espressif/arduino-esp32 #3796: I try to add Factory partition,but failed

Lucy2003, 2020-03-08:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			01/29/2020
IDE name:							Arduino IDE 1.8.9
Flash Frequency:					80Mhz
PSRAM enabled:                                          no
Upload Speed:						921600
Computer OS:                                           Windows 7 x64

### Description:

#### I try to use Factory Partition (I added it by ESP Partition GUI),as this:
[image]
and changed boards.txt like this:
`esp32.menu.PartitionScheme.custom=Custom (1M APP with OTA Factory/896K SPIFFS)
esp32.menu.PartitionScheme.custom.build.partitions=custom
esp32.menu.PartitionScheme.custom.upload.maximum_size=1048576`

#### But it didn't boot from Factory partition ,it booted from app0 / app1 (I update it by OTA before)
The OTA Data(boot_app0.bin) showed that it should be booted from 0x10000.

##### I just want to keep the factory programs in the ESP32 so that I can boot from factory program when I need.If there is another way to do this,please tell me.

### Sketch: 
```cpp
// I used example->HttpUpdate -> HttpUpdate  sketch,and added this line in the factory sketch and uploaded by Serial:
Serial.println("Hello,I'm Factory");  
// And add this line to OTA sketch and put its .bin to my local server:
Serial.println("Hello,I'm OTA");
```

### Debug Messages:
```
Hello,I'm OTA
```

closed 2020-03-09 by Lucy2003

Sergio-Navarro-M, 2021-12-16:
> ### I read document again,and find the way to solve it : Just need to erase the flash !

Hello Lucy. I want to be able to do this (use OTA but go back to factory if needed). Could you give me a hint ablut what you have done?
 I have tried to generate the bootloader with the factory reset option but the results are not satisfactory (a couple of chips bricked)

Lucy2003, 2021-12-19:
Oh,I'm sorry to tell you that since this problem is solved long ago,I didn't remember it clearly.All the steps are discribed above.Good luck!