espressif/arduino-esp32 #3797: Incorrect DST for Europe with NTP Time

thierry7100, 2020-03-08:
### Hardware:
Board:							WEMOS LOLIN D32
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                  ?
Upload Speed:					115200
Computer OS:                                        Ubuntu

### Description:
Daylight Saving is not correct for Europe. Today, March 8th getLocalTime report as in summer time, but in Europe we switch from winter time to summer time the last sunday of march, not the first one. 

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
#include <WiFi.h>
#include "time.h"

const char* ssid       = "YOUR_SSID";
const char* password   = "YOUR_PASS";

const char* ntpServer = "pool.ntp.org";
const long  gmtOffset_sec = 3600;    //  France, CET (GMT+1)
const int   daylightOffset_sec = 3600;

void printLocalTime()
{
  struct tm timeinfo;
  if(!getLocalTime(&timeinfo)){
    Serial.println("Failed to obtain time");
    return;
  }
  Serial.println(&timeinfo, "%A, %B %d %Y %H:%M:%S");
}

void setup()
{
  Serial.begin(115200);
  
  //connect to WiFi
  Serial.printf("Connecting to %s ", ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
  }
  Serial.println(" CONNECTED");
  
  //init and get the time
  configTime(gmtOffset_sec, daylightOffset_sec, ntpServer);
  printLocalTime();

  //disconnect WiFi as it's no longer needed
  WiFi.disconnect(true);
  WiFi.mode(WIFI_OFF);
}
```
Result is 
Sunday, March 08 2020 11:31:31
but correct time is 10:31:31

robin-debug, 2020-03-08:
same error in chian!
`const char* ntpServer = "pool.ntp.org";
const long  gmtOffset_sec = 8 * 3600;
const int   daylightOffset_sec = 8 * 3600;
...
configTime(gmtOffset_sec, daylightOffset_sec, ntpServer);`

output: Monday, March 09 2020 07:27:35
should be Sunday, March 08 2020 23:27:35

CelliesProjects, 2020-03-08:
You can use a TZ string to explicitly set TZ and DST moments.
I use something like this to set TZ:

````c++
  const char * NTPpool         = "nl.pool.ntp.org";
  const char * defaultTimezone = "CET-1CEST,M3.5.0/2,M10.5.0/3";

  configTzTime( defaultTimezone, NTPpool);

  struct tm timeinfo = {0};

  while ( !getLocalTime( &timeinfo, 0 ) )
    vTaskDelay( 10 / portTICK_PERIOD_MS );
````
`defaultTimezone` sets the TZ and DST settings where:

- M3 = month 3
- 5 = week 5
- 0 = day 0 (sunday)
- 2 = hour 2 (when to switch to DST)

Same applies for switching back to 'normal' time:
- M10 = month 10
- 5 = week 5 
- etc

See a list of timezones [here](https://remotemonitoringsystems.ca/time-zone-abbreviations.php).

closed 2020-05-21 by stale[bot]