espressif/arduino-esp32 #3798: ESP32 stuck in boot loop

Polishfuze, 2020-03-08:
I'm trying to boot my DIY ESP32 Board.
I can't even flash firmware to my esp32, cause there is a boot error.
Plugging in the board into my computer gives me this message on the serial monitor.

```
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
```

I don't even know how should I approach this problem

I have the ESP32 VROVER module on my board (from LCSC C129145), and the board schematic is basically the devkit module, without the communication circuitry, as that is handled by a USB to UART converter. (this is the schematic link https://imgur.com/a/rd0YE1V)

stickbreaker, 2020-03-08:
Do you have anything hooked up to pins 6..11 or 12?

Chuck.

Polishfuze, 2020-03-08:
No, my board is wired up exactly as in the schematic with the addition of a UART to USB adapter hooked up to pins RX TX 3V3 and GND ,

chegewara, 2020-03-08:
> ets Jun  8 2016 00:22:57
rst:0x10 (RTCWDT_RTC_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371

Most likely esp32 is clean, nothing flashed on it.

> No, my board is wired up exactly as in the schematic with the addition of a UART to USB adapter hooked up to pins RX TX 3V3 and GND ,

How about CTS and RTS pins? You need to switch esp32 into download mode.
If you dont have CTS/RTS you need to use pin0 and EN to enter download mode.

stickbreaker, 2020-03-08:
As @chegewara suggested you might have a new "clean" esp32 mondule without any code on it.
[flash read err: 1000](https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection#early-flash-read-error)

Chuck.

Polishfuze, 2020-03-08:
Yeah it may be the case as the esp32 boards arrived in tape and reel packaging. I'm sorry to bother you a bit more but how should I go about flashing the bootloader? If you could direct me to someone kind of tutorial that would be great. Again thank you so much for your help

stickbreaker, 2020-03-08:
@Polishfuze The "boot loader" is built into the ROM. You have a couple of choices of, if you want to enable auto loading from the Arduino or IDF you will need to:
* add the circuit to Reset the module and HOLD GPIO0 low to enter the boot loader.
* or you can add grounding switches to EN and IO0. 

 After you get the hardware setup , you need to write your application and use the ` esptool.py` to upload the code.

Chuck.

Polishfuze, 2020-03-08:
I was able to upload a single program (GetChipID example) and now I get this error, and the controller is stuck in a boot loop. Now when I try to upload a file I either get timeout or

A fatal error occurred: Failed to connect to ESP32: Invalid head of packet (0x65)
A fatal error occurred: Failed to connect to ESP32: Invalid head of packet (0x65)

The error that shows up on the serial monitor is:
`
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x00000003,len:53674752
ets Jun  8 2016 00:22:57
`
Any help would be appreciated

Polishfuze, 2020-03-08:
Turns out that my first issue of not being able to flash anything new came down to low quality cables being low quality, but my second issue still stands, even with different sketches

stickbreaker, 2020-03-09:
It looks to be a sequencing problem between EN and GPIO0,  GPIO0 needs to be held low after reset(EN) to get the boot loader to listen for an upload.  The normal sequence is hold GPIO low, toggle EN low , release GPIO0 after the upload starts.  esptool.py will be outputing a ...---.... pattern while it tries to communicate [esptool.py docuementation](https://github.com/espressif/esptool#entering-the-bootloader)

Chuck.

Polishfuze, 2020-03-09:
Yeah yeah, thanks for the help with uploading but now I have the problem that no sketch no matter how simple doesn't want to work, instead the ESP32 gets stuck in a boot loop sending out the message below on the serial monitor each 300ms. I'm sorry to still bug you with this problem, but this is the first time I have encountered this issue.

ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x00000003,len:53674752

And this is literally the code that sparks this issue

```
void setup()
{
  Serial.begin(115200);
  delay(1000);
  Serial.println("ESP32 Test");
}

void loop()
{
  Serial.println("Hello");  
  delay(1000);
}

```

stickbreaker, 2020-03-09:
I would say your flash is corrupt,  That last line:
```
load:0x00000003,len:53674752
```
Is the Boot loader loading a block into address 0 that is 53,674,752 byte long?

Erase the flash, upload a partition table that matches you hardware and reload your code.

Chuck.

stickbreaker, 2020-03-09:
@Polishfuze this is what your code produced on a "Heltec WiFi LoRa 32" with 4MB flash, no PSDRAM.
```

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6364
entry 0x400806b8
ESP32 Test
Hello
Hello
Hello
Hello
Hello
Hello
```

Chuck.

chegewara, 2020-03-11:
> I was able to upload a single program (GetChipID example) and now I get this error, and the controller is stuck in a boot loop. Now when I try to upload a file I either get timeout or
>
> A fatal error [...]

Ok, i got recently the same issue. Its not usual to check this line:
`rst:0x10 (RTCWDT_RTC_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)`
but it is very important.

`boot: 0x37` means that you have pin12 in high state and most likely flash is running with different VDD_SDIO voltage:
https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection#boot-mode-message
https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection#other-pins

closed 2020-05-24 by stale[bot]

Mr-Newlove, 2020-07-27:
Sorry to try to re-open a closed issue, but I am having this problem I have a simple sketch:

`

void setup() {
	Serial.begin(115200);
	while (!Serial);
	Serial.println(F("Test"));
	Serial.flush();
	delay(100);
	Serial.end();

	esp_sleep_enable_timer_wakeup(1000000 * 5);

	esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_SLOW_MEM, ESP_PD_OPTION_OFF);
	esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_FAST_MEM, ESP_PD_OPTION_OFF);
	esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);
	esp_sleep_pd_config(ESP_PD_DOMAIN_XTAL, ESP_PD_OPTION_OFF);
	esp_sleep_pd_config(ESP_PD_DOMAIN_MAX, ESP_PD_OPTION_OFF);

	esp_deep_sleep_start();
`

Then it will run alone on its own for quite some time, but then randomly(seems like about 0.5% of the time) It will then get stuck in a boot look with:

`rst:0x7 (TG0WDT_SYS_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371`

HenniePeters, 2023-08-06:
I had the same problem with the development board connected to the PC USB connector. It is a power problem. I could solve it by soldering a 1500uF capacitor over the 3.3V pins.

sebosfato, 2023-11-10:
Hello i kind of had the same problem it seems that i used the max3485 hardwired to RX pin it causes this error…. adding a 2.2kohm resistor solved the issue.. the output impedance of the max is too low so it wont allow the comunication to serial.

santoshurlam, 2024-07-01:
[image]
I am using esp32 wroom module , when I'm uploading the esp_rainmaker switch example code it is uploading without any error but in serial monitor it is giving bootloop issue .
My tool settings picture is below;
[image]
Provide me a solution to this issue, it should generate a QR code in serial monitor instead of this it is generating bootloop issue.

santoshurlam, 2024-11-25:
> @santoshurlam did you solve your problem? I have the same boot loop issue on 0x13

No