espressif/arduino-esp32 #3801: BLE Disconnected when calling esp_ota_begin() method

dflogeras, 2020-03-09:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                        Linux

### Description:
We are attempting to send packets of data over BLE to do an OTA update.  We devised a protocol which first queries the device and opens the OTA partition, then subsequently sends data packets in chunks of 1024 bytes to be written to the OTA partition.

We have noticed that very frequently, the BLE client gets disconnected when we call esp_ota_begin().  We've tried with an iPhone running our custom app, as well as a Linux laptop and an Android phone running nRF Connect.  The iPhone and laptop fail almost every time.  nRF Connect works more often, but still will lose connection some times.

This is strikingly similar to the IDF bug reported here:
https://github.com/espressif/esp-idf/issues/3551#

However, I have attempted to set the connection parameters as reported as fixing their issue, but it did not fix it for me. (I added a call to BLEServer::updateConnParams() in my onConnect() callback).

I have boiled it down to a simple test case.  It is based on the BLE Uart example provided with ESP32 arduino, then I added logic so that if a client sends any data to the WRITE characteristic, it waits one second then attempts to esp_ota_begin().  This reproduced the issue without the rest of my firmware.

I am running my cpu at 80MHz and with a 1.9MB OTA/190kB SPIFFS, but I tried with the defaults and it loses connection with that as well.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

/*
    Video: https://www.youtube.com/watch?v=oCMOYS71NIU
    Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleNotify.cpp
    Ported to Arduino ESP32 by Evandro Copercini

   Create a BLE server that, once we receive a connection, will send periodic notifications.
   The service advertises itself as: 6E400001-B5A3-F393-E0A9-E50E24DCCA9E
   Has a characteristic of: 6E400002-B5A3-F393-E0A9-E50E24DCCA9E - used for receiving data with "WRITE" 
   Has a characteristic of: 6E400003-B5A3-F393-E0A9-E50E24DCCA9E - used to send data with  "NOTIFY"

   The design of creating the BLE server is:
   1. Create a BLE Server
   2. Create a BLE Service
   3. Create a BLE Characteristic on the Service
   4. Create a BLE Descriptor on the characteristic
   5. Start the service.
   6. Start advertising.

   In this example rxValue is the data received (only accessible inside that function).
   And txValue is the data to be sent, in this example just a byte incremented every second. 
*/
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>
#include <esp_ota_ops.h>

BLEServer *pServer = NULL;
BLECharacteristic * pTxCharacteristic;
bool deviceConnected = false;
bool oldDeviceConnected = false;
uint8_t txValue = 0;

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID           "6E400001-B5A3-F393-E0A9-E50E24DCCA9E" // UART service UUID
#define CHARACTERISTIC_UUID_RX "6E400002-B5A3-F393-E0A9-E50E24DCCA9E"
#define CHARACTERISTIC_UUID_TX "6E400003-B5A3-F393-E0A9-E50E24DCCA9E"

static bool do_open( false );

class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer, esp_ble_gatts_cb_param_t* param ) {
[... 162 lines cut ...]
*********
[D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
opening
done opening
[D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:571] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:200] start(): - no services advertised
[D][BLEDevice.cpp:571] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:495] handleGAPEvent(): handleGAPEvent [event no: 0]
[D][BLEDevice.cpp:571] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:495] handleGAPEvent(): handleGAPEvent [event no: 1]
[D][BLEDevice.cpp:571] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:495] handleGAPEvent(): handleGAPEvent [event no: 6]
[D][BLEDevice.cpp:571] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:200] start(): - no services advertised
start advertising
[D][BLEDevice.cpp:571] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:495] handleGAPEvent(): handleGAPEvent [event no: 0]
[D][BLEDevice.cpp:571] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:495] handleGAPEvent(): handleGAPEvent [event no: 1]
[D][BLEDevice.cpp:571] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:495] handleGAPEvent(): handleGAPEvent [event no: 6]

```

chegewara, 2020-03-09:
When you are using nRF connect you should be able to find disconnect reason. It would be very helpful.

dflogeras, 2020-03-09:
Do you know where they show that?  I cannot seem to find it.

dflogeras, 2020-03-09:
Nevermind, I see it in the 'show log' tab.  Now I cannot seem to reproduce it on nrf connect.  I'll keep trying.

felixcollins, 2022-05-26:
We are experiencing the same issue but only when starting OTA over BLE from iOS. Our code works for Android but fails with disconnect due to timeout when using iOS.  What values should I try in  "pServer->updateConnParams( )" ?  

It should also be noted that not only does the C++ wrapper (updateConnParams()) silently ignore failed calls to  esp_ble_gap_update_conn_params(), but the parameter checking in the esp_ble_gap_update_conn_params() itself is buggy. https://github.com/espressif/esp-idf/issues/9037

edwardocano, 2023-12-14:
I have the exact error that @felixcollins with ios, could you solve it?