espressif/arduino-esp32 #3802: Bluetooth: SerialBT.write not working adding extra lines

ZiTAL, 2020-03-09:
SerialBT.write not working adding extra lines, If I remove some **bluetoothPrintLine** works... when I try to add the 15th line program stop working...

```
#include "BluetoothSerial.h"

void bluetoothPrintLine(String line)
{
    unsigned l = line.length();
    for(int i=0; i<l; i++)
    {
        if(line[i]!='\0')
            SerialBT.write(byte(line[i]));
    }
    SerialBT.write(10); // \n
}

String bluetoothReadLine()
{
    String response = "";
    while(SerialBT.available())
    {
        byte r = SerialBT.read();
        if(r!=13 && r!=10 && char(r)!='\0')
            response = response + char(r);
    }
    return response;      
}

void bluetoothCallback(esp_spp_cb_event_t event, esp_spp_cb_param_t *param)
{
    if(event == ESP_SPP_SRV_OPEN_EVT)
    {
        Serial.println("Connection established");
        menu(0);
    }
    else if(event == ESP_SPP_CLOSE_EVT)
    {
        Serial.println("Connection closed");
        ESP.restart();
    }
    else if(event == ESP_SPP_DATA_IND_EVT)
    {
        Serial.println("Data received");
        String response = bluetoothReadLine();
        if(response=="")
        {
            Serial.println("EMPTY");
        }
    }
    else if(event == ESP_SPP_WRITE_EVT)
    {
        Serial.println("Write operation complete");
    }
}

void setup() {
    Serial.begin(115200);
    while (!Serial)
    {
    }
    SerialBT.register_callback(bluetoothCallback);
    bluetoothPrintLine("1");
    bluetoothPrintLine("2");
    bluetoothPrintLine("3");
    bluetoothPrintLine("4");
    bluetoothPrintLine("5");
    bluetoothPrintLine("6");
    bluetoothPrintLine("7");
    bluetoothPrintLine("8");
    bluetoothPrintLine("9");
    bluetoothPrintLine("10");
    bluetoothPrintLine("11");
    bluetoothPrintLine("12");
    bluetoothPrintLine("13");
    bluetoothPrintLine("14");
//    bluetoothPrintLine("15"); // if i add this line stop working WTF
}

void loop() {
}
```

```
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			?
IDE name:							VSCODE
Flash Frequency:					?
PSRAM enabled:                                          ?
Upload Speed:						115200
Computer OS:                                            debian gnu/linux
```

sansillusion, 2020-03-09:
Try adding delay(1);
Between each lines. I noticed while playing with Bluetooth serial that it
seems to need to be given cpu cycles to empty it's buffer.

ZiTAL, 2020-03-10:
adding delay doesn't work, I try to ```SerialBT.flush();``` at the end of **bluetoothPrintLine** but still not working :(

copercini, 2020-03-11:
Isn't easier use `SerialBT.println("your string");`  ?

ZiTAL, 2020-03-11:
breaks at number 17:
```
            SerialBT.println("1");
            SerialBT.println("2");
            SerialBT.println("3");
            SerialBT.println("4");
            SerialBT.println("5");
            SerialBT.println("6");
            SerialBT.println("7");
            SerialBT.println("8");
            SerialBT.println("9");
            SerialBT.println("10");
            SerialBT.println("11");
            SerialBT.println("12");
            SerialBT.println("13");
            SerialBT.println("14");
            SerialBT.println("15");
            SerialBT.println("16");
            SerialBT.println("17"); // breaks here
```

copercini, 2020-03-11:
ok, it's because the callback wasn't designed to run all this code, try just set a flag on it and keep running the SerialBT.println() in loop() for example:

```c
#include "BluetoothSerial.h"

bool BT_connected = false;
int i =0;

String bluetoothReadLine()
{
    String response = "";
    while(SerialBT.available())
    {
        byte r = SerialBT.read();
        if(r!=13 && r!=10 && char(r)!='\0')
            response = response + char(r);
    }
    return response;      
}

void bluetoothCallback(esp_spp_cb_event_t event, esp_spp_cb_param_t *param)
{
    if(event == ESP_SPP_SRV_OPEN_EVT)
    {
        Serial.println("Connection established");
        BT_connected = true;
    }
    else if(event == ESP_SPP_CLOSE_EVT)
    {
        Serial.println("Connection closed");
        BT_connected = false;
        ESP.restart();
    }
    else if(event == ESP_SPP_DATA_IND_EVT)
    {
        Serial.println("Data received");
        String response = bluetoothReadLine();
        if(response=="")
        {
            Serial.println("EMPTY");
        }
    }
    else if(event == ESP_SPP_WRITE_EVT)
    {
        Serial.println("Write operation complete");
    }
}

void setup() {
    Serial.begin(115200);
    while (!Serial)
    {
    }
    SerialBT.register_callback(bluetoothCallback);
}

void loop() {

    if(BT_connected ){
        SerialBT.println(i);
        i++;
        delay(100);
    }
}
```

ZiTAL, 2020-03-12:
```
#include "BluetoothSerial.h"

BluetoothSerial SerialBT;

bool BT_connected = false;
int i =0;

String bluetoothReadLine()
{
    String response = "";
    while(SerialBT.available())
    {
        byte r = SerialBT.read();
        if(r!=13 && r!=10 && char(r)!='\0')
            response = response + char(r);
    }
    return response;      
}

void bluetoothCallback(esp_spp_cb_event_t event, esp_spp_cb_param_t *param)
{
    if(event == ESP_SPP_SRV_OPEN_EVT)
    {
        Serial.println("Connection established");
        BT_connected = true;
    }
    else if(event == ESP_SPP_CLOSE_EVT)
    {
        Serial.println("Connection closed");
        BT_connected = false;
        ESP.restart();
    }
    else if(event == ESP_SPP_DATA_IND_EVT)
    {
        Serial.println("Data received");
        String response = bluetoothReadLine();
        if(response=="")
        {
            Serial.println("EMPTY");
        }
    }
    else if(event == ESP_SPP_WRITE_EVT)
    {
        Serial.println("Write operation complete");
    }
}

void setup() {
    Serial.begin(115200);
    while (!Serial)
    {
    }
    SerialBT.register_callback(bluetoothCallback);
    SerialBT.begin("BLE");
}

void loop() {

    if(BT_connected ){
        SerialBT.println(i);
        i++;
        delay(100);
    }
}
```
@copercini I fixed your code to work, but it's not the envinromnet that i'm looking for, anyway I will used the bluetooth callback to edit an global state var and work with it in the loop function, thank you all

ZiTAL, 2020-03-12:
```
#include "BluetoothSerial.h"

BluetoothSerial BT_SERIAL;

String BT_CALLBACK = "";
String BT_READ_LINE = "";

void BTCallback(esp_spp_cb_event_t event, esp_spp_cb_param_t *param)
{
    if(event == ESP_SPP_SRV_OPEN_EVT)               // CONNECTION STABLISHED
    {
        BT_CALLBACK = "BTonConnect";
    }
    else if(event == ESP_SPP_CLOSE_EVT)             // CONNECTION CLOSED
    {
        ESP.restart();
    }
    else if(event == ESP_SPP_DATA_IND_EVT)          // DATA RECEIVED
    {
        BT_CALLBACK = "BTonReceivedStart";
    }
    else if(event == ESP_SPP_WRITE_EVT)             // DATA WRITED
    {
        BT_CALLBACK = "BTonWrite";
    }
}

void BTonReceivedStart()
{
    if(BT_SERIAL.available())
    {
        byte r = BT_SERIAL.read();
        BT_READ_LINE = BT_READ_LINE + char(r);
    }
    else
        BTonReceivedEnd();
}

void BTonReceivedEnd()
{
    Serial.println(BT_READ_LINE);
    BT_CALLBACK = "";
    BT_READ_LINE = "";
}
 
void setup()
{
    Serial.begin(115200);
    while (!Serial)
    {
    }
    BT_SERIAL.register_callback(BTCallback);
    BT_SERIAL.begin("BT");
}

void loop()
{
    if(BT_CALLBACK=="BTonReceivedStart")
    {
        BTonReceivedStart();
    }
}
```

this is the way to do it, i don't proud of myself for this kind of code, but it works

closed 2020-05-25 by stale[bot]