espressif/arduino-esp32 #3803: [QUESTION]: WiFiSTA - Why is autoConnect Code commented out?

hasenradball, 2020-03-09:
Why ist the code for WiFi.autoConnect commented out?

```
/**
 * Setting the ESP32 station to connect to the AP (which is recorded)
 * automatically or not when powered on. Enable auto-connect by default.
 * @param autoConnect bool
 * @return if saved
 */
bool WiFiSTAClass::setAutoConnect(bool autoConnect)
{
    /*bool ret;
    ret = esp_wifi_set_auto_connect(autoConnect);
    return ret;*/
    return false;//now deprecated
}

/**
 * Checks if ESP32 station mode will connect to AP
 * automatically or not when it is powered on.
 * @return auto connect
 */
bool WiFiSTAClass::getAutoConnect()
{
    /*bool autoConnect;
    esp_wifi_get_auto_connect(&autoConnect);
    return autoConnect;*/
    return false;//now deprecated
}
```

lbernstone, 2020-03-10:
It has been deprecated in esp-idf

hasenradball, 2020-03-10:
What is the reason for that?
Does it not work?

atanisoft, 2020-03-10:
@hasenradball it is enabled by default if you call WiFi.begin() (no args). The esp-idf code doesn't have a public API to expose this or one that is supported.

atanisoft, 2020-03-10:
You could also look at that commented code as "planned support that never materialized"

hasenradball, 2020-03-10:
@atanisoft thanks for your answer.
Did I understood you right, that the autoconnect should work by default?
I tried but it doesn‘t with my esp32.

atanisoft, 2020-03-10:
It depends on how you use the WiFi library. It will only reconnect to the last SSID if persistence is enabled and you call the WiFi.begin() API. Note that if you lose connection without a restart it may not reconnect without your app recalling WiFi.begin().

hasenradball, 2020-03-10:
Ok understood, so this is a difference to the small brother 8266, there is no need to call WiFi.begin() when connection is lost, or during restart without calling WiFi.begin().

lbernstone, 2020-03-10:
Autoconnect means the device uses saved credentials to connect immediately at boot.
Reconnect means to retry if a connection is lost.

hasenradball, 2020-03-11:
Yes so is my understanding! And when the esp32 needs a wifi.begin() to connect, then the esp32 has actually NO autoconnnect and autoreconnect!
But why???

hasenradball, 2020-03-11:
Why can‘t we do like the 8266?

hasenradball, 2020-05-10:
A Anwer is left, actaually.

ElToberino, 2020-06-07:
Hello stale bot... ,
this question is still pending. 
I'd also prefer an auto connect and auto reconnect mode without the need of calling WiFi.begin()