espressif/arduino-esp32 #3804: HTTP GET failed... connection refused

manish-hello-gupta, 2020-03-10:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                      no
Upload Speed:					115200
Computer OS:                                         Windows 10

### Description:
 I am trying to fetch values of columns from a table name **traffic1** in database **test** of a phpmyadmin in localserver. I am using Xamppserver. 2-3 months before the code works perfectly. But now it shows error in connecting.The error is
**HTTPCODE: -1
[HTTP] GET... failed, error: connection refused
0
0
0
0**
 The four zeroes are of x[4] array's elements values which do not get change 
### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>
#include <WiFi.h>
#include <WiFiMulti.h>
#include <HTTPClient.h>

WiFiMulti wifiMulti;
void setup() {

  Serial.begin(115200);
  Serial.println();
  for (uint8_t t = 4; t > 0; t--) {
    Serial.printf("[SETUP] WAIT %d...\n", t);
    Serial.flush();
    delay(1000);
  }

  wifiMulti.addAP("Redmi", "IDEA!23#");

}

void loop() {
  int x[4] = {0} ;
  // wait for WiFi connection
  if ((wifiMulti.run() == WL_CONNECTED)) {

    HTTPClient http;

    Serial.print("[HTTP] begin...\n");

    // configure server and url
    http.begin("http://localhost/testcode/show1.php");
    //http.begin("192.168.1.12", 80, "/test.html");
    
    Serial.print("[HTTP] GET...\n");
    // start connection and send HTTP header
    int httpCode = http.GET();
    Serial.print("HTTPCODE: ");
    Serial.println(httpCode);
    if (httpCode > 0) {
      // HTTP header has been send and Server response header has been handled
      Serial.printf("[HTTP] GET... code: %d\n", httpCode);

      // file found at server
      if (httpCode == HTTP_CODE_OK) {

        // get lenght of document (is -1 when Server sends no Content-Length header)
        int len = http.getSize();
        Serial.println(len);
        // create buffer for read
        uint8_t buff[128] = { 0 };

        // get tcp stream
        WiFiClient * stream = http.getStreamPtr();

        // read all data from server
        size_t size = stream->available();
        while (http.connected() && (len > 0 || len == -1)) {
          // get available data size
          if (size) {
            // read up to 128 byte
            int c = stream->readBytes(buff, ((size > sizeof(buff)) ? sizeof(buff) : size));

            // write it to Serial
            Serial.write(buff, c);
            if (len > 0) {
              len -= c;
            }
          }
          delay(1);
        }
        Serial.println();
        int j = 0;
        for (int i = 0; i < size; i++)
        {
          if (j == 4)
          {
            break;
          }
          if (buff[i] == ' ')
          {
            j++;
          }
          else {
            x[j] *= 10;
            x[j] += (int)buff[i];
            x[j] -= 48;
          }
        }

        Serial.print("[HTTP] connection closed or file end.\n");
      }
    }
    else {
      Serial.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
    }
    http.end();
  }
  for (int k = 0; k < 4; k++)
  {
    Serial.println(x[k]);
  }
  delay(10000);
}
```

### Debug Messages:
```
[E][WiFiClient.cpp:258] connect(): socket error on fd 60, errno: 104, "Connection reset by peer"
[W][HTTPClient.cpp:1318] returnError(): error(-1): connection refused
```

atanisoft, 2020-03-10:
You need to provide the actual hostname where your php service is running. Localhost means exactly what it sounds like, local to the device (ie the esp32 itself).

manish-hello-gupta, 2020-03-10:
@atanisoft  do you mean the IP address of the server? If so , I've tried that before also, but still connection refused.

atanisoft, 2020-03-10:
@manish-hello-gupta Yes, that is precisely what is needed. Your code sample only shows using localhost as the destination. If you have tested with the actual hostname or ip address and it still shows connection refused does the log statements you posted above still say localhost or does it show the ip address? If it shows the ip address make sure the port etc are correct.

manish-hello-gupta, 2020-03-10:
@atanisoft I just change the network(basically trying on another mobile's hotspot) and also apply what you say(though i was expecting the same result) but this time my code works.And then i again try the same on former network with applying what you say but still connection refused.
The error message was **[E][WiFiClient.cpp:258] connect(): socket error on fd 61, errno: 113, "Software caused connection abort"
[D][HTTPClient.cpp:1018] connect(): failed connect to 192.168.43.181:80
[W][HTTPClient.cpp:1318] returnError(): error(-1): connection refused
[HTTP] GET... failed, error: connection refused
[D][HTTPClient.cpp:383] disconnect(): tcp is closed** So, the conclusion is may be there is a problem in former network.Thank you for help.

lbernstone, 2020-03-10:
You are failing to connect to the server.  Are you sure the server is listening on port 80 (unencrypted)?  Are you sure you have a route to the server?

manish-hello-gupta, 2020-03-11:
@lbernstone problem is resolved. probelm was in the network i using. So, i connect the devices on another network and my code works.

closed 2020-03-11 by manish-hello-gupta