espressif/arduino-esp32 #3805: is there way to reduce data receiving speed in ESP32 WROOM module??

Cembedded, 2020-03-11:
We are sending converted image data from **mobile app to ESP32 WROOM module** through **Bluetooth in Classic SPP mode**. 
Data receiving speed is faster,this makes the receiving buffer(size 55000 bytes) overflow and loss of data after overflow.
Received image data takes some microseconds time for processing, Meanwhile receive buffer overflow and data loss occurs.
Need solution to solve this issue , is there way to reduce data receiving speed in ESP32 WROOM module.

manish-hello-gupta, 2020-03-11:
i don't know but try after lowering down the cpu frequency

chegewara, 2020-03-11:
Only way is to use arduino as component and change settings in menuconfig or recompile bluetooth library with new settings (speed you need).
Bluetooth library is pre-compiled and changes in sdkconfig in arduino wont make any difference.

closed 2020-05-24 by stale[bot]