espressif/arduino-esp32 #3806: Sleep mode still pulling 10mA

niallmcauley, 2020-03-11:
Hi, when I use the deep sleep example code on my PCB  I am still drawing 10mA in deep sleep mode. Does anyone know of a further measure to reduce the current or why my current is not falling to the uA range?
PCB: https://www.banggood.com/Geekcreit-SX1278-LoRa-ESP32-bluetooth-WIFI-Lora-Internet-Antenna-Development-Board-p-1214812.html?utm_source=EUsite20200303&cur_warehouse=CN

chegewara, 2020-03-11:
https://esp32.com/viewtopic.php?t=10222
and few more similar on forum.

rtek1000, 2020-03-15:
See also the MCP1700 regulator (1.6µA of quiescent current), as an alternative to minimize energy waste. (AP7361C: Quiescent Current (IQ): 60μA Typical)

G6EJD, 2020-04-10:
You have a development board issue nothing to do with ESP32

closed 2020-06-23 by stale[bot]