espressif/arduino-esp32 #3809: How to enable CONFIG_MBEDTLS_SSL_PROTO_DTLS with "Arduino IDE Boards Manager"?

alorbach, 2020-03-11:
I am developing an LED Controller on ESP32 and  for my project I need support for DTLS in mbedtls. 
Unfortunatelly it turns out that support MBEDTLS_SSL_PROTO_DTLS is not enabled by default in sdkconfig. 

So have no clue on ESP-IDF yet - but all references go to there. 

Is there any easy way ( I hope so )  to change the sdkconfig, enable "CONFIG_MBEDTLS_SSL_PROTO_DTLS" and rebuild this for "Arduino IDE Boards Manager"?

lbernstone, 2020-03-11:
There is no easy way.
You will need to [compile your own libraries](https://github.com/espressif/arduino-esp32/blob/master/docs/lib_builder.md)

alorbach, 2020-03-12:
I found a hint how to compile the user libraries here: 
https://github.com/espressif/arduino-esp32/issues/3410

I just replaced sdkconfig and lib/include files for mbedtls in my Arduino IDE Boards Manager esp32 files, and got DTLS support. 

Now I am fighting with exceptions during the DTLS Handshake ;)

alorbach, 2020-03-13:
I got it working finally. 
I created a wrapper class based on WifiClientSecure, maybe this is help for somebody else: 

https://github.com/alorbach/wificlientsecureudp

alorbach, 2020-05-05:
Any chance to get DTLS enabled (CONFIG_MBEDTLS_SSL_PROTO_DTLS=y) with the next esp32 release?

closed 2020-07-18 by stale[bot]