espressif/arduino-esp32 #3813: Olimex ESP32-EVB smtp connection fails only with Ethernet

atlantisguru, 2020-03-12:
Board:							Olimex ESP32-EVB
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    yes
Upload Speed:				      115200
Computer OS:                                        Windows 10

Description:
I would like to send email with only via Ethernet, but it doesn't works. SMTP connection fails on identification (not authentication) part. 
With WiFi it works perfect. I don't want to use WiFi.
It's a mail service independent problem. Credentials are all ok.
I already discussed the problem with the creator of ESP32_MailClient library and that works perfect.

//Relevant parts of the code
#include <ETH.h>
#include "ESP32_MailClient.h"
...

void setup() {
...ethernet started/connected
...got IP from DHCP
}

void loop() {
...sending email triggered by an event
}

Debug Messages:
Connecting to SMTP server...
[DEBUG] - INFO: connecting to SMTP server...
[DEBUG] - smtp.gmail.com
[DEBUG] - 587
[DEBUG] - INFO: starting socket
[DEBUG] - INFO: connecting to Server...
[DEBUG] - INFO: server connected
[DEBUG] - INFO: begin STARTTLS handshake
[DEBUG] - 220 smtp.gmail.com ESMTP c85sm13831667wmd.48 - gsmtp
⸮!Pk
[DEBUG] - INFO: send SMTP command extended HELO
[DEBUG] - 250-smtp.gmail.com at your service, [5.63.208.133]
250-SIZE 35882577
250-8BITMIME
250-STARTTLS
250-ENHANCEDSTATUSCODES
250-PIPELINING
250-CHUNKING
250 SMTPUTF8

[DEBUG] - INFO: send STARTTLS protocol command
[DEBUG] - 220 2.0.0 Ready to start TLS

[DEBUG] - INFO: seeding the random number generator
[DEBUG] - INFO: setting up the SSL/TLS structure...
[DEBUG] - INFO: setting hostname for TLS session...
[DEBUG] - INFO: performing the SSL/TLS handshake...
[DEBUG] - INFO: verifying peer X.509 certificate...
[DEBUG] - INFO: certificate verified
[DEBUG] - INFO: smtp server connected
Identification...
[DEBUG] - INFO: send smtp HELO command
Error, identification failed    //<- With WiFi can pass trough, with Ethernet can't
[DEBUG] - ERROR: identification failed
[DEBUG] - INFO: cleaning SSL connection
Error sending Email, identification failed

atlantisguru, 2020-03-14:
Additional information:
With Wireshark i did catch the communication between SMTP server and ESP.
It seems when the "Identification..." comes, the board suddenly resetting the communication (Line ID 3066).

No.	Time	Source	Destination	Protocol	Length	Info
3033	1591.392626	192.168.137.2	108.177.15.108	TCP	58	52839 → 587 [SYN] Seq=0 Win=5744 Len=0 MSS=1436
3034	1591.451031	108.177.15.108	192.168.137.2	TCP	58	587 → 52839 [SYN, ACK] Seq=0 Ack=1 Win=62920 Len=0 MSS=1430
3035	1591.451720	192.168.137.2	108.177.15.108	TCP	54	52839 → 587 [ACK] Seq=1 Ack=1 Win=22976 Len=0
3036	1591.510841	108.177.15.108	192.168.137.2	SMTP	107	S: 220 smtp.gmail.com ESMTP n1sm40021609wrj.77 - gsmtp
3037	1591.510913	192.168.137.2	108.177.15.108	TCP	54	52839 → 587 [ACK] Seq=1 Ack=54 Win=22976 Len=0
3038	1591.920478	192.168.137.1	224.0.0.251	MDNS	102	Standard query response 0x0000 A, cache flush 192.168.137.1
3039	1592.452915	192.168.137.2	108.177.15.108	SMTP	65	C: EHLO DUDE
3040	1592.540947	108.177.15.108	192.168.137.2	TCP	54	587 → 52839 [ACK] Seq=54 Ack=12 Win=62920 Len=0
3041	1592.548404	108.177.15.108	192.168.137.2	SMTP	221	S: 250-smtp.gmail.com at your service, [5.63.214.69] | SIZE 35882577 | 8BITMIME | STARTTLS | ENHANCEDSTATUSCODES | PIPELINING | CHUNKING | SMTPUTF8
3042	1592.548522	192.168.137.2	108.177.15.108	TCP	54	52839 → 587 [ACK] Seq=12 Ack=221 Win=22976 Len=0
3043	1593.462134	192.168.137.2	108.177.15.108	SMTP	64	C: STARTTLS
3044	1593.520641	108.177.15.108	192.168.137.2	SMTP	84	S: 220 2.0.0 Ready to start TLS
3045	1593.520684	192.168.137.2	108.177.15.108	TCP	54	52839 → 587 [ACK] Seq=22 Ack=251 Win=22976 Len=0
3046	1594.475617	192.168.137.2	108.177.15.108	TLSv1	357	Client Hello
3047	1594.551024	108.177.15.108	192.168.137.2	SSL	152	[TCP Previous segment not captured] , Continuation Data
3048	1594.551408	108.177.15.108	192.168.137.2	TCP	1484	[TCP Out-Of-Order] 587 → 52839 [ACK] Seq=251 Ack=325 Win=63784 Len=1430
3049	1594.553589	192.168.137.2	108.177.15.108	TCP	54	[TCP Dup ACK 3045#1] 52839 → 587 [ACK] Seq=325 Ack=251 Win=22976 Len=0
3050	1594.553722	192.168.137.2	108.177.15.108	TCP	54	52839 → 587 [ACK] Seq=325 Ack=1681 Win=22976 Len=0
3051	1594.558306	108.177.15.108	192.168.137.2	TCP	1484	[TCP Out-Of-Order] 587 → 52839 [ACK] Seq=1681 Ack=325 Win=63784 Len=1430
3052	1594.558355	192.168.137.2	108.177.15.108	TCP	54	52839 → 587 [ACK] Seq=325 Ack=3209 Win=22976 Len=0
3053	1594.925986	192.168.137.1	224.0.0.251	MDNS	93	Standard query response 0x0000 A, cache flush 192.168.8.101
3054	1595.313210	192.168.137.2	108.177.15.108	TLSv1.2	129	Client Key Exchange
3055	1595.317151	192.168.137.2	108.177.15.108	TLSv1.2	60	Change Cipher Spec
3056	1595.318389	192.168.137.2	108.177.15.108	TLSv1.2	99	Encrypted Handshake Message
3057	1595.380724	108.177.15.108	192.168.137.2	TCP	54	587 → 52839 [ACK] Seq=3209 Ack=406 Win=63784 Len=0
3058	1595.388265	108.177.15.108	192.168.137.2	TLSv1.2	346	New Session Ticket, Change Cipher Spec, Encrypted Handshake Message
3059	1595.401198	192.168.137.2	108.177.15.108	TCP	54	52839 → 587 [ACK] Seq=451 Ack=3501 Win=22976 Len=0
3060	1595.416261	192.168.137.2	108.177.15.108	TLSv1.2	92	Application Data
3061	1595.416793	192.168.137.2	108.177.15.108	TLSv1.2	85	Application Data
3062	1595.422297	192.168.137.2	108.177.15.108	TCP	54	52839 → 587 [FIN, ACK] Seq=520 Ack=3501 Win=22976 Len=0
3063	1595.470711	108.177.15.108	192.168.137.2	TCP	54	587 → 52839 [ACK] Seq=3501 Ack=520 Win=63784 Len=0
3064	1595.478523	108.177.15.108	192.168.137.2	TLSv1.2	119	Application Data
3065	1595.478549	192.168.137.2	108.177.15.108	TCP	54	52839 → 587 [ACK] Seq=521 Ack=3566 Win=22976 Len=0
**3066	1595.479195	192.168.137.2	108.177.15.108	TCP	54	52839 → 587 [RST, ACK] Seq=521 Ack=3566 Win=5744 Len=0**
3067	1595.490816	108.177.15.108	192.168.137.2	TCP	54	587 → 52839 [FIN, ACK] Seq=3566 Ack=521 Win=63784 Len=0
3068	1595.492336	192.168.137.2	108.177.15.108	TCP	54	52839 → 587 [RST] Seq=521 Win=0 Len=0

atlantisguru, 2020-03-15:
Any starting point? Any suggestion? Any hint or something?

atlantisguru, 2020-03-17:
Core level debug:
```
20:28:35.976 -> ets Jun  8 2016 00:22:57
20:28:35.976 -> 
20:28:35.976 -> rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
20:28:35.976 -> configsip: 0, SPIWP:0xee
20:28:35.976 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:28:36.016 -> mode:DIO, clock div:1
20:28:36.016 -> load:0x3fff0018,len:4
20:28:36.016 -> load:0x3fff001c,len:1044
20:28:36.016 -> load:0x40078000,len:8896
20:28:36.016 -> load:0x40080400,len:5816
20:28:36.016 -> entry 0x400806ac
20:28:39.012 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 21 - ETH_STOP
20:28:39.012 -> ETH Started
20:28:42.991 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 25 - MAX
20:28:42.991 -> [D][WiFiGeneric.cpp:417] _eventCallback(): ETH IP: 192.168.137.2, MASK: 255.255.255.0, GW: 192.168.137.1
20:28:42.991 -> 24:0A:C4:21:50:6B, IPv4: 192.168.137.2, FULL_DUPLEX, 100Mbps
20:28:42.991 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 25 - MAX
20:28:43.038 -> [D][WiFiGeneric.cpp:417] _eventCallback(): ETH IP: 192.168.137.2, MASK: 255.255.255.0, GW: 192.168.137.1
20:28:43.038 -> 24:0A:C4:21:50:6B, IPv4: 192.168.137.2, FULL_DUPLEX, 100Mbps
20:28:45.896 -> Teszt történt.
20:28:47.885 -> Connecting to SMTP server...
20:28:47.885 -> [DEBUG] - INFO: connecting to SMTP server...
20:28:47.885 -> [DEBUG] - smtp.gmail.com
20:28:47.931 -> [DEBUG] - 587
20:28:47.931 -> [DEBUG] - INFO: starting socket
20:28:47.931 -> [V][ssl_client32.cpp:81] start_ssl_client(): Free internal heap before TLS 237632
20:28:47.931 -> [V][ssl_client32.cpp:83] start_ssl_client(): Starting socket
20:28:47.978 -> [DEBUG] - INFO: connecting to Server...
20:28:48.025 -> [DEBUG] - INFO: server connected
20:28:48.025 -> [DEBUG] - INFO: begin STARTTLS handshake
20:28:49.035 -> [DEBUG] - 220 smtp.gmail.com ESMTP m17sm6109964wrw.3 - gsmtp
20:28:49.035 -> ⸮l⸮⸮⸮Kt⸮⸮⸮nn
20:28:49.035 -> [DEBUG] - INFO: send SMTP command extended HELO
20:28:50.005 -> [DEBUG] - 250-smtp.gmail.com at your service, [5.63.207.1]
20:28:50.005 -> 250-SIZE 35882577
20:28:50.068 -> 250-8BITMIME
20:28:50.068 -> 250-STARTTLS
20:28:50.068 -> 250-ENHANCEDSTATUSCODES
20:28:50.068 -> 250-PIPELINING
20:28:50.068 -> 250-CHUNKING
20:28:50.068 -> 250 SMTPUTF8
20:28:50.068 -> 
20:28:50.068 -> [DEBUG] - INFO: send STARTTLS protocol command
20:28:51.012 -> [DEBUG] - 220 2.0.0 Ready to start TLS
20:28:51.046 -> 
20:28:51.046 -> [DEBUG] - INFO: seeding the random number generator
20:28:51.046 -> [V][ssl_client32.cpp:152] start_ssl_client(): Seeding the random number generator
20:28:51.046 -> [DEBUG] - INFO: setting up the SSL/TLS structure...
20:28:51.046 -> [V][ssl_client32.cpp:176] start_ssl_client(): Setting up the SSL/TLS structure...
20:28:51.046 -> [I][ssl_client32.cpp:287] start_ssl_client(): WARNING: Use certificates for a more secure communication!
20:28:51.081 -> [V][ssl_client32.cpp:334] start_ssl_client(): Setting hostname for TLS session...
20:28:51.081 -> [DEBUG] - INFO: performing the SSL/TLS handshake...
20:28:51.081 -> [V][ssl_client32.cpp:375] start_ssl_client(): Performing the SSL/TLS handshake...
20:28:52.053 -> [DEBUG] - INFO: verifying peer X.509 certificate...
20:28:52.053 -> [V][ssl_client32.cpp:415] start_ssl_client(): Verifying peer X.509 certificate...
20:28:52.053 -> [V][ssl_client32.cpp:431] start_ssl_client(): Certificate verified.
20:28:52.053 -> [V][ssl_client32.cpp:449] start_ssl_client(): Free internal heap after TLS 196604
20:28:52.053 -> [DEBUG] - INFO: smtp server connected
20:28:52.053 -> Identification...
20:28:52.053 -> [DEBUG] - INFO: send smtp HELO command
20:28:52.053 -> [V][ssl_client32.cpp:502] send_ssl_data(): Writing HTTP request...
20:28:52.053 -> [V][ssl_client32.cpp:502] send_ssl_data(): Writing HTTP request...
20:28:52.053 -> Error, identification failed
20:28:52.100 -> [DEBUG] - ERROR: identification failed
20:28:52.100 -> [DEBUG] - INFO: cleaning SSL connection
20:28:52.100 -> [V][ssl_client32.cpp:458] stop_ssl_socket(): Cleaning SSL connection.
20:28:52.100 -> Error sending Email, identification failed
```

PDE-Paul, 2020-03-25:
Hi All,

Has any resolution be found for this? As I have the same problem. WiFi works perfectly, but Ethernet via the LAN7810A does not work.

Thanking you in advance.

Paul

atlantisguru, 2020-03-25:
Hi Paul,

your ethernet completely not works or just the SMTP connection fails at some point?
As i experienced, i've got almost nothing response about ethernet related problems from Expressif forum nor the manufacturer's forum for weeks. Stackoverflow is the same.

Attila

PDE-Paul, 2020-03-25:
Hi,

My Ethernet connection works as I have an Azure connection in place, I need to implement a fail over so if my application cannot access Azure, I need to email via Ethernet, the WiFi route works as I want but WiFi is also a fail over, Ethernet will always be my primary comms route as my hardware is PoE powered. So I need to email via Ethernet, I get authentication passing but the process falls over with the response from Gmail. Tested same code with Ethernet disconnected, WiFi connects and sends email perfectly.

Cheers
Paul

atlantisguru, 2020-03-25:
Somebody suggested me thats maybe something MTU related problem (maybe the ethernet can't handle the response package size from the SMTP server), but the esp32 ethernet library is so primitive there is no way to make soft changes and some parts of it still depends on WiFi library. It seems to me nobody works on it right now.

PDE-Paul, 2020-03-25:
Well I just got this working...over Ethernet and Wifi, 

https://github.com/AIWIndustries/ESP32_Gmail_Sender

You just need to encode your logon details with this site https://www.base64encode.org/

Hope this helps...

Regards
Paul

atlantisguru, 2020-03-26:
Thank you Paul,

unfortunately it's another dead end for me. I've got SMTP Auth error constantly right after "Cannot Decode response" and "Unrecognized command".

```
20:21:48.725 -> 501 5.5.2 Cannot Decode response o9sm4960176wrx.48 - gsmtp
20:21:48.725 -> [V][ssl_client.cpp:279] send_ssl_data(): Writing HTTP request...
20:21:48.725 -> [V][ssl_client.cpp:279] send_ssl_data(): Writing HTTP request...
20:21:48.762 -> 502 5.5.1 Unrecognized command. o9sm4960176wrx.48 - gsmtp
20:21:48.762 -> [V][ssl_client.cpp:248] stop_ssl_socket(): Cleaning SSL connection.
20:21:48.795 -> Error sending message: SMTP AUTH error
```

atlantisguru, 2020-03-26:
Thanks Paul!

Finally it works with Ethernet! :)

closed 2020-03-26 by atlantisguru