espressif/arduino-esp32 #3815: esp_now_mod_peer() crash-bug

WereCatf, 2020-03-13:
As of current (13. March, 2020) git-version of Arduino-esp32 and with multiple different ESP32-boards, esp_now_mod_peer() causes a hard crash, if you first add a peer with encryption set as disabled, but then try to modify the aforementioned peer with encryption enabled.

Simplistic example below:
```
#include <esp_now.h>
#include <WiFi.h>

esp_now_peer_info_t nowHost;

void setup() {
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  esp_now_init();
  memset((void *) &nowHost, 0, sizeof(esp_now_peer_info_t));
  nowHost.encrypt = 0; //Set this to 1 and the following esp_now_mod_peer won't crash!
  esp_now_add_peer(&nowHost);

  nowHost.encrypt = 1;
  esp_now_mod_peer(&nowHost); //CRASH
  while(true) delay(1000); //Never reaches this
}

void loop() {
}
```

lbernstone, 2020-03-14:
Please follow the [issue template](https://github.com/espressif/arduino-esp32/blob/master/docs/ISSUE_TEMPLATE.md)

lbernstone, 2020-03-14:
What makes you think the original add_peer is working?  You are giving it an empty structure.  See the example at https://github.com/espressif/esp-idf/blob/master/examples/wifi/espnow/main/espnow_example_main.c#L302-L315.  They always check to see if the structure is empty before doing anything with it...

WereCatf, 2020-03-15:
@lbernstone  No, they aren't. They are using malloc to allocate the memory for the structure and then checking if the pointer to the structure is NULL, ie. if the malloc succeeded. In the above code, I am not using a malloc/free and am just allocating the structure from the get-go.

closed 2020-05-28 by stale[bot]