espressif/arduino-esp32 #3816: AsyncUDPServer receive message has a very long time delay

Peter-haotian-cui, 2020-03-13:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			11/jul/2017
IDE name:							Platform.io
Flash Frequency:					         40Mhz
PSRAM enabled:                                                no
Upload Speed:						115200
Computer OS:                                                Mac OSX

### Description:
I am trying to use two ESP32 communicated with each other, one is sending a UDP packet to another. But I observer a very long time delay from the server. And the delay is very unstable, it is about 10ms to several hundred ms. I want to build a remote controller using esp32 wifi, this delay is unacceptable. 

Two ESP32, one as an AP is the client, the other is a server with STA mode. The client send message to the server.

I use GPIO2 to output edge signal and oscilloscope to measure the delay between two devices. 

While, if I connect my macbook to the client AP and let the client send UDP message to the a python UDP receive program running on OSX, the delay seems normal, it is about 1~3ms. So I guess the client is normal.
[image]
[image]
[image]
[image]

### Client.cpp
```cpp

//Change the code below by your sketch
#include <WiFi.h>
#include <WiFiUdp.h>
#include <WiFiAP.h>
#include <Arduino.h>

const char *ssid = "ESP32";
const char *password = "password";

//IP address to send UDP data to:
// either use the ip address of the server or 
// a network broadcast address
// const char * udpAddress = "192.168.4.2";
const char * udpAddress = "192.168.4.2";
const int udpPort = 1234;

//Are we currently connected?
uint8_t connected = false;

//The udp library class
WiFiUDP udp;

//wifi event handler
void WiFiEvent(WiFiEvent_t event){
    switch(event) {
      case SYSTEM_EVENT_STA_GOT_IP:
          //When connected set 
          Serial.print("WiFi connected! IP address: ");
          Serial.println(WiFi.localIP());  
          //initializes the UDP state
          //This initializes the transfer buffer
          udp.begin(WiFi.localIP(),udpPort);
          connected = true;
          break;
      case SYSTEM_EVENT_STA_DISCONNECTED:
          Serial.println("WiFi lost connection");
          connected = false;
          break;
      default: break;
    }
}

void connectToWiFi(const char * ssid, const char * pwd){
  Serial.println("Connecting to WiFi network: " + String(ssid));

  // delete old config
  WiFi.disconnect(true);
  //register event handler
  WiFi.onEvent(WiFiEvent);
  
  //Initiate connection
  WiFi.begin(ssid, pwd);

  Serial.println("Waiting for WIFI connection...");
}

void setup()
{
    Serial.begin(115200);
    pinMode(2, OUTPUT);
    //Connect to the WiFi network
    // connectToWiFi(ssid, password);

    Serial.println();
    Serial.println("Configuring access point...");
    // You can remove the password parameter if you want the AP to be open.
    WiFi.softAP(ssid, password);
    IPAddress myIP = WiFi.softAPIP();
    Serial.print("AP IP address: ");
    Serial.println(myIP);

    // connected = true;

}

const uint8_t message[5] = {0x31,0x32,0x33,0x34,0x35};

void loop()
{
    delay(10);
    
    connected = Serial.read();
    //only send data when connected
    if(connected == 0x30){
        digitalWrite(2, HIGH);
        //Send a packet
        udp.beginPacket(udpAddress,udpPort);
        udp.printf("Seconds since boot: %lu", millis());
        udp.endPacket();
        digitalWrite(2, LOW);
    }

}
```
### Server.cpp
```cpp
#include "WiFi.h"
#include "AsyncUDP.h"

const char *ssid = "ESP32";
const char *password = "password";

AsyncUDP udp;

unsigned long time_1, time_2, i = 0;

void onPacketCallBack(AsyncUDPPacket packet)
{
    

    digitalWrite(2, HIGH);
    if(i == 0) {
      time_1 = millis();
      i++;
      Serial.printf("intervel is %d ms\n",time_1 - time_2);
    }else if(i == 1) {
      time_2 = millis();
      i = 0;
      Serial.printf("intervel is %d ms\n",time_2 - time_1);
    }

    // Serial.print("UDP Packet Type: ");
    // Serial.print(packet.isBroadcast()?"Broadcast":packet.isMulticast()?"Multicast":"Unicast");
    // Serial.print(", From: ");
    // Serial.print(packet.remoteIP());
    // Serial.print(":");
    // Serial.print(packet.remotePort());
    // Serial.print(", To: ");
    // Serial.print(packet.localIP());
    // Serial.print(":");
    // Serial.print(packet.localPort());
    // Serial.print(", Length: ");
    // Serial.print(packet.length());
    // Serial.print(", Data: ");
    Serial.write(packet.data(), packet.length());
    packet.flush();
    Serial.println();
    delay(1);
    digitalWrite(2, LOW);
    //reply to the client
    // packet.printf("Got %u bytes of data", packet.length());

}

void setup()
{
    Serial.begin(115200);
    pinMode(2, OUTPUT);
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid, password);
    if (WiFi.waitForConnectResult() != WL_CONNECTED) {
        Serial.println("WiFi Failed");
        while(1) {
            delay(1000);
        }
    }
    if(udp.listen(1234)) {
        Serial.print("UDP Listening on IP: ");
        Serial.println(WiFi.localIP());
        udp.onPacket(onPacketCallBack);
    }
}

void loop()
{
    delay(1000);
    //Send broadcast
    // udp.broadcast("Anyone here?");
}
```

lbernstone, 2020-03-13:
Does the lag decrease if you comment out all those serial prints? (serial is pretty slow response).
Is there a router in between the devices?  Try using the server as an AP and see if that improves the transaction time.

Peter-haotian-cui, 2020-03-14:
I try to comment out all the print, the lag is still there. I use the client as the AP and let the server connect to the client AP, so there is no router between them, just two ESP32. That is odd.

lbernstone, 2020-03-14:
The only other thing I can recommend is to ensure that WiFi sleep is set off
`WiFi.setSleep(false);`

Peter-haotian-cui, 2020-03-14:
It really WORKS !!! The delay is down to 400uS to 5ms, average is about 1ms.
[image]
[image]

Peter-haotian-cui, 2020-03-14:
While, there is another little problem, is the server first connect to the client and I set the client transfer frequency from 10Hz to 100Hz, I got five packets bundled together. Just like before.  But if I set down to 1Hz and then set to 100Hz, all back to normal. After that, no matter how I change the transmit frequency,  received packets are normal. 
[image]

closed 2020-05-27 by stale[bot]