espressif/arduino-esp32 #3817: LAN8720 does not appear to be restarting (Need to recycle the power supply)

rtek1000, 2020-03-14:
I managed to get the LAN8720 board to work with STM32F407VG using STM32CubeIDE and Arduino IDE, but I'm not having complete success with ESP32.

I am using this wire connection sequence:

```
GPIO17 - PHY_POWER   : NC - Osc. Enable - 4k7 Pulldown
GPIO22 - EMAC_TXD1   : TX1
GPIO19 - EMAC_TXD0   : TX0
GPIO21 - EMAC_TX_EN  : TX_EN
GPIO26 - EMAC_RXD1   : RX1
GPIO25 - EMAC_RXD0   : RX0
GPIO27 - EMAC_RX_DV  : CRS
GPIO00 - EMAC_TX_CLK : nINT/REFCLK (50MHz) - 4k7 Pullup
GPIO23 - SMI_MDC     : MDC
GPIO18 - SMI_MDIO    : MDIO
GND                  : GND
3V3                  : VCC
```
Source: https://sautter.com/blog/ethernet-on-esp32-using-lan8720/

And I set up the sample sketch for these settings:
```C++
/*
    This sketch shows the Ethernet event usage

*/

#include <ETH.h>

#ifdef ETH_CLK_MODE
#undef ETH_CLK_MODE
#endif
#define ETH_CLK_MODE    ETH_CLOCK_GPIO0_IN

// Pin# of the enable signal for the external crystal oscillator (-1 to disable for internal APLL source)
#define ETH_POWER_PIN   17

// Type of the Ethernet PHY (LAN8720 or TLK110)
#define ETH_TYPE        ETH_PHY_LAN8720

// I²C-address of Ethernet PHY (0 or 1 for LAN8720, 31 for TLK110)
#define ETH_ADDR        1

// Pin# of the I²C clock signal for the Ethernet PHY
#define ETH_MDC_PIN     23

// Pin# of the I²C IO signal for the Ethernet PHY
#define ETH_MDIO_PIN    18

static bool eth_connected = false;

void WiFiEvent(WiFiEvent_t event)
{
  switch (event) {
    case SYSTEM_EVENT_ETH_START:
      Serial.println("ETH Started");
      //set eth hostname here
      ETH.setHostname("esp32-ethernet");
      break;
    case SYSTEM_EVENT_ETH_CONNECTED:
      Serial.println("ETH Connected");
      break;
    case SYSTEM_EVENT_ETH_GOT_IP:
      Serial.print("ETH MAC: ");
      Serial.print(ETH.macAddress());
      Serial.print(", IPv4: ");
      Serial.print(ETH.localIP());
      if (ETH.fullDuplex()) {
        Serial.print(", FULL_DUPLEX");
      }
      Serial.print(", ");
      Serial.print(ETH.linkSpeed());
      Serial.println("Mbps");
      eth_connected = true;
      break;
    case SYSTEM_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");
      eth_connected = false;
      break;
    case SYSTEM_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");
      eth_connected = false;
      break;
    default:
      break;
  }
}

void testClient(const char * host, uint16_t port)
{
  Serial.print("\nconnecting to ");
  Serial.println(host);

  WiFiClient client;
  if (!client.connect(host, port)) {
    Serial.println("connection failed");
    return;
  }
  client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
  while (client.connected() && !client.available());
  while (client.available()) {
    Serial.write(client.read());
  }

  Serial.println("closing connection\n");
  client.stop();
}

void setup()
{
//  pinMode(ETH_POWER_PIN, OUTPUT);
//  digitalWrite(ETH_POWER_PIN, HIGH);
  Serial.begin(115200);
  WiFi.onEvent(WiFiEvent);
  ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE);
}

void loop()
{
  if (eth_connected) {
    testClient("google.com", 80);
  }
  delay(10000);
}
```

But whenever I press the reset button, or after the IDE finishes uploading the sketch, only those outputs appear and appear to be stuck:
```
02:02:07.939 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
02:02:07.939 -> configsip: 0, SPIWP:0xee
02:02:07.939 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
02:02:07.939 -> mode:DIO, clock div:1
02:02:07.939 -> load:0x3fff0018,len:4
02:02:07.939 -> load:0x3fff001c,len:1216
02:02:07.939 -> ho 0 tail 12 room 4
02:02:07.939 -> load:0x40078000,len:9720
02:02:07.939 -> ho 0 tail 12 room 4
02:02:07.939 -> load:0x40080400,len:6352
02:02:07.939 -> entry 0x400806b8
02:02:08.220 -> ETH Started

```

For STM32, when I pressed the reset button, the LEDs flashed (from the RJ45 connector on the LAN8720). But when pressing the reset of the ESP32, the same behavior does not occur with the LEDs.

So I turned off the power, removed the USB cable, and then reconnected the cable. After turning the power back on, the LAN8720 card worked, and a response was received from the internet.

In this way I conclude that there is no reset of the LAN8720.

How can I solve this?

Thank you.

rtek1000, 2020-03-14:
Note: the problem of a lack of RESET when starting ESP32 for LAN8720 also occurs here: https://github.com/espressif/arduino-esp32/issues/3818

rtek1000, 2020-03-14:
Note: the problem of a lack of RESET when starting ESP32 for LAN8720 also occurs here: https://github.com/espressif/arduino-esp32/issues/3819

rtek1000, 2020-03-14:
Note: the problem of a lack of RESET when starting ESP32 for LAN8720 also occurs here: https://github.com/espressif/arduino-esp32/issues/3821

rtek1000, 2020-03-15:
I managed to make ESP32 work in a similar way to STM32, when ESP32 is reset.

Looking at the LAN8720 datasheet, we have the following:

> 3.8.5.1Hardware Reset
> ...During a Hardware reset, an external clock must be supplied to the XTAL1/CLKIN signal.

The next part that says about Software Reset does not inform about the need for the clock to be working, but that must be what is happening. When the GPIO17 pin is placed in a low logic level, the clock is stopped and soon afterwards the ESP32 must be trying to reset the LAN8720.

To solve this, it was enough to disable the use of the GPIO17 pin via library initialization (ETH_POWER_PIN), and to perform the high logic level on the GPIO17 pin. So that it is possible to start ESP32 with GPIO0 at a high logic level, which can only be achieved by stopping the clock.

Before:
```C++
#define ETH_POWER_PIN  17
```
After:
```C++
#define ETH_POWER_PIN  -1
```

And the rise of the GPIO17 pin:

```C++
#define ETH_POWER_PIN_ALTERNATIVE 17

void setup() {
  pinMode(ETH_POWER_PIN_ALTERNATIVE, OUTPUT);
  digitalWrite(ETH_POWER_PIN_ALTERNATIVE, HIGH);

  ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE); // Enable ETH
```

rtek1000, 2020-03-15:
Modified code (Working):

```C++
/*
    This sketch shows the Ethernet event usage

*/

#include <ETH.h>

#ifdef ETH_CLK_MODE
#undef ETH_CLK_MODE
#endif
#define ETH_CLK_MODE    ETH_CLOCK_GPIO0_IN

// Pin# of the enable signal for the external crystal oscillator (-1 to disable for internal APLL source)
#define ETH_POWER_PIN   -1 // Do not use it, it can cause conflict during the software reset.
#define ETH_POWER_PIN_ALTERNATIVE 17

// Type of the Ethernet PHY (LAN8720 or TLK110)
#define ETH_TYPE        ETH_PHY_LAN8720

// I²C-address of Ethernet PHY (0 or 1 for LAN8720, 31 for TLK110)
#define ETH_ADDR        1

// Pin# of the I²C clock signal for the Ethernet PHY
#define ETH_MDC_PIN     23

// Pin# of the I²C IO signal for the Ethernet PHY
#define ETH_MDIO_PIN    18

static bool eth_connected = false;

void WiFiEvent(WiFiEvent_t event)
{
  switch (event) {
    case SYSTEM_EVENT_ETH_START:
      Serial.println("ETH Started");
      //set eth hostname here
      ETH.setHostname("esp32-ethernet");
      break;
    case SYSTEM_EVENT_ETH_CONNECTED:
      Serial.println("ETH Connected");
      break;
    case SYSTEM_EVENT_ETH_GOT_IP:
      Serial.print("ETH MAC: ");
      Serial.print(ETH.macAddress());
      Serial.print(", IPv4: ");
      Serial.print(ETH.localIP());
      if (ETH.fullDuplex()) {
        Serial.print(", FULL_DUPLEX");
      }
      Serial.print(", ");
      Serial.print(ETH.linkSpeed());
      Serial.println("Mbps");
      eth_connected = true;
      break;
    case SYSTEM_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");
      eth_connected = false;
      break;
    case SYSTEM_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");
      eth_connected = false;
      break;
    default:
      break;
  }
}

void testClient(const char * host, uint16_t port)
{
  Serial.print("\nconnecting to ");
  Serial.println(host);

  WiFiClient client;
  if (!client.connect(host, port)) {
    Serial.println("connection failed");
    return;
  }
  client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
  while (client.connected() && !client.available());
  while (client.available()) {
    Serial.write(client.read());
  }

  Serial.println("closing connection\n");
  client.stop();
}

void setup()
{
  pinMode(ETH_POWER_PIN_ALTERNATIVE, OUTPUT);
  digitalWrite(ETH_POWER_PIN_ALTERNATIVE, HIGH);
  Serial.begin(115200);
  WiFi.onEvent(WiFiEvent);
  ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE);
}

void loop()
{
  if (eth_connected) {
    testClient("google.com", 80);
  }
  delay(10000);
}
```

closed 2020-05-28 by stale[bot]

AH-007, 2022-01-21:
For love and money can't get the lights on the Ethernet Port to come on.... any help anyone?

rtek1000, 2022-01-21:
https://m.youtube.com/watch?v=zJS2dj4mQxg

rtek1000, 2022-01-21:
What schematic diagram are you using?

[image]

https://media.digikey.com/pdf/Data%2520Sheets/SMSC/LAN8720.pdf