espressif/arduino-esp32 #3819: [LAN8720] Please add this example of TCP server connection

rtek1000, 2020-03-14:
```C++
/*
    Copyright statement: This article is the original article of the CSDN blogger "Naisu_kun",
    which follows the CC 4.0 BY-SA copyright agreement.
    For reprinting, please attach the original source link and this statement.

    Original link：https://blog.csdn.net/Naisu_kun/article/details/87125845
 */

#include <ETH.h> // quote to use ETH
#include <WiFi.h>

#define ETH_ADDR        1
#define ETH_POWER_PIN  17
#define ETH_MDC_PIN    23
#define ETH_MDIO_PIN   18
#define ETH_TYPE       ETH_PHY_LAN8720
#define ETH_CLK_MODE    ETH_CLOCK_GPIO0_IN

WiFiServer server; // Declare server objects

void setup() {
  delay(5000);
  Serial.begin(115200);
  Serial.println();

  ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE); // Enable ETH
  while(!((uint32_t)ETH.localIP())) // Waiting for IP
  {

  }
  Serial.println("Connected");
  Serial.print("IP Address:");
  Serial.println(ETH.localIP());

  server.begin(8888); // Server starts listening on port number 8888
}

void loop() {
    WiFiClient client = server.available(); // Try to create a client object
    if (client) // If current customer is available
    {
        Serial.println("[Client connected]");
        String readBuff;
        while (client.connected()) // If the client is connected
        {
            if (client.available() > 0) // If there is readable data
            {
                char c = client.read(); // Read a byte
                                        // You can also use readLine() and other methods

                client.write(c); // Send to client
                Serial.write(c); // Printing from the serial port
                
//                readBuff += c;
//                if(c == '\r') // Received carriage return
//                {
//                    client.print("Received: " + readBuff); // Send to client
//                    Serial.println("Received: " + readBuff); // Printing from the serial port
//                    readBuff = "";
//                    break; // Out of the loop
//                }
            }
        }
        client.stop(); //End the current connection
        Serial.println("[Client disconnected]");
    }
}
```

Serial monitor output:
> 04:33:31.936 -> 
> 04:33:36.046 -> Connected
> 04:33:36.046 -> IP Address:192.168.1.106
> 04:33:44.036 -> [Client connected]
> 04:33:46.436 -> MMMMMMMMMMMMMMMMMMM
> 04:33:51.436 -> 
> 04:33:52.156 -> 
> 04:33:52.396 -> 
> 04:33:52.636 -> 0000
> 04:33:53.676 -> 0000
> 04:33:53.876 -> 0000
> 04:33:54.116 -> 0000
> 04:33:54.316 -> 0000
> 04:33:54.516 -> 0000
> 04:33:54.716 -> 0000
> 04:33:55.036 -> 0000

[image]

(Note: the problem of a lack of RESET when starting ESP32 for LAN8720 also occurs here)

rtek1000, 2020-03-14:
Explanation of the code and other parameters in the link (already mentioned in the code above):
https://blog.csdn.net/Naisu_kun/article/details/87125845

Original language: Chinese (You can use the Google translator to read English, or another language)

rtek1000, 2020-03-14:
Static IP example:

```C++
/*
    Copyright statement: This article is the original article of the CSDN blogger "Naisu_kun",
    which follows the CC 4.0 BY-SA copyright agreement.
    For reprinting, please attach the original source link and this statement.

    Original link：https://blog.csdn.net/Naisu_kun/article/details/87125845
    - Explanation of the code and other parameters: see the link above.
    - Note: Original language: Chinese (You can use the Google translator to read English, or another language)
 */

#include <ETH.h> // quote to use ETH
#include <WiFi.h>

#define ETH_ADDR        1
#define ETH_POWER_PIN  17
#define ETH_MDC_PIN    23
#define ETH_MDIO_PIN   18
#define ETH_TYPE       ETH_PHY_LAN8720
#define ETH_CLK_MODE    ETH_CLOCK_GPIO0_IN

WiFiServer server; // Declare server objects

IPAddress local_ip(192,168,1,112);
IPAddress gateway(192,168,1,1);
IPAddress subnet(255,255,255,0);
IPAddress dns1(8,8,8,8);
IPAddress dns2 = (uint32_t)0x00000000;

static bool eth_connected = false;

void WiFiEvent(WiFiEvent_t event)
{
  switch (event) {
    case SYSTEM_EVENT_ETH_START:
      Serial.println("ETH Started");
      //set eth hostname here
      ETH.setHostname("esp32-ethernet");
      break;
    case SYSTEM_EVENT_ETH_CONNECTED:
      Serial.println("ETH Connected");
      break;
    case SYSTEM_EVENT_ETH_GOT_IP:
      Serial.print("ETH MAC: ");
      Serial.print(ETH.macAddress());
      Serial.print(", IPv4: ");
      Serial.print(ETH.localIP());
      if (ETH.fullDuplex()) {
        Serial.print(", FULL_DUPLEX");
      }
      Serial.print(", ");
      Serial.print(ETH.linkSpeed());
      Serial.println("Mbps");
      eth_connected = true;
      break;
    case SYSTEM_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");
      eth_connected = false;
      break;
    case SYSTEM_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");
      eth_connected = false;
      break;
    default:
      break;
  }
}

void setup() {
  delay(5000);
  Serial.begin(115200);
  Serial.println();

  WiFi.onEvent(WiFiEvent);

  ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE); // Enable ETH
  
  ETH.config(local_ip, gateway, subnet, dns1, dns2); // Static IP, leave without this line to get IP via DHCP
    
//  while(!((uint32_t)ETH.localIP())) // Waiting for IP (leave this line group to get IP via DHCP)
//  {
//
//  }
  
//  Serial.println("Connected");
//  Serial.print("IP Address:");
//  Serial.println(ETH.localIP());

  server.begin(8888); // Server starts listening on port number 8888
}

void loop() {
    WiFiClient client = server.available(); // Try to create a client object
    if (client) // If current customer is available
    {
        Serial.println("[Client connected]");
        String readBuff;
        while (client.connected()) // If the client is connected
        {
            if (client.available() > 0) // If there is readable data
            {
                char c = client.read(); // Read a byte
                                        // You can also use readLine() and other methods

                client.write(c); // Send to client
                Serial.write(c); // Printing from the serial port
                
//                readBuff += c;
//                if(c == '\r') // Received carriage return
//                {
//                    client.print("Received: " + readBuff); // Send to client
//                    Serial.println("Received: " + readBuff); // Printing from the serial port
//                    readBuff = "";
//                    break; // Out of the loop
//                }
            }
        }
        client.stop(); //End the current connection
        Serial.println("[Client disconnected]");
    }
}
```

rtek1000, 2020-03-15:
Updated code to avoid conflict at startup. See: #3817

closed 2020-05-28 by stale[bot]

ahmadreza-hadidi, 2021-08-01:
Does [AsyncTCP](https://github.com/me-no-dev/AsyncTCP) library work with ETH interface (instead of using WiFiServer)?

wesee-dev, 2021-09-13:
@ahmadreza-hadidi Yes it does :)

ttuzz, 2023-01-01:
tnx for example i love it