espressif/arduino-esp32 #3820: retrieveCharacteristics(): >> getCharacteristics() exception

gereblye79, 2020-03-14:
When I try to get the characteristics of my BLE device I always get the below error message. The source was working 3 days ago, but now it causes crash during the execution on the device. I could not do too much with the EspExceptionDecoder. Maybe my fault.

I get this message:
`Stack smashing protect failure!
abort() was called at PC 0x400dea3c on core 1

Backtrace: 0x40091f40:0x3ffceb00 0x40092171:0x3ffceb20 0x400dea3c:0x3ffceb40 0x400d5ab3:0x3ffceb60 0x400d5ada:0x3ffcec20 0x400d8fcb:0xe4893916`

If I set the debug core level I see the following exception:
`[D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:436] getServices(): << getServices
[D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:396] getService(): << getService: found the service with uuid: 47e9ee00-47e9-11e4-8939-164230d1df67
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLERemoteService.cpp:168] retrieveCharacteristics(): >> getCharacteristics() for service: 47e9ee00-47e9-11e4-8939-164230d1df67
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400812d0  PS      : 0x00060c30  A0      : 0x800d7159  A1      : 0x3ffcec60  
A2      : 0xe911e489  A3      : 0x00000000  A4      : 0x00000019  A5      : 0x00000048  
A6      : 0x3ffcecd2  A7      : 0x3ffcecbc  A8      : 0xe911e489  A9      : 0x3ffcec20  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x00000000  
A14     : 0x3ffcecbc  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xe911e489  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  

Backtrace: 0x400812d0:0x3ffcec60 0x400d7156:0x3ffcec80 0x400d74fa:0x3ffced40 0x400db63b:0xe4893916`

sketch part:
`int bluRetVal = device->connect("CometBlue", 5234);`

library part (this is where the exception was thrown I think:
'uint8_t CometBlue::connect(std::string deviceName, uint32_t pin) {
    // check if connected
    if (_connected) {
        return 0;
    }

    // connect to device
    BLEDevice::init(deviceName);
	BLEDevice::setPower(ESP_PWR_LVL_P4);
    _client = BLEDevice::createClient();
    if (!_client->connect(_deviceAddress)) {
        return 1;
    }

    // check service uuid
    _remoteService = _client->getService(BLEUUID(SERVICE_UUID));
    if (_remoteService == nullptr) {
        _client->disconnect();
        return 2;
    }

    // authenticate
    BLERemoteCharacteristic* pRemotePin = _remoteService->getCharacteristic(BLEUUID(PIN_UUID));
    if (pRemotePin == nullptr) {
        _client->disconnect();
        return 3;
    }
    uint8_t pinencoded[4] = {(uint8_t)(pin & 0xFF), (uint8_t)((pin >> 8) & 0xFF), (uint8_t)((pin >> 16) & 0xFF), (uint8_t)((pin >> 24) & 0xFF)};
    pRemotePin->writeValue(pinencoded, sizeof(pinencoded), true);

    // verify if successful
    std::string value = _remoteService->getValue(BLEUUID(BAT_UUID));
    if (value.length() == 0) {
        _client->disconnect();
        return 4;
    }

    _connected = true;
    return 0;
}'

chegewara, 2020-03-14:
> Stack smashing protect failure!

Too small task stack.

gereblye79, 2020-03-14:
Thanks! How can I increase the task stack?

closed 2020-05-27 by stale[bot]

covert-, 2022-07-22:
To help others. I resolved this issue by using the ArduinoBLE lib instead of the ESP32_BLE one I was using.