espressif/arduino-esp32 #3822: Native resolving .local domain

Cicicok, 2020-03-14:
ESP8266 Arduino library supports native resolving of local domains like `serve.local`. But why ESP32 don't?
This one works perfectly on ESP8266:
```
client.connect(serve.local, 80);
```
but on ESP32 produces error `hostByName(): DNS Failed for server.local`

Where is the difference? Which part of code or library is here missing?

lbernstone, 2020-03-14:
If you have a local DNS server with avahi/Bonjour (mDNS) capabilities, it will resolve just fine.  I am able to resolve .local mDNS addresses on my esp32.

asier70, 2020-03-23:
This probably depends on LWIP library. Arduino ESP32 library uses very old version, ESP8266 library inludes newer version.

Cicicok, 2020-03-23:
That could be it. esp8266 community uses the latest 2.1.2 version https://github.com/d-a-v/esp82xx-nonos-linklayer, but here we're stuck with 2.0.3
https://github.com/espressif/arduino-esp32/blob/0f772270fbb32099cad72053755d6b4c7e8e1f82/tools/sdk/include/lwip/lwip/init.h#L52-L57

closed 2020-06-05 by stale[bot]

MikaAK, 2020-06-14:
This stale bot seems like a real kicker, stale issues just get closed when they're actually valid 😮 What's the solution to this?

markg85, 2020-12-12:
Un-stale please.
I too have this issue and am looking for a solution. One that does allow me to keep using .local names and does not require me to setup a local dns server.

NorthernMan54, 2021-01-16:
Latest release candidate doesn't appear to resolve this

moimart1, 2021-01-16:
Any workaround at least ?

moimart1, 2021-01-16:
Finally, I found this https://www.tutorialfor.com/questions-324359.htm

```
  #include<ESPmDNS.h>

  mdns_init ();// Add
  IPAddress ipaddr = MDNS.queryHost("my-host-without-dot-local", 5000 /* ms */);
  if (ipaddr == INADDR_NONE)
  {
    // If not valid
    ipaddr.fromString("192.168.0.1");
  }
```

markg85, 2021-01-17:
> Finally, I found this https://www.tutorialfor.com/questions-324359.htm
> 
> ```c++
>   #include<ESPmDNS.h>
> 
>   mdns_init ();// Add
>   IPAddress ipaddr = MDNS.queryHost("my-host-without-dot-local", 5000 /* ms */);
>   if (ipaddr == INADDR_NONE)
>   {
>     // If not valid
>     ipaddr.fromString("192.168.0.1");
>   }
> ```

Hi!

While that is a nice trick, it might or might not work. It entirely depends on other factors in your home network (particularly your router) if it works.

In my case i have a pc that i can only ping by name if i ping with the explicit .local suffix.
If i leave that out then it won't resolve and won't ping.
This is a DNS thingy where it depends on your network settings if you can "ping both", in that case the above trick works.

I'm still searching for a way to make the .local thingy work too... I'll let it know if i find something.

markg85, 2021-02-21:
I took a deep dive into this issue and documented my findings here: https://github.com/espressif/esp-idf/issues/2507#issuecomment-761836300

The actual issue is that for whatever reason the ESP32 SDK blindly adds ".local" to mdns queries. That's downright wrong as ".local" is something a user can change (.local is the default) and doing this makes it a lot more difficult to lookup names.