espressif/arduino-esp32 #3825: first upload worked, second did not (Olimex ESP32-PoE)

karls0, 2020-03-15:
### Hardware:
Board:							Olimex ESP32-PoE
Core Installation version:			Core 4.2.1  Home 3.1.1
IDE name:						platform.io 
Flash Frequency:					default
PSRAM enabled:                                    don't know
Upload Speed:					tried default, now at 115200
Computer OS:                                         Linux Mint 19.3

### Description:
I recently bought an Olimex ESP32-PoE and an Olimex ESP32-EVB and uploaded a test-sketch to both. No problem with this.
Next day I wanted to try another sketch, but the upload failed with 
 "A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header"
From my experience with ESP8266 I checked the supply, but I measure 5.1V at the board. The USB-cable is the same as yesterday, but nevertheless I tried another and it did not help.

I also tried different versions of pressing RST- and BUT-button - no help.

Since I could upload a sketchd yesterday, I really do not understand what could be wrong. Thank you for any tips to help me.
Regards, Karl

atlantisguru, 2020-03-15:
Same happened to me today with an Olimex ESP32-EVB, but i'm using Windows and Arduino IDE.
Did you installed the driver of the board?
For me the solution was disconnect USB, restart OS, reconnect USB and start Arduino IDE again.

karls0, 2020-03-16:
Hi,
thank you for the quick reply and your tips.
First I installed the driver from the OLIMEX-repository, rebooted and tried again. 
Uploading to a  OLIMEX ESP32-PoE, Olimex ESP32-EVB and AZ-delivery ESP32 Dev did not work - uploading to a ESP8266 works. This makes me think, that it cannot be the serial-driver.

I then installed Arduino 1.8.12 and the ESP-boards too (following https://www.heise.de/ct/artikel/Arduino-IDE-installieren-und-fit-machen-fuer-ESP8266-und-ESP32-4130814.html ) with the same results. No uploading to ESP32-boards, but everything OK with a ESP8266.  

Since the probability that all 3 new ESP32-boards are defective is very low, I must have missed something basic :-((  I have also tried to use a lower upload speed of 115200 - same results. 
Any further tips?

TIA, Karl

karls0, 2020-03-16:
I don't think it is three defective boards. It is something trivial, like Linux gave new COM port number for the devices, or some other peripheral used and blocked the COM port, or the driver associated changed and it stopped toggling reset, maybe wrong board configuration was selected. What is the chip in the ESP8266 that works? Is it CH340T? If not - the comparison in behavior is inconclusive, since it probably uses different driver. If the ESP8266 board has CH340T then it is less likely to be a driver issue but still not completely out of question.

I think it is a popular problem so make sure to check for what others might have suggested online.

Also share the test program that you used. I might run it on our boards to compare.

closed 2020-03-16 by karls0

karls0, 2020-03-18:
I recently bought two ESP32-boards from OLIMEX and had problems uploading. It was very strange, 
DanKoloff from Olimex gave me some tips too, but it was a capacitor at EN that helped. 
See https://randomnerdtutorials.com/solved-failed-to-connect-to-esp32-timed-out-waiting-for-packet-header/#comment-438162
From curiosity I tested different caps, diffent resistors and the two Olimex-boards plus a AZ-delivery ESP32-Dev which I had bought the week before.
You can find the table of my tests at
https://github.com/OLIMEX/ESP32-POE/issues/11
Regards, Karl