espressif/arduino-esp32 #3829: esp32-cam flash led doesnt turn off

ligantx, 2020-03-17:
### Hardware:
Board:							ESP32-cam
Core Installation version:			1.04
IDE name:						Arduino IDE
Upload Speed:					115200
Computer OS:                                        Windows 10

```cpp

    ledcSetup(2, 5000, 8);
   //flashled = GPIO4, onboard led
    ledcAttachPin(flashLed, 1);
    ledcWrite(2, 50); //flashLed in bright
    delay(1000);
    ledcWrite(2,  0); //flashLed in still bright, it should turn off
    delay(1000);
    ledcWrite(2, 50);
    delay(1000);
    ledcWrite(2,  0);

```

ligantx, 2020-03-18:
made a little more research and found more interferences with esp32-cam.
I have identified problem when using 0 - 2 channels and camera at the same time. The onboard flashLed stays on forever. BUT, if i change to channel 15 everything works fine:
```
ledcSetup(15, 5000, 8);
ledcAttachPin(flashLed, 15);
ledcWrite(15, 150);
  
  Serial.println("capturing photo");
  fb = esp_camera_fb_get();
  if (!fb) {
    ESP_LOGE(TAG, "Camera Capture Failed");
    return;
  }
  ledcWrite(15, 0);

```

closed 2020-05-31 by stale[bot]

mbilsky, 2020-11-13:
I was having the exact same problem. Found out that we need to use timer 3 (14 &15) according to: https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-ledc.c