espressif/arduino-esp32 #3830: SMTP connection fails only with onboard Ethernet

atlantisguru, 2020-03-17:
Board: Olimex ESP32-EVB
Core Installation version: 1.0.4
IDE name: Arduino IDE
Flash Frequency: 80Mhz
PSRAM enabled: yes
Upload Speed: 115200
Computer OS: Windows 10

Description:
I would like to send email with only via Ethernet, but it doesn't works. SMTP connection fails on identification (not authentication) part.
With WiFi it works perfect. I don't want to use WiFi.
It's a mail service independent problem. Credentials are all ok.
I already discussed the problem with the creator of ESP32_MailClient library and that works perfect.

//Relevant parts of the code
#include <ETH.h>
#include "ESP32_MailClient.h"
...

void setup() {
...ethernet started/connected
...got IP from DHCP
}

void loop() {
...sending email triggered by an event
}

Debug Messages:
Connecting to SMTP server...
[DEBUG] - INFO: connecting to SMTP server...
[DEBUG] - smtp.gmail.com
[DEBUG] - 587
[DEBUG] - INFO: starting socket
[DEBUG] - INFO: connecting to Server...
[DEBUG] - INFO: server connected
[DEBUG] - INFO: begin STARTTLS handshake
[DEBUG] - 220 smtp.gmail.com ESMTP c85sm13831667wmd.48 - gsmtp
⸮!Pk�
[DEBUG] - INFO: send SMTP command extended HELO
[DEBUG] - 250-smtp.gmail.com at your service, [5.63.208.133]
250-SIZE 35882577
250-8BITMIME
250-STARTTLS
250-ENHANCEDSTATUSCODES
250-PIPELINING
250-CHUNKING
250 SMTPUTF8

[DEBUG] - INFO: send STARTTLS protocol command
[DEBUG] - 220 2.0.0 Ready to start TLS

[DEBUG] - INFO: seeding the random number generator
[DEBUG] - INFO: setting up the SSL/TLS structure...
[DEBUG] - INFO: setting hostname for TLS session...
[DEBUG] - INFO: performing the SSL/TLS handshake...
[DEBUG] - INFO: verifying peer X.509 certificate...
[DEBUG] - INFO: certificate verified
[DEBUG] - INFO: smtp server connected
Identification...
[DEBUG] - INFO: send smtp HELO command
Error, identification failed //<- With WiFi can pass trough, with Ethernet can't
[DEBUG] - ERROR: identification failed
[DEBUG] - INFO: cleaning SSL connection
Error sending Email, identification failed

Additional information:
With Wireshark i did catch the communication between SMTP server and ESP.
It seems when the "Identification..." comes, the board suddenly resetting the communication (Line ID 3066).

No. Time Source Destination Protocol Length Info
3033 1591.392626 192.168.137.2 108.177.15.108 TCP 58 52839 → 587 [SYN] Seq=0 Win=5744 Len=0 MSS=1436
3034 1591.451031 108.177.15.108 192.168.137.2 TCP 58 587 → 52839 [SYN, ACK] Seq=0 Ack=1 Win=62920 Len=0 MSS=1430
3035 1591.451720 192.168.137.2 108.177.15.108 TCP 54 52839 → 587 [ACK] Seq=1 Ack=1 Win=22976 Len=0
3036 1591.510841 108.177.15.108 192.168.137.2 SMTP 107 S: 220 smtp.gmail.com ESMTP n1sm40021609wrj.77 - gsmtp
3037 1591.510913 192.168.137.2 108.177.15.108 TCP 54 52839 → 587 [ACK] Seq=1 Ack=54 Win=22976 Len=0
3038 1591.920478 192.168.137.1 224.0.0.251 MDNS 102 Standard query response 0x0000 A, cache flush 192.168.137.1
3039 1592.452915 192.168.137.2 108.177.15.108 SMTP 65 C: EHLO DUDE
3040 1592.540947 108.177.15.108 192.168.137.2 TCP 54 587 → 52839 [ACK] Seq=54 Ack=12 Win=62920 Len=0
3041 1592.548404 108.177.15.108 192.168.137.2 SMTP 221 S: 250-smtp.gmail.com at your service, [5.63.214.69] | SIZE 35882577 | 8BITMIME | STARTTLS | ENHANCEDSTATUSCODES | PIPELINING | CHUNKING | SMTPUTF8
3042 1592.548522 192.168.137.2 108.177.15.108 TCP 54 52839 → 587 [ACK] Seq=12 Ack=221 Win=22976 Len=0
3043 1593.462134 192.168.137.2 108.177.15.108 SMTP 64 C: STARTTLS
3044 1593.520641 108.177.15.108 192.168.137.2 SMTP 84 S: 220 2.0.0 Ready to start TLS
3045 1593.520684 192.168.137.2 108.177.15.108 TCP 54 52839 → 587 [ACK] Seq=22 Ack=251 Win=22976 Len=0
3046 1594.475617 192.168.137.2 108.177.15.108 TLSv1 357 Client Hello
3047 1594.551024 108.177.15.108 192.168.137.2 SSL 152 [TCP Previous segment not captured] , Continuation Data
3048 1594.551408 108.177.15.108 192.168.137.2 TCP 1484 [TCP Out-Of-Order] 587 → 52839 [ACK] Seq=251 Ack=325 Win=63784 Len=1430
3049 1594.553589 192.168.137.2 108.177.15.108 TCP 54 [TCP Dup ACK 3045#1] 52839 → 587 [ACK] Seq=325 Ack=251 Win=22976 Len=0
3050 1594.553722 192.168.137.2 108.177.15.108 TCP 54 52839 → 587 [ACK] Seq=325 Ack=1681 Win=22976 Len=0
3051 1594.558306 108.177.15.108 192.168.137.2 TCP 1484 [TCP Out-Of-Order] 587 → 52839 [ACK] Seq=1681 Ack=325 Win=63784 Len=1430
3052 1594.558355 192.168.137.2 108.177.15.108 TCP 54 52839 → 587 [ACK] Seq=325 Ack=3209 Win=22976 Len=0
3053 1594.925986 192.168.137.1 224.0.0.251 MDNS 93 Standard query response 0x0000 A, cache flush 192.168.8.101
3054 1595.313210 192.168.137.2 108.177.15.108 TLSv1.2 129 Client Key Exchange
3055 1595.317151 192.168.137.2 108.177.15.108 TLSv1.2 60 Change Cipher Spec
3056 1595.318389 192.168.137.2 108.177.15.108 TLSv1.2 99 Encrypted Handshake Message
3057 1595.380724 108.177.15.108 192.168.137.2 TCP 54 587 → 52839 [ACK] Seq=3209 Ack=406 Win=63784 Len=0
3058 1595.388265 108.177.15.108 192.168.137.2 TLSv1.2 346 New Session Ticket, Change Cipher Spec, Encrypted Handshake Message
3059 1595.401198 192.168.137.2 108.177.15.108 TCP 54 52839 → 587 [ACK] Seq=451 Ack=3501 Win=22976 Len=0
3060 1595.416261 192.168.137.2 108.177.15.108 TLSv1.2 92 Application Data
3061 1595.416793 192.168.137.2 108.177.15.108 TLSv1.2 85 Application Data
3062 1595.422297 192.168.137.2 108.177.15.108 TCP 54 52839 → 587 [FIN, ACK] Seq=520 Ack=3501 Win=22976 Len=0
3063 1595.470711 108.177.15.108 192.168.137.2 TCP 54 587 → 52839 [ACK] Seq=3501 Ack=520 Win=63784 Len=0
3064 1595.478523 108.177.15.108 192.168.137.2 TLSv1.2 119 Application Data
3065 1595.478549 192.168.137.2 108.177.15.108 TCP 54 52839 → 587 [ACK] Seq=521 Ack=3566 Win=22976 Len=0
3066 1595.479195 192.168.137.2 108.177.15.108 TCP 54 52839 → 587 [RST, ACK] Seq=521 Ack=3566 Win=5744 Len=0
3067 1595.490816 108.177.15.108 192.168.137.2 TCP 54 587 → 52839 [FIN, ACK] Seq=3566 Ack=521 Win=63784 Len=0
3068 1595.492336 192.168.137.2 108.177.15.108 TCP 54 52839 → 587 [RST] Seq=521 Win=0 Len=0

atanisoft, 2020-03-17:
So because https://github.com/espressif/arduino-esp32/issues/3813 wasnt' acted upon you filed another?

atlantisguru, 2020-03-17:
> So because #3813 wasn't acted upon you filed another?

Exactly. Nobody write to tell me, my problem is totally impossible or not. Mail Client library creator force me to write here, Olimex force me to write here, StackOverflow force me to write here, but nobody wants to answer an ethernet related question here or Espressif forum for days (or weeks as i saw other ethernet related questions/issues).

atanisoft, 2020-03-17:
Filing a second issue won't bring more attention to the issue and nor will what you have posted so far on either of the issues. Please turn on core level debug output and post it in #3813 and please close this duplicate issue.

closed 2020-03-17 by atlantisguru