espressif/arduino-esp32 #3831: ESP32 + LAN8720 only detects network cable and does not show ip

inaieAlves, 2020-03-17:
I'm trying to connect the LAN8720 module with esp32_lora_ttgo_v1.But I can't get an IP or effectively connect the network, I can only detect that the network cable is connected or disconnected.

I recorded the signals with the 70MHz oscilloscope and the 50Mhz clock looks ok. The Tx1, Tx0, MDIO, MDC pins cannot see communication, since the Rx0 and Rx1 pins seem to me to be ok.

Below I present the registered signals.

CLOCK GPIO0
[image]

RX0 ( RX1 is similar)
[image]

MDIO
[image]

MDC
[image]

TX0(TX1 is similar)
[image]

#1938 

### Hardware:
Board:							ESP32 ttgo_lora v1
Framework						ESP-IDF v3.3.1-dirty
Flash Frequency:					26Mhz
Upload Speed:				       115200
Computer OS:                                         Ubuntu

my prototyping board (I tried to assemble similar to sautter (https://sautter.com/blog/ethernet-on-esp32-using-lan8720) previously and got the same problem so I set it up as follows, following the topic # 1938)

[image]
[image]
[image]

I also tried to add 33ohm resistors and 100nF capacitors on the Tx, Rx, CLK, MDIO and MDC pins and it didn't work.

I tried 3 different routers, I tried to set fixed IP, I tried to run on version 4.0 of ESP-IDF, I tried to use the arduino IDE, I tried to use external power supply and all this without result. 

I just needed to try another LAN7820, because I don't have another one.

### Sketch: 

`#include <stdio.h>
#include <string.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_event_loop.h"
#include "esp_event.h"
#include "esp_log.h"
#include "esp_eth.h"
#include "rom/gpio.h"
#include "tcpip_adapter.h"
#include "driver/gpio.h"
#include "driver/periph_ctrl.h"

#if CONFIG_PHY_LAN8720
#include "eth_phy/phy_lan8720.h"
#define DEFAULT_ETHERNET_PHY_CONFIG phy_lan8720_default_ethernet_config
#elif CONFIG_PHY_TLK110
#include "eth_phy/phy_tlk110.h"
#define DEFAULT_ETHERNET_PHY_CONFIG phy_tlk110_default_ethernet_config
#elif CONFIG_PHY_IP101
#include "eth_phy/phy_ip101.h"
#define DEFAULT_ETHERNET_PHY_CONFIG phy_ip101_default_ethernet_config
#endif

static const char *TAG = "eth_example";

#define PIN_PHY_POWER CONFIG_PHY_POWER_PIN
#define PIN_SMI_MDC CONFIG_PHY_SMI_MDC_PIN
#define PIN_SMI_MDIO CONFIG_PHY_SMI_MDIO_PIN

#ifdef CONFIG_PHY_USE_POWER_PIN
/**
 * @brief re-define power enable func for phy
 *
 * @param enable true to enable, false to disable
 *
 * @note This function replaces the default PHY power on/off function.
 * If this GPIO is not connected on your device (and PHY is always powered),
 * you can use the default PHY-specific power on/off function.
 */
static void phy_device_power_enable_via_gpio(bool enable)
{
    assert(DEFAULT_ETHERNET_PHY_CONFIG.phy_power_enable);

    if (!enable) {
        DEFAULT_ETHERNET_PHY_CONFIG.phy_power_enable(false);
    }

    gpio_pad_select_gpio(PIN_PHY_POWER);
    gpio_set_direction(PIN_PHY_POWER, GPIO_MODE_OUTPUT);
    if (enable == true) {
        gpio_set_level(PIN_PHY_POWER, 1);
        ESP_LOGI(TAG, "Power On Ethernet PHY");
    } else {
        gpio_set_level(PIN_PHY_POWER, 0);
        ESP_LOGI(TAG, "Power Off Ethernet PHY");
    }

    vTaskDelay(1); // Allow the power up/down to take effect, min 300us

    if (enable) {
        /* call the default PHY-specific power on function */
        DEFAULT_ETHERNET_PHY_CONFIG.phy_power_enable(true);
    }
}
#endif

/**
 * @brief gpio specific init
 *
 * @note RMII data pins are fixed in esp32:
 * TXD0 <=> GPIO19
 * TXD1 <=> GPIO22
 * TX_EN <=> GPIO21
 * RXD0 <=> GPIO25
 * RXD1 <=> GPIO26
 * CLK <=> GPIO0
 *
 */
static void eth_gpio_config_rmii(void)
{
    phy_rmii_configure_data_interface_pins();
    phy_rmii_smi_configure_pins(PIN_SMI_MDC, PIN_SMI_MDIO);
}

/**
 * @brief event handler for ethernet
 *
 * @param ctx
 * @param event
 * @return esp_err_t
 */
static esp_err_t eth_event_handler(void *ctx, system_event_t *event)
{
    tcpip_adapter_ip_info_t ip;

    switch (event->event_id) {
    case SYSTEM_EVENT_ETH_CONNECTED:
        ESP_LOGI(TAG, "Ethernet Link Up");
        break;
    case SYSTEM_EVENT_ETH_DISCONNECTED:
        ESP_LOGI(TAG, "Ethernet Link Down");
        break;
    case SYSTEM_EVENT_ETH_START:
        ESP_LOGI(TAG, "Ethernet Started");
        break;
    case SYSTEM_EVENT_ETH_GOT_IP:
        memset(&ip, 0, sizeof(tcpip_adapter_ip_info_t));
        ESP_ERROR_CHECK(tcpip_adapter_get_ip_info(ESP_IF_ETH, &ip));
        ESP_LOGI(TAG, "Ethernet Got IP Addr");
        ESP_LOGI(TAG, "~~~~~~~~~~~");
        ESP_LOGI(TAG, "ETHIP:" IPSTR, IP2STR(&ip.ip));
        ESP_LOGI(TAG, "ETHMASK:" IPSTR, IP2STR(&ip.netmask));
        ESP_LOGI(TAG, "ETHGW:" IPSTR, IP2STR(&ip.gw));
        ESP_LOGI(TAG, "~~~~~~~~~~~");
        break;
    case SYSTEM_EVENT_ETH_STOP:
        ESP_LOGI(TAG, "Ethernet Stopped");
        break;
    default:
        break;
    }
    return ESP_OK;
}

void app_main()
{
    tcpip_adapter_init();

            ESP_ERROR_CHECK(esp_event_loop_init(eth_event_handler, NULL));

    eth_config_t config = DEFAULT_ETHERNET_PHY_CONFIG;
    config.phy_addr = CONFIG_PHY_ADDRESS;
    config.gpio_config = eth_gpio_config_rmii;
    config.tcpip_input = tcpip_adapter_eth_input;
    config.clock_mode = CONFIG_PHY_CLOCK_MODE;
#ifdef CONFIG_PHY_USE_POWER_PIN
    /* Replace the default 'power enable' function with an example-specific one
     that toggles a power GPIO. */
    config.phy_power_enable = phy_device_power_enable_via_gpio;
#endif

    ESP_ERROR_CHECK(esp_eth_init(&config));
            ESP_ERROR_CHECK(esp_eth_enable()) ;
}`

### Debug Messages:
```
I (60) boot: Chip Revision: 1
I (35) boot: ESP-IDF v3.3.1-dirty 2nd stage bootloader
I (35) boot: compile time 13:50:45
I (44) boot: Enabling RNG early entropy source...
I (44) boot: SPI Speed      : 40MHz
I (44) boot: SPI Mode       : DIO
I (48) boot: SPI Flash Size : 4MB
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00100000
I (85) boot: End of partition table
I (89) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0f25c ( 62044) map
I (120) esp_image: segment 1: paddr=0x0001f284 vaddr=0x3ffb0000 size=0x00d8c (  3468) load
I (122) esp_image: segment 2: paddr=0x00020018 vaddr=0x400d0018 size=0x29e84 (171652) map
0x400d0018: _stext at ??:?

I (187) esp_image: segment 3: paddr=0x00049ea4 vaddr=0x3ffb0d8c size=0x01160 (  4448) load
I (189) esp_image: segment 4: paddr=0x0004b00c vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /root/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (194) esp_image: segment 5: paddr=0x0004b414 vaddr=0x40080400 size=0x08568 ( 34152) load
I (222) boot: Loaded app from partition at offset 0x10000
I (222) boot: Disabling RNG early entropy source...
I (223) cpu_start: Pro cpu up.
I (226) cpu_start: Application information:
I (231) cpu_start: Project name:     ethernet_demo
I (237) cpu_start: App version:      1
I (241) cpu_start: Compile time:     Mar 16 2020 13:50:36
I (247) cpu_start: ELF file SHA256:  1bf49e2a520ef43d...
I (253) cpu_start: ESP-IDF:          v3.3.1-dirty
I (259) cpu_start: Starting app cpu, entry point is 0x40080f3c
0x40080f3c: call_start_cpu1 at /root/esp/esp-idf/components/esp32/cpu_start.c:269

I (0) cpu_start: App cpu up.
I (269) heap_init: Initializing. RAM available for dynamic allocation:
I (276) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (282) heap_init: At 3FFB4080 len 0002BF80 (175 KiB): DRAM
I (288) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (294) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (301) heap_init: At 40088968 len 00017698 (93 KiB): IRAM
I (307) cpu_start: Pro cpu start user code
I (102) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (104) eth_example: Power On Ethernet PHY
I (114) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (124) emac: emac reset done
I (124) eth_example: Ethernet Started
I (4124) eth_example: Ethernet Link Up
I (14124) eth_example: Ethernet Link Down
I (18124) eth_example: Ethernet Link Up
```
I get a link up whenever I connect the ethernet cable and link down whenever I disconnect the cable, but I don't get the ip address or an "effective" connection.

I'm sure my settings are ok, as seen below.

[image]
qq
Note: I tried to provide as much information as possible, if any information is missing I am available to provide.

I know that there are similar topics however so far nothing has solved my problem and I am days looking for the solution.

thank anyone who can help, grateful

xenpac, 2020-03-25:
Hi inaieAlves,

dont despair...things can go wrong ;-)
Asto your logs...
50Mhz is coming in,  "emac reset done".
everything looks fine...except you dont get an Ip address.
Soo...you are not communicating with your network.

I also have one of my LAN8720 boards from china that has exact same symptoms.
i suspect some defect in the connector (that integrates a transformer).
It looks like the chip on the LAN8720-board is up and running, as you talk to him.
But somehow there is a broken connection to that Jack/connector/transformer that gets you out to the LAN.
nice you have this mega-scope for investigating.

I will investigating my board in the next few days to see whats wrong.
maybe we can update here..

inaieAlves, 2020-03-27:
@xenpac Hi,

Thanks for looking at my case.

I am following the topic and any necessary information I am available to provide:

In fact, link up / link dow only provides me with information that the module is communicating with esp32 in some way.

Success and thanks!

xenpac, 2020-04-01:
The link up/down is detected by some control-signals being exchanged between esp and the router. that way they know a cable is inserted. So it tells yo "cable connected".
However data traffic is missing......somehow i guess its the transformer in that jack on the board that has some bad connection. havent investigated further jet.

sauttefk, 2020-04-01:
Link-status and configuration is communicated over the TWI-interface (MDC / MDIO), which are relatively low frequency signals.
The actual ethernet data is transmitted via TXD0/TXD1/RXD0/RDX1/CLK, which are 25MHz-signals, that most probably are a problematic in your setup.

xenpac, 2020-04-01:
Success ;-)
My Problem solved by resoldering the SMC chip.

I had two problems:
- 1 EMAC timout/init failed/regs=0xffff       fix: resoldered just Pin 10
- 2 after fix 1, got LINK UP, but no IP-address.     fix: resolder whole chip

So it was not the transformer/connector.

PHY init errors usually are caused by WRONG TWI address. (being 0 instead of 1)
The lan8720's default address is determined by the state of its PHYAD0 pin(10) coming out of reset. 

The Problem of no communication over LAN was caused by bad connections of the SMC-chip pins(20 to 24) related to LINE side.

To resolder, put some Flux on the pins and slide the "iron" gently from the chip outbound. 
Repeat if you get a "short".

xenpac, 2020-04-01:
ps.  maybe it is of interest, that i did not modify my LAN8720 boards with that 50mhz enable control.

instead (on my custom esp-board) i placed a 1K2 resistor in series of the 50mhz line coming from the PHY-board.......going to IO0 of esp.
Also i added a 6K8 resistor from IO0 to 3.3V as pullup. (experiment with lower values! until reset clean)

That way esp is not going into Bootloadermode after reset.......as the 50mhz are dampened but still there.

Then adding a little RF-choke to IO0 (to isolate the 50mhz clock )  -> i can connect my serial interface PROG-enable line to IO0.
This way i can leave the serial programmer connected and still have ethernet.

closed 2020-06-14 by stale[bot]

dienvx1997bn, 2020-12-07:
i can use module lan8720, just remove 50Mhz clock, connect nINT/RETCLK to GPIO_0 and have 4.7k pull up resistor. Change default config to use ETH_CLOCK_GPIO0_OUT and ETH_ADDR = 1
[image]

datedave, 2021-01-11:
> i can use module lan8720, just remove 50Mhz clock, connect nINT/RETCLK to GPIO_0 and have 4.7k pull up resistor. Change default config to use ETH_CLOCK_GPIO0_OUT and ETH_ADDR = 1
> [image]

HI dienvx1997bn, any chance you can post the wiring digarmme or connection table as I would like to have a go and reproduce what you have here!

thanks

brano1990, 2021-02-22:
Hi does anyone have a manual schematic for wiring esp32 lan8720 as shown here? remove 50Mhz clock

nardev, 2021-09-28:
The issue description looks so similar to what just had happened with my boards.

But the "solution" was something quite different.

1. When i used the BLACK LAN8720 board from aliexpress, actually, everything was working fine.
[image]

2. When i tried almost 1:1 same board by electrodragon, it was detecting the cable but i was not able to verify 2 layer connection.
[image]

**However, i had local switch in between the router with DHCP server, when i connected directly the cable to the ROUTER, it was able to fetch the IP and everything worked fine.**

I have no idea what was causing the issue, the small switch is just a dummy 5 port one and it works well with 100Mbps.

The mystery is that basically same board with same connections works through sitch and another one doesn't get the IP

brano1990, 2021-09-28:
@nardev wiring diagram BLACK LAN8720

rakzz14, 2022-01-19:
Hello all,

I am facing a similar issue with my FPGA connected to LAN8720 

I am giving a brief info on my connections 
-Clk/ref int derived from LAN8720 to pmod 1 of fpga
-Txd0 /txd1 along with tx_en are connected to randomly 3 pmod pins of fpga 
-I can see activities on tx_data lines ,4 V peak to peak and 2.5 MHz
- clock of 50Mhz 8V peak to peak 
-Not connected MDC/MDIO 

Scenario;
I am generating UDP frames from my verilog code
But not sure what values should i assign these for ?
Destination ip address
Source ip address
Source port 
Dest port
Dest address 
Source address

But when i connect my ethernet cable to LAN8720 and analyse the traffic on wireshark i see nothing 

But on oscillosope i see all the activities on my pins at LAN8720 board

I have attached few  snapshots 
Request you to help me out get the frames out of LAN8720

rakzz14, 2022-01-19:
![PhotoGrid_1642604419669](https://user-
[image]
[image]
images.githubusercontent.com/98060560/150222343-06cd4721-83a3-4d70-ab55-cc95d4cbe7b3.jpg)
[image]

DatanoiseTV, 2022-02-28:
I have a similar problem. I see received packets, but TX doesnt work.
Did you resolve the issue?