espressif/arduino-esp32 #3832: How can i send a .bin file to esp32 through bluetooth?

testter2, 2020-03-19:
I have an esp32, I need to update the firmware through Bluetooth.

so that .bin file wants to send through Bluetooth.., could you please tell me,
how can it possible?

Thanks.

abdset, 2020-03-22:
When flashing, bluetooth is switched off at that time. this mean you just cant do it.

testter2, 2020-03-22:
> 
> 
> When flashing, bluetooth is switched off at that time. this mean you just cant do it.

doesn't there any way to transfer the whole bin file to the esp first ?!
then after receiving the data ,may be stored in another partition of the ESP, Start flashing the bin file.

abdset, 2020-03-22:
Take a look here (seems like i was wrong):
https://esp32.com/viewtopic.php?f=2&t=2221&sid=c74c6ac78f89458ae3e92bce1f7b0be3&start=10

testter2, 2020-03-23:
> 
> 
> Take a look here (seems like i was wrong):
> https://esp32.com/viewtopic.php?f=2&t=2221&sid=c74c6ac78f89458ae3e92bce1f7b0be3&start=10

i saw  it.
that's meant that it really could be done.
but i think i need to know how to divide the bin file so i could send the data.
and how i could save it first in ESP?
if found anything related to that !
thanks in advance

closed 2020-06-05 by stale[bot]