espressif/arduino-esp32 #3838: Error with CameraWebServer Sketch - Guru Meditation Error: Core 0 panic'ed (IllegalInstruction).

adbailey1, 2020-03-20:
### Hardware:
Board:							ESP32-S (I think this is the AI-Thinker board according to Wiki) using with FT232RL FTDI USB to TTL Serial Converter
Core Installation version:			1.0.4 (ESP32 by Espressif Systems)
IDE name:						Arduino 1.8.10
Upload Speed:				       115200
Computer OS:                                        Windows 10

### Description:
I am trying to run the CameraWebServer example sketch with arduino IDE, my first attempt was the example sketch WiFiScan and this worked fine. It manages to upload (I then remove connection of GND->IO0 and press reset) but when I try and run the serial monitor to obtain the IP address, I runs into an exception and reboots to try again. 

I tried external power supply for the ESP32 Module but this didn't work, along with trying 3.3V instead of 5V. I tried a separate USB cable to be sure it wasn't this. 

I am very new to all this so any pointers would be greatly appreciated.

### Sketch: 
```

#include "esp_camera.h"
#include <WiFi.h>

//
// WARNING!!! Make sure that you have either selected ESP32 Wrover Module,
//            or another board which has PSRAM enabled
//

// Select camera model

#define CAMERA_MODEL_AI_THINKER

#include "camera_pins.h"

const char* ssid = "***";
const char* password = "***";

void startCameraServer();

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.println();

  camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sscb_sda = SIOD_GPIO_NUM;
  config.pin_sscb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.pixel_format = PIXFORMAT_JPEG;
  //init with high specs to pre-allocate larger buffers
  if(psramFound()){
    config.frame_size = FRAMESIZE_UXGA;
    config.jpeg_quality = 10;
    config.fb_count = 2;
  } else {
    config.frame_size = FRAMESIZE_SVGA;
    config.jpeg_quality = 12;
    config.fb_count = 1;
  }

#if defined(CAMERA_MODEL_ESP_EYE)
  pinMode(13, INPUT_PULLUP);
  pinMode(14, INPUT_PULLUP);
#endif

  // camera init
  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    Serial.printf("Camera init failed with error 0x%x", err);
    return;
  }

  sensor_t * s = esp_camera_sensor_get();
  //initial sensors are flipped vertically and colors are a bit saturated
  if (s->id.PID == OV3660_PID) {
    s->set_vflip(s, 1);//flip it back
    s->set_brightness(s, 1);//up the blightness just a bit
    s->set_saturation(s, -2);//lower the saturation
  }
  //drop down frame size for higher initial frame rate
  s->set_framesize(s, FRAMESIZE_QVGA);

#if defined(CAMERA_MODEL_M5STACK_WIDE)
  s->set_vflip(s, 1);
  s->set_hmirror(s, 1);
#endif

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");

  startCameraServer();

  Serial.print("Camera Ready! Use 'http://");
  Serial.print(WiFi.localIP());
  Serial.println("' to connect");
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(10000);
}
```

### Debug Messages Serial Output:

ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400db308: ffffffff ffffffff ffffffff
Core 0 register dump:
PC      : 0x400db30c  PS      : 0x00060e30  A0      : 0x800d776c  A1      : 0x3ffbb9f0  
A2      : 0x00000001  A3      : 0x00000001  A4      : 0x3ffc2900  A5      : 0x00000000  
A6      : 0x00000080  A7      : 0x00000000  A8      : 0x800852c1  A9      : 0x3ffbb9d0  
A10     : 0x3ffc2900  A11     : 0x00c86016  A12     : 0x3ff49060  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000017  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x400db30c:0x3ffbb9f0 0x400d7769:0x3ffbba20 0x400d34c8:0x3ffbba40 0x400d3441:0x3ffbba60 0x400d3db3:0x3ffbba80 0x400d73ae:0x3ffbbab0 0x4008db91:0x3ffbbad0

Rebooting...

### Debug Messages Using ESP Exception Decoder:

PC: 0x400db30c: psram_read_id at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/spiram_psram.c line 404
EXCVADDR: 0x00000000

Decoding stack results
0x400db30c: psram_read_id at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/spiram_psram.c line 404
0x400d7769: esp_spiram_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/spiram.c line 148
0x400d34c8: psramInit at C:\Users\Andrew\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\esp32-hal-psram.c line 28
0x400d3441: initArduino at C:\Users\Andrew\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\esp32-hal-misc.c line 200
0x400d3db3: app_main() at C:\Users\Andrew\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\main.cpp line 26
0x400d73ae: main_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/cpu_start.c line 506
0x4008db91: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

adbailey1, 2020-03-20:
I also tried to erase the flash using esptools.py (https://github.com/espressif/esptool), however this did not work either.

WereCatf, 2020-03-20:
@adbailey1 The sketch rather clearly says at the top that you need a board with PSRAM. ESP32-S doesn't have PSRAM.

adbailey1, 2020-03-20:
@WereCatf thanks for the reply. I was following this tutorial (https://randomnerdtutorials.com/esp32-cam-video-streaming-face-recognition-arduino-ide/), they seem to be using the same ESP32-CAM (although their chip has Ai-Thinker underneath the ESP32-S) and FTDI boards as I am but do not mention having to configure or add PSRAM. And while I don't doubt that PSRAM might be the problem, I'm not sure how the tutorial works as their board shouldn't have PSRAM either according to https://en.wikipedia.org/wiki/ESP32. 

Also from the sketch this condition statment suggests that if there is no PSRAM available then the image quality will be reduced to compensate? 

`  if(psramFound()){
    config.frame_size = FRAMESIZE_UXGA;
    config.jpeg_quality = 10;
    config.fb_count = 2;
  } else {
    config.frame_size = FRAMESIZE_SVGA;
    config.jpeg_quality = 12;
    config.fb_count = 1;
  }`

If I am still wrong, the datasheet says that 4M external PSRAM is possibles. Is there a way to do this simply or would the best solution be to return my current module and get a replacement?

lbernstone, 2020-03-20:
Do you see the part at the top of the sketch which has WARNING in all caps, that says you must have a board with PSRAM?
Make sure you have selected a board in Arduino that has PSRAM enabled (the AI Thinker ESP32-cam does).
This little sketch will verify that you have psram:
```
void setup() {
  Serial.begin(115200);
  Serial.printf("PSRAM size: %d\n", ESP.getPsramSize());
  Serial.printf("PSRAM free: %d\n", ESP.getFreePsram());
}
void loop() {}
```

adbailey1, 2020-03-20:
@lbernstone thanks for your reply and helpful sketch! 

Just for clarity, even though this module's chip (https://uk.banggood.com/ESP32-CAM-WiFi-bluetooth-Camera-Module-Development-Board-ESP32-With-Camera-Module-OV2640-p-1394679.html?rmmds=detail-top-buytogether-auto&cur_warehouse=UK) says AI-Thinker on it, the PSRAM is still external and therefore not suitable? This is the case with mine, as the datasheet says 4M external PSRAM. 

If this is the case then I think that's where my confusion has been.

WereCatf, 2020-03-20:
@adbailey1 If you have that board, you need to select the "AI Thinker ESP32-CAM" as your board, not "ESP32-S"

adbailey1, 2020-03-20:
@WereCatf my current board is this: [https://www.banggood.com/ESP32-CAM-WiFi-bluetooth-Development-Board-ESP32-with-FT232RL-FTDI-USB-to-TTL-Serial-Converter-40-Pin-Jumper-p-1605463.html?rmmds=detail-left-hotproducts__5&cur_warehouse=UK](url) but I am thinking about getting the one I linked above. 

However, on the description for both it says that PSRAM is external.

WereCatf, 2020-03-20:
@adbailey1 Those both are the same board, just a different manufacturer and yes, you already said about PSRAM being external -- that's irrelevant. It just means the 4MB PSRAM is outside the metal shield, instead of inside it.

In Arduino IDE, set the board as "AI Thinker ESP32-CAM" to fix your problem.

adbailey1, 2020-03-20:
@WereCatf ok, sorry, so I did that and that still brought me the error seen at the start of this thread.

So to confirm, from the sketch above I use: #define CAMERA_MODEL_AI_THINKER

and I selected the board to be: AI Thinker ESP32-CAM

WereCatf, 2020-03-21:
@adbailey1 
> The output gave the infinite loop serial message as in my first post

It should work and report PSRAM as 4MB, if you really have the board you linked to and you have the correct board ("AI Thinker ESP32-CAM") selected in Arduino IDE. It seems that either you actually have some other board, there's something wrong with your board itself, or there's something wrong with your installation.

Just to verify you have the right board, why don't you take a couple of pictures of it and post links to the pictures here? Try to take a good, clear picture from both sides.

adbailey1, 2020-03-21:
@WereCatf thanks for the message. Here is a link to the photos of the CAM: [https://imgur.com/gallery/8qtHG44](url)

If there is something wrong with the installation, I'm not sure what it could be, I have already uninstalled the ESP32 board libraries and re-installed them and still met this error. I think there may be something wrong with the board itself.

beegee-tokyo, 2020-03-21:
Maybe not of help, but I have a very similar [ESP32-CAM](https://circuit.rocks/product:2659) board. The module on it says as well ESP32-S. The board looks the same, just the of the prints are missing.

I wrote a [tutorial for RTSP streaming](https://learn.circuit.rocks/esp32-cam-with-rtsp-video-streaming) and it works without problems (beside of being very slow and a 2 seconds delay).

The sources are on [Github](https://github.com/circuitrocks/ESP32-RTSP). 

In ArduinoIDE I selected _**AI Thinker ESP32-CAM**_ and it compiles without problems.
[image]

adbailey1, 2020-03-22:
@beegee-tokyo thanks very much for this link. I uploaded the code via arduino IDE while holding the upload button down. This worked. When I pressed reset, it said "waiting for download" on the serial monitor (still holding upload button down) but the moment I pressed the reset button on the back of the board and released the upload button, I received the repeating error message stated at the start of this post. 

Unless I am doing something basic incorrectly, I think my board might be faulty?

beegee-tokyo, 2020-03-23:
The code works on my board.
Your debug says it crashes on **`psramInit`**, so I think something is wrong with your board.

To check your PSRAM, try a very simple sketch like
```cpp
#include <Arduino.h>

void setup(void)
{
	// Start Serial
	Serial.begin(115200);
	Serial.println("\n\n##################################");
	Serial.printf("Internal Total heap %d, internal Free Heap %d\n", ESP.getHeapSize(), ESP.getFreeHeap());
	Serial.printf("SPIRam Total heap %d, SPIRam Free Heap %d\n", ESP.getPsramSize(), ESP.getFreePsram());
	Serial.printf("ChipRevision %d, Cpu Freq %d, SDK Version %s\n", ESP.getChipRevision(), ESP.getCpuFreqMHz(), ESP.getSdkVersion());
	Serial.printf("Flash Size %d, Flash Speed %d\n", ESP.getFlashChipSize(), ESP.getFlashChipSpeed());
	Serial.println("##################################\n\n");
}
void loop()
{
	delay(100);
}
```
If your PSRAM is working, it should tell you how much PSRAM is available.
If your board has no PSRAM, it should zero.
If you PSRAM is faulty, it should crash the same way as the code you tried before.

WereCatf, 2020-03-23:
@adbailey1 I looked at the pics and sure enough, that is a clone of the "AI Thinker ESP32-CAM" and all the code in this thread should work without an issue, if you have the correct board in the Arduino IDE selected. As such, it looks like something's wrong with your board.

adbailey1, 2020-03-25:
@beegee-tokyo Thanks for this test. I received the same recursive error as before. 

@WereCatf @lbernstone @beegee-tokyo Thank you all for your help, I will be returning the board as faulty.

adbailey1, 2020-04-01:
Hi, Just to confirm, I ordered a new camera and tested it today. It works perfectly so I'll close this as it has been confirmed that the PSRAM was faulty. Thanks for everyone's help again.

closed 2020-04-01 by adbailey1

pimentdubois, 2021-11-30:
Hardware:

Board: ESP32-S (I think this is the AI-Thinker board according to Wiki) 
	using with FT232RL FTDI USB to TTL Serial Converter and Micro Interface USB, ESP32-CAM-MB
Core Installation version: 1.0.4 (ESP32 by Espressif Systems)
IDE name: Arduino 1.8.16
Upload Speed: 115200
Computer OS: Windows 10

Description:

Here, I am at the end of my resources. I went around the web to try to correct the errors
initialization of PSRAM psramInit (); with the ESP32-CAM card.

I have an error 500 which prevents me from activating the "Face Detection" and "Face Recognition" functions

PSRAM test

Arduino IDE serial monitor output

19:51:05.985 -> ets Jun  8 2016 00:22:57
19:51:05.985 -> 
19:51:05.985 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
19:51:05.985 -> configsip: 0, SPIWP:0xee
19:51:05.985 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
19:51:05.985 -> mode:DIO, clock div:1
19:51:05.985 -> load:0x3fff0030,len:1420
19:51:05.985 -> ho 0 tail 12 room 4
19:51:05.985 -> load:0x40078000,len:13540
19:51:05.985 -> load:0x40080400,len:3604
19:51:05.985 -> entry 0x400805f0
19:51:06.975 -> 
19:51:06.975 -> 
19:51:06.975 -> ##################################
19:51:06.975 -> Internal Total heap 366004, internal Free Heap 341224
19:51:06.975 -> SPIRam Total heap 4192139, SPIRam Free Heap 4192139
19:51:06.975 -> ChipRevision 1, Cpu Freq 240, SDK Version v4.4-dev-3569-g6a7d83af19-dirty
19:51:06.975 -> Flash Size 4194304, Flash Speed 80000000
19:51:06.975 -> ##################################
19:51:07.022 -> 
19:51:07.022 -> 

So the ESP32-CAM has 4.2MB Flash memory.

It does not initialize with the Random Nerd Tutorials sketch, nor the CameraWebServer sketch

During my experiments, I noticed several small problems very often encountered by
other AI Thinker ESP32-CAM (ESP32-S) users.

Arduino IDE -> Put the IO0 pin to Ground and press the reset button to launch
downloading the sketch.

Same with Visual Studio Code and PlatformIO.

It works and after a few times it's the opposite more NOTHING ...

Here is one of the many errors I encountered:

Arduino IDE monitor output

17:43:15.440 -> ets Jun  8 2016 00:22:57
17:43:15.440 -> 
17:43:15.440 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
17:43:15.488 -> configsip: 0, SPIWP:0xee
17:43:15.488 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
17:43:15.488 -> mode:DIO, clock div:2
17:43:15.488 -> load:0x3fff0018,len:4
17:43:15.488 -> load:0x3fff001c,len:1044
17:43:15.488 -> load:0x40078000,len:10124
17:43:15.488 -> load:0x40080400,len:5828
[... 184 lines cut ...]
  Serial.println("WiFi connected");

  startCameraServer();

  Serial.print("Camera Ready! Use 'http://");
  Serial.print(WiFi.localIP());
  Serial.println("' to connect");
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(10000);
}

In the folder, I add the following files:
app_httpd.cpp
camera_index.h
camera_pins.h

to CameraWebServer.ino file

The folder path is as follows:
C: \ Users \ Trouble \ Documents \ Arduino \ libraries \ ESP32 \ examples \ Camera \ CameraWebServer

Please, I would need some help.