espressif/arduino-esp32 #3842: When changing IP address of AP results in network timeouts

brownrb, 2020-03-23:
### Hardware:
Board:							ESP32 Dev Module VROOM
Core Installation version:			104
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                     Disabled
Upload Speed:					115200
Computer OS:                                        Windows 10

### **Description:
I have the AP working with no issues on 192.168.4.1
I have now coded according the guidelines using a delay.
The intent was to change the AP address.
the code is
    WiFi.mode(WIFI_AP);
    delay(200);
    if ( WiFi.softAPConfig(ip, gateway, subnet) == true )
    {
      DebugPrintln("Error setting IP address of AP");
    }
    else
    {
      DebugPrintln("IP address of AP set correctly");
    }
    WiFi.softAP(mySSID, myPASSWORD);  
where ip = 192.168.5.1, gateway=192.168.5.1 and subnet=255.255.255.0

the Ip address gets changed as expected, BUT the result is increased response times to network requests.
I have used PING command to ping the controller at 192.168.5.1 and regularly get time outs with delay times in excess of 1s and greater. Most of the pings timeout. I am running no other code on the ESP32. 

When I configure the AP using this code
  WiFi.config(ip, dns, gateway, subnet);
  WiFi.mode(WIFI_AP);
  WiFi.softAP(mySSID, myPASSWORD);
using an address of 192.168.4.1 it works perfectly and ping response times are < 2ms.

i believe there is a bug causing increased network response times when changing the AP ip address.

#include <WiFiServer.h>
#include <WiFiClient.h>
#include <ArduinoJson.h>

#include <WiFi.h>
#include "SPIFFS.h"
#include <SPI.h>

char mySSID[64] = "mysqmproesp";
char myPASSWORD[64] = "mysqmproesp";

IPAddress ip(192, 168, 5, 1);                 // AP static IP - you can change these values to change the IP
IPAddress dns(192, 168, 5, 1);                // just set it to the same IP as the gateway
IPAddress gateway(192, 168, 5, 1);
IPAddress subnet(255, 255, 255, 0);

String  ipStr;                                  // shared between modes
IPAddress   myIP;                               // local IP address of this controller

void setup()
{
  WiFi.config(ip, dns, gateway, subnet);
  WiFi.mode(WIFI_AP);
  WiFi.softAP(mySSID, myPASSWORD);
  /*
    WiFi.mode(WIFI_AP);
    delay(200);
    if ( WiFi.softAPConfig(ip, gateway, subnet) == true )
    {
      DebugPrintln("Error setting IP address of AP");
    }
    else
    {
      DebugPrintln("IP address of AP set correctly");
    }
    WiFi.softAP(mySSID, myPASSWORD);              // this call has a limit of 4 simultaneous connections to the WiFi network
  */
  myIP = WiFi.localIP();
}

void loop()
{
}

brownrb, 2020-10-09:
This is still relevant. Can someone please confirm.
It really is simple to replicate.
Just load up the code on an ESP32 device.
When AP is 192.168.4.1 any Ping to that IP is returned immediately < 2ms
When AP is NOT, like 192.168.5.1, Ping is NOT returned immediately, most often times out of > 200ms

#include <WiFiServer.h>
#include <WiFiClient.h>
#include <ArduinoJson.h>

#include <WiFi.h>
#include "SPIFFS.h"
#include <SPI.h>

char mySSID[64] = "mysqmproesp";
char myPASSWORD[64] = "mysqmproesp";

IPAddress ip(192, 168, 5, 1);                 // AP static IP - you can change these values to change the IP
IPAddress dns(192, 168, 5, 1);                // just set it to the same IP as the gateway
IPAddress gateway(192, 168, 5, 1);
IPAddress subnet(255, 255, 255, 0);

String  ipStr;                                  // shared between modes
IPAddress   myIP;                               // local IP address of this controller

void setup()
{
  WiFi.config(ip, dns, gateway, subnet);
  WiFi.mode(WIFI_AP);
  WiFi.softAP(mySSID, myPASSWORD);
  /*
    WiFi.mode(WIFI_AP);
    delay(200);
    if ( WiFi.softAPConfig(ip, gateway, subnet) == true )
    {
      DebugPrintln("Error setting IP address of AP");
    }
    else
    {
      DebugPrintln("IP address of AP set correctly");
    }
    WiFi.softAP(mySSID, myPASSWORD);              // this call has a limit of 4 simultaneous connections to the WiFi network
  */

  myIP = WiFi.localIP();
}

void loop()
{

}

brownrb, 2020-10-09:
Updated

brownrb, 2020-10-09:
Issue is still there

lbernstone, 2020-10-09:
Calling WiFi.mode while the service is active will cause bad behavior.   This works for me:
```
#include <WiFi.h>
char mySSID[64] = "mysqmproesp";
char myPASSWORD[64] = "mysqmproesp";
IPAddress ip(192, 168, 5, 1);
IPAddress gateway(192, 168, 5, 1);
IPAddress subnet(255, 255, 255, 0);
void setup() {
  WiFi.mode(WIFI_AP);
  WiFi.softAP(mySSID, myPASSWORD);
  delay(2000);
  WiFi.softAPConfig(ip, gateway, subnet);
}
void loop() {}
```

r2d2r01, 2020-11-30:
Hi, I have exactly the same problem, the solution IS the delay.

closed 2021-02-13 by stale[bot]