espressif/arduino-esp32 #3846: BLE Mesh support (compilation failure)

faxik, 2020-03-24:
Versions:
=======
- esp-idf: git latest (6330b3345e87eb4401e7be7c8b6fea2870c35d9f)
- arduino-esp32: git latest  (0f772270fbb32099cad72053755d6b4c7e8e1f82)

### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			git
IDE name:						IDF component
Computer OS:                                        Ubuntu

### Description:
I'm trying to use BLE Mesh. As I understand, its support is available in IDF 4.
However, when I try to compile (either `master` or `idf-4.0` branches), a lot of errors emerge. 

```
In file included from [PROJECT-DIR[/components/arduino/cores/esp32/esp32-hal-gpio.c:18:
[ESP-IDF-DIR]/components/esp32/include/rom/ets_sys.h:1:2: warning: #warning rom/ets_sys.h is deprecated, please use esp32/rom/ets_sys.h instead [-Wcpp]
 #warning rom/ets_sys.h is deprecated, please use esp32/rom/ets_sys.h instead
  ^~~~~~~
In file included from [PROJECT-DIR[/components/arduino/cores/esp32/esp32-hal-gpio.c:20:
[ESP-IDF-DIR]/components/esp32/include/esp_intr.h:16:2: warning: #warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead [-Wcpp]
 #warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead
  ^~~~~~~
In file included from [PROJECT-DIR[/components/arduino/cores/esp32/esp32-hal-gpio.c:21:
[ESP-IDF-DIR]/components/esp32/include/rom/gpio.h:1:2: warning: #warning rom/gpio.h is deprecated, please use esp32/rom/gpio.h instead [-Wcpp]
 #warning rom/gpio.h is deprecated, please use esp32/rom/gpio.h instead
  ^~~~~~~
[PROJECT-DIR[/components/arduino/cores/esp32/esp32-hal-gpio.c: In function '__pinMode':
[PROJECT-DIR[/components/arduino/cores/esp32/esp32-hal-gpio.c:90:24: error: 'rtc_gpio_desc' undeclared (first use in this function); did you mean 'rtc_io_desc'?
     uint32_t rtc_reg = rtc_gpio_desc[pin].reg;
```

I saw that there are a lot of incompatibilities in `soc` component; furthermore there are two of them: in `ESP-IDF` and in `arduino`, and they are not the same. I don't know which of the suggested by compiler changes are correct, if any.

Is it any work on that? I'm stuck at it for several days. Please, help!

lbernstone, 2020-03-24:
If you are using the [lib-builder](https://github.com/espressif/arduino-esp32/blob/master/docs/lib_builder.md), you should make sure to check out the release/v4.0 branch.

faxik, 2020-03-24:
Yes, I do use `lib-builder`. Checked this in `release/v4.0` branch and in
master. Probably, I should perform some 'deep clean' to make sure the
correct versions of files are picked up? Please, advise.

lbernstone, 2020-03-24:
If you are working on the bleeding edge, you should know how to throw a stitch.
Add `#define rtc_gpio_desc rtc_io_desc` to the top of components/esp32-camera/driver/twi.c and components/arduino/cores/esp32/esp32-hal-gpio.c
Then run `IDF_COMMIT=463a9d8b7f9af8205222b80707f9bdbba7c530e1 ./build.sh`
This is the commit of the IDF 4.0 release version that lib-builder was pegged against.

atanisoft, 2020-03-24:
Also make sure you are using the arduino-esp32 idf-release/v4.0 branch

closed 2020-06-06 by stale[bot]

phoenixxd, 2020-09-23:
Were you able to make some progress @faxik ?