espressif/arduino-esp32 #3854: reorder as error

neu-rah, 2020-03-27:
this turn of warning into error is disabling some libs to compile, is this really needed?

https://github.com/espressif/arduino-esp32/blob/1977370e6fc069e93ffd8818798fbfda27ae7d99/tools/platformio-build.py#L63

liebman, 2020-03-27:
You can undo that effect with [build_unflags](https://docs.platformio.org/en/latest/projectconf/section_env_build.html#build-unflags) like this (cut from one of my projects):
```
build_unflags =
    ; esp32 core disabled these warnings, re-enable them
    -Wno-maybe-uninitialized
    -Wno-unused-function
    -Wno-unused-but-set-variable
    -Wno-unused-variable
    -Wno-deprecated-declarations
    -Wno-unused-parameter
    -Wno-sign-compare
    -Wno-old-style-declaration
    -Wno-error=maybe-uninitialized
    -Wno-error=unused-function
    -Wno-error=unused-but-set-variable
    ;-Wno-error=unused-variable
    -Wno-error=deprecated-declarations
```

neu-rah, 2020-03-27:
indeed, and that is what I'm doing, but them I have to pass that to users of my library because they can not build and in this case its from a dependency lib.
Removing that seems more accurate to me as it should really be a warning, not an error.

liebman, 2020-03-27:
or submit pull requests against libraries you have issues with.  Personally I prefer 0 warnings and strict checks.....

neu-rah, 2020-03-28:
me too but
this is a warning turned into error by `-Werror=reorder`
and I wont force the world for this.

closed 2020-03-28 by neu-rah