espressif/arduino-esp32 #3855: Uploading to WeMos D1 R2

nsiatras, 2020-03-27:
Hello everyone,

I followed the Installation instructions using Arduino IDE Boards Manager
(https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md)

But I do not know what board to select in order to compile the code.
So far I selected 
1) WEMOS LOLIN32
2) WEMOS D1 MINI ESP32

And every time I get the error **"A fatal error occurred: Timed out waiting for packet header"**

My board is the following **WEMOS D1 R2**
https://d.scdn.gr/images/sku_main_images/010578/10578958/xlarge_20161208120411_wemos_d1_r2_v2_1.jpeg

CelliesProjects, 2020-03-27:
Are you sure it is an esp32 board? I think it is an ESP8266 board.

https://alexnld.com/product/new-wemos-d1-r2-v2-1-0-wifi-uno-based-esp8266-for-arduino-nodemcu-compatible/

Looks much the same.

nsiatras, 2020-03-27:
Well... It is an ESP8266 board. I am looking for ESP32 now.
Thanks all for your help

closed 2020-06-09 by stale[bot]