espressif/arduino-esp32 #3856: Stack smashing protect failure (using the BLE library)

jaggil, 2020-03-27:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			esp32-1.0.4
BLE lib :                                                    Included in the Arduino ESP32 core 1.0.4
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:					115200
Computer OS:                                         Windows 10

### Description:
My program aims to test the Xiaomi-Mijia-LYWSD03MMC humidity and temperature sensor.
This device can only be read through notifications.
To try to save battery, I want to turn notifications on and off from time to time, for example every minute, but keeping the BLE connection with the device.
The program basically works, but every so often and randomly, the error "Stack smashing protect failure!" Occurs.
I have done several tests and introduced some modifications after reading many posts, but there have been no results.
There is always a sequence when the error occurs, which is that the "Notifications turned on" command stops appearing on the serial port, and after receiving four reads / notifications the error occurs and the ESP32 automatically restarts.

Other tests carried out are the following:
1.- Cancel the on / off commands of the loop (), canceling the two programming lines
// pRemoteCharacteristic_THB-> getDescriptor (BLEUUID ((uint16_t) 0x2902)) -> writeValue ((uint8_t *) notificationOff, 2, true);
 and the
// pRemoteCharacteristic_THB-> getDescriptor (BLEUUID ((uint16_t) 0x2902)) -> writeValue ((uint8_t *) notificationOn, 2, true);
In this case, and after several days, NO error occurs.
2.- Cancel only the loop () off command, canceling the programming line
// pRemoteCharacteristic_THB-> getDescriptor (BLEUUID ((uint16_t) 0x2902)) -> writeValue ((uint8_t *) notificationOff, 2, true);
 In this case, and after several days, no error occurs either.

Therefore, the error occurs when both commands are used in loop ().

By the way, no additional information of the origin of the problem is obtained if I copy the error message in the Exception Decoder in the ArduinoIDE
If I activate the Core Debug Level option in Verbose mode, the messages are not enlightening either

Please can you give me some idea to locate my fault.
Am I using the BLE library commands correctly?
Thanks in advance

### Sketch: 
```cpp

//https://github.com/karolkalinski/esp32-snippets
//https://randomnerdtutorials.com/esp32-bluetooth-low-energy-ble-arduino-ide/
//https://medium.com/home-wireless/how-to-program-an-esp32-in-arduino-while-using-esp-idf-functions-90033d860f75
#define BTU_TASK_STACK_SIZE (6144 + BT_TASK_EXTRA_STACK_SIZE)
#include <BLEDevice.h>
#include <Esp.h>

//#define LYWSD03MMC_ADDR "a4:c1:38:06:78:d8"
#define LYWSD03MMC_ADDR "a4:c1:38:89:78:1c"

BLEClient* pClient;
static BLEAddress htSensorAddress(LYWSD03MMC_ADDR);

bool connectionSuccessful = false;

// The remote service we wish to connect to.
static BLEUUID serviceUUID("ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6");
// The characteristic of the remote service we are interested in.
static BLEUUID    charUUID("ebe0ccc1-7a0a-4b0c-8a1a-6ff2997da3a6");
// La caracteristica para cambiar el tiempo de intervalo de comunicaciones.
static BLEUUID charUUID_SetCommInterval("ebe0ccd8-7a0a-4b0c-8a1a-6ff2997da3a6");
static BLEUUID charUUID_SetNotify("00002902-0000-1000-8000-00805f9b34fb");

// Obtain a reference to the service we are after in the remote BLE server.
BLERemoteService* pRemoteService;
// Obtain a reference to the characteristic in the service of the remote BLE server.
BLERemoteCharacteristic* pRemoteCharacteristic_THB;
  
const uint8_t notificationOff[] = {0x0, 0x0};
const uint8_t notificationOn[] = {0x1, 0x0};
[... 328 lines cut ...]
14:15:42.324 -> T= 0xd3 0x8 : H= 0x34 : BatmV= 0xcd 0xb 
14:15:42.324 -> temp= 22.59 ºC : humidity= 52.00 : bat= 3.021 v : TempRocio= 12.26 ºC 
14:15:42.324 -> Notifications turned off ->esp_get_free_heap_size: 107184 
14:15:42.324 -> 
14:15:48.810 -> T= 0xd3 0x8 : H= 0x34 : BatmV= 0xcd 0xb 
14:15:48.810 -> temp= 22.59 ºC : humidity= 52.00 : bat= 3.021 v : TempRocio= 12.26 ºC 
14:15:48.810 -> Notifications turned off ->esp_get_free_heap_size: 107108 
14:15:48.844 -> 
14:15:50.650 -> 
14:15:50.650 -> Stack smashing protect failure!
14:15:50.650 -> 
14:15:50.650 -> Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
14:15:50.650 -> Debug exception reason: BREAK instr 
14:15:50.650 -> Core 1 register dump:
14:15:50.650 -> PC      : 0x00000000  PS      : 0x00000000  A0      : 0x00000000  A1      : 0x800d282a  
14:15:50.650 -> A2      : 0x3ffc8730  A3      : 0x01c9d2ae  A4      : 0x28ddf108  A5      : 0x00000004  
14:15:50.683 -> A6      : 0x00000000  A7      : 0x00000000  A8      : 0x3ffc6860  A9      : 0x800d84d0  
14:15:50.683 -> A10     : 0x3ffc8750  A11     : 0x28ddf108  A12     : 0x00000000  A13     : 0x00000001  
14:15:50.683 -> A14     : 0x00000000  A15     : 0x3ffc86ec  SAR     : 0x3ffc86ec  EXCCAUSE: 0x00000001  
14:15:50.683 -> EXCVADDR: 0xffffffe0  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
14:15:50.717 -> 
14:15:50.717 -> Backtrace: 0x00000000:0x800d282a
14:15:50.717 -> 
14:15:50.717 -> Rebooting...
```

chegewara, 2020-03-27:
You are not doing anything wrong. Default task in arduino is looperTask, which is created by arduino implementation. It has 8kB stack, which seems to be too few in your case, thats why you get this error:
`Stack smashing protect failure!`

You can create new task, let say at the very beginning of setup and use it instead of loop, in the next line you could just destroy default task to reclaim memory from task stack with `vTaskDelete(NULL)`. Since now setup and loop no longer exist.

jaggil, 2020-03-29:
Thank you for your suggestion   @chegewara.
Before trying to create another task, I thought about monitoring the changes taking place in the stack.
This is why I have introduced the "uxTaskGetStackHighWaterMark (NULL)" function before and after the BLE write commands.
As you can see, the unused stack size is 6068, which seems quite large.
The problem persists, but now the message is associated with "core 0" which surprises me.
Does the BLE library run on "core 0"?
Now we have some information in the Exception Decoder in the ArduinoIDE about "bta_gattc_"
It appears that there are some stored BLE write commands that are pending execution.
Is there an easy way in the BLE library to check that the write function has been fully executed?

what's your opinion about it? 
Thanks in advance
### Sketch: 
```cpp

void loop() 
{
  //currentMillis = millis();
  currentMillis = micros();  
  looptime = currentMillis;
  UBaseType_t uxHighWaterMark;

  if (!connectionSuccessful){ Serial.println("Reconectar...."); connectSensor(); Serial.println("Reconectado y registrado"); }
  else
  {
    if(Nueva_Notificacion==true)
    {
      Nueva_Notificacion=false; 
      uxHighWaterMark = uxTaskGetStackHighWaterMark( NULL);Serial.println( uxHighWaterMark);
      pRemoteCharacteristic_THB->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t*)notificationOff, 2, true);    
      uxHighWaterMark = uxTaskGetStackHighWaterMark( NULL );Serial.println( uxHighWaterMark);
      Serial.printf("temp= %.2f ºC : humidity= %.2f \% : bat= %.3f v : TempRocio= %.2f ºC \n", temp, humi, bat, dewPointC(temp, humi));
      Serial.print("Notifications turned off ->"); Serial.printf("esp_get_free_heap_size: %d \n\n", esp_get_free_heap_size());
    }
    else
    if ( ((currentMillis - previousMillis) > interval) ) 
    {
      uxHighWaterMark = uxTaskGetStackHighWaterMark( NULL );
      Serial.printf("Notifications turned on ->esp_get_free_heap_size: %d : %d \n", esp_get_free_heap_size(), uxHighWaterMark); 
      pRemoteCharacteristic_THB->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t*)notificationOn, 2, true);
      uxHighWaterMark = uxTaskGetStackHighWaterMark( NULL );Serial.println( uxHighWaterMark);
      previousMillis = currentMillis;
      Serial.println( micros());    
    }     
  }
  
  //looptime = millis() - looptime;
  looptime = micros() - looptime;   
  if (looptime > looptimeMax){ looptimeMax= looptime;Serial.printf("LooptimeMax: %d \n", looptimeMax);}
  if (looptime < looptimeMin){ looptimeMin= looptime;Serial.printf("LooptimeMin: %d \n", looptimeMin);}
  //if (looptime >= 1000){ Serial.printf("Looptime > 1s: %d \n", looptime);}  //Indica si el "loop" dura mas de 1 s.
  if (looptime >= 1000000){ Serial.printf("Looptime > 1s: %d \n", looptime);}  //Indica si el "loop" dura mas de 1 s.  
}
```

### Debug Messages:
```
01:11:52.672 -> 
01:12:47.180 -> Notifications turned on ->esp_get_free_heap_size: 108292 : 6068 
01:12:47.180 -> 6068
01:12:47.180 -> 1562100982
01:12:48.662 -> T= 0xca 0x7 : H= 0x45 : BatmV= 0xcf 0xb 
01:12:48.662 -> 6068
01:12:48.662 -> 6068
01:12:48.662 -> temp= 19.94 ºC : humidity= 69.00 : bat= 3.023 v : TempRocio= 14.10 ºC 
01:12:48.662 -> Notifications turned off ->esp_get_free_heap_size: 107456 
01:12:48.696 -> 
01:12:56.678 -> T= 0xc9 0x7 : H= 0x45 : BatmV= 0xcf 0xb 
01:12:56.678 -> 6068
01:12:56.678 -> 6068
[... 43 lines cut ...]
0x4013b899: bta_gattc_pop_command_to_send at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_act.c line 1519
0x4013cac7: bta_gattc_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_act.c line 1170
0x4013a102: bta_gattc_sm_execute at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_main.c line 288
0x4013b899: bta_gattc_pop_command_to_send at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_act.c line 1519
0x4013cac7: bta_gattc_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_act.c line 1170
0x4013a102: bta_gattc_sm_execute at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_main.c line 288
0x4013b899: bta_gattc_pop_command_to_send at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_act.c line 1519
0x4013cac7: bta_gattc_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_act.c line 1170
0x4013a102: bta_gattc_sm_execute at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_main.c line 288
0x4013b899: bta_gattc_pop_command_to_send at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_act.c line 1519
0x4013cac7: bta_gattc_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_act.c line 1170
[... cut ...]
0x4013b899: bta_gattc_pop_command_to_send at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_act.c line 1519
0x4013cac7: bta_gattc_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_act.c line 1170
0x4013a102: bta_gattc_sm_execute at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_main.c line 288
```

chegewara, 2020-03-29:
> Thank you for your suggestion @chegewara.
> Before trying to create another task, I thought about monitoring the changes taking place in the stack.
> This is why I have introduced the [...]

That was really good idea, because i just assumed the problem is in main loop, but in fact stack smashing could be triggered by other task.

From backtrace i can tell that something bad and unusual i think happen in esp-idf bluetooth stack:
```
Decoding stack results
0x4009165c: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x4009188d: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x4008fa5a: xTaskRemoveFromEventList at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 3091
0x4008dcbd: xQueueGiveFromISR at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 1367
0x4008500d: semphr_give_from_isr_wrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bt.c line 528
0x4011c3c9: GATTC_Write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/gatt/gatt_api.c line 1005
```
If its reproducible then i suggest to report it on esp-idf repo with code that can be used to reproduce it if possible.
In fact maybe @me-no-dev can pass it to bluetooth team and ask for opinion.

jaggil, 2020-03-29:
I think it is reproducible, although it is random and sometimes you have to wait hours.
On the other hand, to be sure to reproduce it 100%, perhaps you need to have the Xiaomi-Mijia-LYWSD03MMC device.

hampshire-al, 2020-04-11:
I've run your Arduino sketch on an ESP32 and it has run without the error continuously for about 9 hours.

I attach the output log (I've slightly edited your sketch to use English) and the only issue was a disconnection when I moved the LYWSD03MMC, which resulted in losing the connection briefly.

[20200411 MijiaTemp2_Log.txt](https://github.com/espressif/arduino-esp32/files/4465377/20200411.MijiaTemp2_Log.txt)

I don't know if this helps, but I could not reproduce the error over this period of 9 hours.

jaggil, 2020-05-27:
Hello @hampshire-al ,
First of all, thanks a lot for testing my code.
Sorry I haven't answered you before, but I've had the project abandoned for several weeks.
Yesterday I recompiled my code and after 16 hours the problem remains the same.
Please, could you try it for a longer time, for example one or two days?
On the other hand, I would like to know how you have integrated the LYWSD03MMC sensors in your system, and how you access the information.
I would also like to know how fast is the consumption of the sensor battery.

Thanks in advance, regards

hampshire-al, 2020-06-23:
Hi @jaggil, I've finally picked this up again briefly...  
I've moved the code to PlatformIO in Visual Studio Code, which requires a couple of small tweaks, but gives a slightly different log, which may help?

I attach the log:
[20200623 Log file.txt](https://github.com/espressif/arduino-esp32/files/4822178/20200623.Log.file.txt)

It seems that while the LYWSD03MMC was further away it had a few issues including a Stack Smashing failure, but it all settled down when the ESP32 and the LYWSD03MMC moved closer together.  This may have just been a coincidence.

I am also attaching a text file with my PlatformIO version of the code:
[20200623 Mijia Temp - VSC Platformio Code.txt](https://github.com/espressif/arduino-esp32/files/4822181/20200623.Mijia.Temp.-.VSC.Platformio.Code.txt)

Sorry, haven't had time to run it longer, but maybe this helps a little?

closed 2020-09-05 by stale[bot]

tuxmartin, 2020-10-18:
@jaggil hi, do you have any progress?

jaggil, 2020-10-28:
Hello, I have not had any progress yet, the program crashes randomly after several hours,
I don't know what to do to move forward in any way.

I have stopped my project to integrate several LYWSD03MMC in Blynk

covert-, 2022-07-22:
To help others. I resolved this issue by using the ArduinoBLE lib instead of the ESP32_BLE one I was using.