espressif/arduino-esp32 #3857: Installation on Mac fails: command line tools are already installed, use "Software Update" to install updates

andreasloe, 2020-03-27:
I am trying to install the code for ESP32 on a Mac.Installation of packages fails.

### Hardware:
Board:							ESP32 Cam Module
Core Installation/update date:			unkown
IDE name:							Arduino IDE
Computer OS:                                            Mac OSX 10.14.6

### Description:
During installation after "git clone https://github.com/espressif/arduino-esp32.git esp32" I get the error message
'xcrun: error: active developer path ("/Developer/Xcode.app/Contents/Developer") does not exist
'Use `sudo xcode-select --switch path/to/Xcode.app` to specify the Xcode that you wish to use for command line developer tools, or use `xcode-select --install` to install the standalone command line developer tools.
'See `man xcode-select` for more details.

After "(sudo) xcode-select --install" I get the error message
'command line tools are already installed, use "Software Update" to install updates

lbernstone, 2020-03-27:
Are you not able to install from [board manager](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md)?

andreasloe, 2020-03-27:
No, it did not work. I do not find the ESP32 when searching for the available packages.

lbernstone, 2020-03-27:
Make sure you have the correct address in the additional boards.  Paste it into a browser to be sure you can get to the URL.

andreasloe, 2020-03-28:
lbernstone: There seems to be something wrong with my installation. If I put the location https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json into my arduino IDE (Version 1.8.12) then searching for ESP32 does not yield a result (see screenshot). I suspect my installation is the villain but do not know where to look.

[image]

andreasloe, 2020-03-28:
I just removed the entire arduino installation and started all over again. The same result: I do not find the ESP32 library.

sansillusion, 2020-03-28:
It's not in libraries but in board manager that you will find esp32

andreasloe, 2020-03-28:
Thank you very much, that solved it!

closed 2020-03-28 by andreasloe