espressif/arduino-esp32 #3858: Would you please make a arduino lib to support Hilink

myie9, 2020-03-28:
ESP8266WiFi library of arduino is very good.
   I used it for 2 years.
   The smartconfig function of it is very nice.

   Would you please make a arduino lib to support Hilink of config network ?

  Hilink is supported by Huawei Wifi router. Users  can config network by just press one button on the router.

Here is some reference:
hilink api doc:
https://developer.huawei.com/consumer/cn/doc/development/smarthome-References/3090501

code:
https://github.com/myie9/esp8266-hilink

lbernstone, 2020-03-28:
ESP32 supports the industry standard WPS for one touch config:
https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFi/examples/WPS

myie9, 2020-03-29:
thank u very much

发自我的iPhone

 

 https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFi/examples/WPS

closed 2020-06-11 by stale[bot]