espressif/arduino-esp32 #3859: Problem with GetChipID Code

cweinhofer, 2020-03-28:
I have a question about the example code:
[GetChipID.ino](https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/ChipID/GetChipID/GetChipID.ino)

Is this code correct? It says "The chip ID is essentially its MAC address" but it outputs the MAC address in reverse. This seems confusing at best and inaccurate at worst.

Also, if this is an attempt to mimic ESP8266's chip ID, it's not very helpful since that one is a 32-bit integer (which can be used in things like switch-case) whereas this produces something with very little usability.

Seems like it would be more helpful for people if the code showed people how to get a 32-bit integer from EfuseMac.

Jeroen88, 2020-03-29:
I use: `Serial.printf("%012llx", ESP.getEfuseMac());` to output the chip ID to the Serial monitor. If you want to use the lower 4 bytes only, just assign it to a uint32_t variable: `uint32_t chipID = ESP.getEfuseMac();`

cweinhofer, 2020-03-30:
`uint32_t chipID = ESP.getEfuseMac();` would be a nice simple solution.
However, the value that getChipId returns on an ESP8266 is only the last 3 bytes of the MAC, whereas this would be 4. That doesn't make it unusable, but if the point of this sketch is to provide a way to mirror ESP8266's getChipId, then `uint32_t chipID = ESP.getEfuseMac();` wouldn't do that.
Also, correct me if I'm wrong, but since getEfuseMac returns an LSB value, wouldn't `uint32_t chipID = ESP.getEfuseMac();` end up giving you the first 4 bytes of the MAC address (the OUI + 1, which would be less likely to be unique) -- and in reverse order at that.

Jeroen88, 2020-04-02:
I do not see your point: ESP8266 `ESP.getChipId()` returns an unique ID for the MCU you are using. So does `ESP.getEfuseMac()`. Both functions are equivalent and can be used as a unique ID.
If you need a MAC, you can not use ESP8266 `ESP.getChipId()` because it returns only 4 bytes and a MAC is 6 bytes, so both functions are not equivalent in this sense. 
However, you can use `uint8_t *MAC = ESP.getEfuseMac()` and read the bytes in any order, e.g. `Serial.printf("%02x%02x%02x%02x%02x%02x\n", MAC[5], MAC[4], MAC[3], MAC[2], MAC[1], MAC[0]);`
Hope this is of any help!

cweinhofer, 2020-04-03:
Appreciate your comments. Sorry, I guess I should have been more clear. My point was not to ask how to get usable info out of ESP.getEfuseMac(), but rather to ask, "What's the point of the [GetChipID.ino](https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/ChipID/GetChipID/GetChipID.ino) example sketch? and Is it even correct?"

The name would seem to imply that the point was to provide an equivalent to ESP8266's ESP.getChipId(). But this sketch doesn't do that at all. And worse yet, what it does do would seem to confuse newbies more than help them. I know it did me at first.

It seems like this would be most helpful if this sketch gave people an exact equivalent to ESP.getChipId() (i.e. the NIC part of the MAC address as a 32-bit integer) -- particularly for those who are using ESP32s and ESP8266s in a mixed environment and need a unique ID that works with both.

I'd be happy to submit the changes. I just thought I'd see what others thought of this sketch before I did.

cweinhofer, 2020-06-07:
The code has been rewritten and a pull request created.

https://github.com/espressif/arduino-esp32/pull/4070

cweinhofer, 2020-09-08:
Not sure if / when my pull request for this might get approved, so I'll post the code here for anyone who needs it.

```
/* The true ESP32 chip ID is essentially its MAC address.
This sketch provides an alternate chip ID that matches 
the output of the ESP.getChipId() function on ESP8266 
(i.e. a 32-bit integer matching the last 3 bytes of 
the MAC address. This is less unique than the 
MAC address chip ID, but is helpful when you need 
an identifier that can be no more than a 32-bit integer 
(like for switch...case).
created 2020-06-07 by cweinhofer
with help from Cicicok */
	
uint32_t chipId = 0;

void setup() {
	Serial.begin(115200);
}

void loop() {
	for(int i=0; i<17; i=i+8) {
	  chipId |= ((ESP.getEfuseMac() >> (40 - i)) & 0xff) << i;
	}

	Serial.print("ESP8266-Style Chip ID: ");
	Serial.println(chipId);
	Serial.println();
	
	delay(5000);

}
```

Electro-boy, 2020-10-16:
Try this ::: For Getting the full CHIP ID --- ESP 32
Unfortunately : ESP_getChipId() doesnt return the full chip ID.

`uint32_t low = ESP.getEfuseMac() & 0xFFFFFFFF; 
uint32_t high = ( ESP.getEfuseMac() >> 32 ) % 0xFFFFFFFF;

 Serial.print(low);
 Serial.print(" : "); 
 Serial.println(high);  `

cweinhofer, 2020-11-12:
Thanks. But the solution I'm suggesting in this case is less about getting the true chip ID -- especially since it's exactly the same as the MAC address -- and more about producing a (relatively) unique 32-bit integer that can be used as an ID number for the SOC.