espressif/arduino-esp32 #3862: time.h functions

peppeg85, 2020-03-29:
hello, this is not a truly issue, i found the simple time example. 

https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Time/SimpleTime/SimpleTime.ino

reading many guides over the internet i fount many examples that uses ntpclient library, than i found this great example that uses just wifi.h and time.h libraries, and it's great! just i would like to know, where can i find this library to understand what functions i can use? 

for example the function getLocalTime, i can't find it inside all the time.h files inside the sdk . is there a guide? sorry and thank you in advance

bertmelis, 2020-03-29:
For ESP8266 the (s)ntp functions are included in the LwIP lib.

lbernstone, 2020-03-29:
The header is here:
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/newlib/time.h
You will see it is mostly external functions provided by the compiler toolchain, which is a pretty standard gcc 5.2.
The arduino core time functions come from https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-time.c, which uses the [esp-idf sntp app](https://github.com/espressif/esp-idf/blob/release/v3.3/components/lwip/include/apps/sntp/sntp.h), which is in turn a piece of LwIP.

peppeg85, 2020-03-29:
ok! tanks for fast response, i will check these files. thank you!

closed 2020-06-11 by stale[bot]