espressif/arduino-esp32 #3863: NimBLE library ready and looking for testers.

h2zero, 2020-03-29:
I just finished refactoring the BLE library to use the NimBLE stack and I thought some of you would like to test it out. 

The resource usage is significantly less than bluedroid, both ram and flash use are nearly half as much.

Compiles in Arduino IDE and platform IO, tested with v1.0.2 and v1.0.4 of arduino-esp32, and I suspect it will work will all.

If interested you can [check it out here](https://github.com/h2zero/NimBLE-Arduino). 

Warning: a few changes to your apps may need to be made, but should be minimal. Check the examples for details.

chegewara, 2020-03-29:
Nice work. If you want some help just let me know.

h2zero, 2020-03-29:
Thanks, I'm open to any PR's or feature requests / ideas. At the moment I'm bug squashing and code cleaning.

wakwak-koba, 2020-03-31:
I mainly use BLEClient.
It is very rude to say this, but I think it is better than the original one !

When reconnecting after disconnection, it is more convenient for NimBLEClient to retain the previous connection parameters, NimBLEDevice :: createClient (NimBLEAdvertisedDevice *) method has been added.
What do you think ?

In the case of connecting multiple devices and receiving a Notify , a Panic error occurred at 3 units, etc., and I am pursuing the cause.
(this may be my implementation is bad...)

Once we have a solution, I would like to PR it.

https://github.com/wakwak-koba/NimBLE-Arduino

chegewara, 2020-03-31:
> It is very rude to say this, but I think it is better than the original one !

Americas government is trying to change this, but lucky to us we are still living in free world and as long as you dont insult other users you can think and say whatever you want.

Its good that users have more options and dont have to rely on just 1 library.

wakwak-koba, 2020-03-31:
Sorry, my choice of words was not good...

h2zero, 2020-03-31:
That's the great thing about open source, it's also open opinion, without that it wouldn't work.

@wakwak-koba I'm glad you like it, if you want to create an issue at my repo I would like to see the nature of the panic you are getting.

closed 2020-06-13 by stale[bot]

kind3r, 2021-01-26:
Great job !

The client part seems to work like a charm for my use case (building a light [BLE to WiFi gateway](https://github.com/kind3r/esp32-ble-gateway)) and saved me about 70K of RAM and 400K of flash storage. Also fragmentation seems lower. Whish I had discovered this before implementing everything using the BLE lib from this repo and having to find workarounds and implement patches for the various issues it has. But at least hopefully those would help someone if they get merged.

Now I have room to implement an UI and even OTA :D

h2zero, 2021-01-26:
Thanks, I'm glad you found it!

I have the same use case in one of my projects and the original library was too resource heavy, that was the motivation for creating this (also a good opportunity to fix a lot of things). 

I still have plenty of room with a webserver UI, telnet and OTA :).

I don't think they want to add new 3rd party libraries to the core, besides updates and fixes are quicker in a focused repo. 
Spread the word to others, the more feedback the better.

kind3r, 2021-01-26:
The current BLE implementation was also a 3rd party repo, but indeed, it would be more confortable for you as a maintainer to manage your own repo, as long as development is still ongoing. But at some point life and other projects will get in the way so it will become more and more difficult to keep it up-to-date, and then it will make sense to have it in a shared repo.

h2zero, 2021-01-26:
>But at some point life and other projects will get in the way so it will become more and more difficult to keep it up-to-date, and then it will make sense to have it in a shared repo.

Good point, I'm not opposed to the idea, I just think they might be.