espressif/arduino-esp32 #3866: How to #define correctly standard ArduinoIDE macros for (arduino-esp32 as component) for using third-party Arduino libraries

Sterpa, 2020-03-31:
I use [ESP-IDF4.0](https://github.com/espressif/esp-idf/tree/release/v4.0) + [arduino-esp32-4.0](https://github.com/espressif/arduino-esp32/tree/idf-release/v4.0) as component (and this pair works very well in VSCode with CMake)
Many perfect third-party libraries for ArduinoIDE use standard Arduino macros in their header files, depending on which the library is configured during compilation.
**Example:**
[Streaming.h](https://github.com/janelia-arduino/Streaming/blob/master/src/Streaming.h) use **ARDUINO** macros
```
#if defined(ARDUINO) && ARDUINO >= 100
#include "Arduino.h"
#else
#include "WProgram.h"
#endif
```
[M5Stack.h](https://github.com/m5stack/M5Stack/blob/master/src/M5Stack.h) use **ESP32**, **M5STACK_MPU6886**, **M5STACK_MPU9250** and **M5STACK_MPU6050** macroses
```
#if defined(ESP32)
...
    #if defined(M5STACK_MPU6886) || defined(M5STACK_MPU9250) || defined(M5STACK_MPU6050)
        #include "utility/MPU6886.h"
        #elif defined M5STACK_200Q
        #include "utility/SH200Q.h"
    #endif
...
#endif
```
Some libraries use **ARDUINO_ARCH_ESP32** macros and so-on.
These macros are usually defined by the Arduino IDE, but they are undefined if you work with ESP-IDF and arduino-esp32 as a component in other IDEs.

So, of course, we don’t want to edit the contents of the library itself to define these macros.
What is the best way to define these macros when using arduino-esp32 as component?
- Configure somewhere in the menuconfig ESP-IDF, if possible?
- Maybe we can place a header file with these macros in the arduino-esp32 framework itself?
- Or maybe in the main.cpp file we can somehow declare these macros before including libraries?
- Make some kind of header file with these definitions, and then insert it at the beginning of each library, similar to C++ (stdafx.h or pch.h - precompiled header files option)?

**e.g. header.h**
```
#define ARDUINO 181
#define ESP32
#define ARDUINO_ARCH_ESP32
#define M5STACK_MPU9250
```
I would be grateful for the most correct vector for solving the problem.

rdvojmoc, 2020-05-03:
@Sterpa, how do you solve this issue at the moment? I have a couple of library in my project that checks for ARDUINO macro, but cannot find it?

lbernstone, 2020-05-03:
This is by design.  These defines are added by ArduinoIDE (through platform.txt), so they can be variables.  Adding them in the library compile would negate that.  So, if you want them in an IDF compile, you need to add them as compiler options (-D).  It might be possible to add them into a file unused in ArduinoIDE, e.g. component.mk, but that could have unpredictable downstream effects for PlatformIO and Eclipse.

rdvojmoc, 2020-05-03:
@lbernstone, can you provide an example of that?

lbernstone, 2020-05-03:
I would if I had the interest.  Instead, I will provide a recommendation.  Use VSCode + platformIO, where all you will need to do is add a define to your platformio.ini file.

rdvojmoc, 2020-05-04:
I'm using VSCode+platformio. ARDUINO define added to build_flags in `platform.ini`
```
build_flags = 
  -D ARDUINO=250
```

Now it compiles without problems. Thanks for the hint @lbernstone .

closed 2020-07-17 by stale[bot]