espressif/arduino-esp32 #3867: Esp32 to stm8 i2c communication

nikhil8333, 2020-04-01:
Hi , 
I was trying to implement a i2c protocol with esp 32 as master and multiple stm8's as slaves. Code was implimented using adruino ide and wire.h lib was used. Initial address scanning was successful but unable to communicate with slave. Can you plese help me

stickbreaker, 2020-04-01:
@nikhil8333 I don't have any experience with your STM micro.  

this is how I would clean up your ESP code:
```c++
#include <Wire.h>

const int SCLpin = 22;
const int SDApin = 21; // For ESP32 Lolin D32

void setup()
{
  Serial.begin(115200);
  Serial.println("I2C Scanner");
  Serial.println("SDA Pin = "+String(SDA));
  Serial.println("SCL Pin = "+String(SCL));
  
  Wire.begin(SDApin, SCLpin);
  Wire.setClock(100000) ; // set i2c frequency to 100k, needs at least 3.3k pullups to 3.3v
}

uint8_t x = 0x00;

void loop()
{
   byte error;
   Serial.println("testing address 0x24");

   Wire.beginTransmission(24);
   Wire.write(x);
   error =Wire.endTransmission();
  
   if (error != 0) {
	   Serial.printf("I2C error =%d(%s)\n",error, Wire.getErrorText(error));
   }
   else Serial.printf("Success!\n");

   delay(2000);
  
}
```

Here is a link to an I2C scanner that will display all slave I2C devices on the bus:
[i2c scanner](https://github.com/stickbreaker/arduino-esp32/blob/master/libraries/Wire/examples/i2c_scan/i2c_scan.ino)

Chuck.

nikhil8333, 2020-04-02:
thanks for the help chuck, still no luck with the code, getting error 1 from your code same as mine

> Here is a link to an I2C scanner that will display all slave I2C devices on the bus:
> [i2c scanner](https://github.com/stickbreaker/arduino-esp32/blob/master/libraries/Wire/examples/i2c_scan/i2c_scan.ino)
already adopted the same code for scanning, and my scanning has no problem. I was able to retrieve all the slave address. only my write and read commands are giving me errors. is there any other way around.

Thanks in advance.

stickbreaker, 2020-04-02:
@nikhil8333 recompile with `CORE_DEBUG_LEVEL` set to `INFO`, the I2c subsection will log any problems.

`I2C_ERROR_DEV` is returned if the hardware was not successfully configured, or the `Wire()` object is corrupted.

Additionally you can enable the i2c Debug buffer by uncommenting line 45 
```
//#define ENABLE_I2C_DEBUG_BUFFER
```
in `cores\esp32\esp32_hal_i2c.c`
here https://github.com/espressif/arduino-esp32/blob/0f772270fbb32099cad72053755d6b4c7e8e1f82/cores/esp32/esp32-hal-i2c.c#L45

closed 2020-06-15 by stale[bot]