espressif/arduino-esp32 #3868: ESP32 MDNS working only on one computer

PC-maniak, 2020-04-01:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			latest
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
MDNS works only on my computer .. on the other (local area network) I have to give an ip address ... and on my Android phone does not work - only ip ..

### Sketch: 
```cpp

//Change the code below by your sketch
#include <WiFi.h>
#include <WiFiClientSecure.h>
#include <WebServer.h>
#include <ESPmDNS.h>
#include <Update.h>
#include <SPIFFS.h>

const char* host = "nas-wifi";
const char* ssid = "****";
const char* password = "****";

WebServer server(80);

void setup() {
Serial.begin(115200);
 
  // Start SPIFFS Files System
  SPIFFS.begin();

  // Connect to WiFi network
  // Wait for connection
   WiFi.begin ( ssid, password );
  while ( WiFi.status() != WL_CONNECTED ) {
    delay ( 500 ); Serial.print ( "." );
  }
  // WIFI OK
  Serial.println ( "" ); 
  Serial.print ( "Connected to " ); Serial.println ( ssid );
  Serial.print ( "IP address: " ); Serial.println ( WiFi.localIP() );

  //use mdns for host name resolution*/
  if (!MDNS.begin(host)) { 
    Serial.println("Error setting up MDNS responder!");
    while (1) {
      delay(1000);
    }
  }
  Serial.println((String)"http://" + host);

 server.onNotFound([]() {                              // If the client requests any URI
    if (!handleFileRead(server.uri()))                  // send it if it exists
      server.send(404, "text/plain", "404: Not Found"); // otherwise, respond with a 404 (Not Found) error
  });
  
  server.begin();
  Serial.println("HTTP server started");
}

void loop() {
server.handleClient();
}

// MORE SERVER CODE HERE ...

```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

CelliesProjects, 2020-04-02:
You will need Avahi, Bonjour or any kind of mDNS support on your phone or any pc connecting to your esp. Which one exactly depends on your OS.

See [https://en.wikipedia.org/wiki/Multicast_DNS](https://en.wikipedia.org/wiki/Multicast_DNS)

PC-maniak, 2020-04-02:
hi ok android mobile i understand .. but i have 3x windows 10 with chrome browser and i can only http: //nas-wifi/ connect only from computer where i wrote arduino program

that's what I don't really understand as much as possible .. all computers are on the same local area network

there is a more reliable way to get DNS work on ESP32 ?

stickbreaker, 2020-04-02:
@PC-maniak you could also be having problems with a VLan config on your network.  Many corporate networks are configured such that ports on the switch are defined as clients or servers, clients only "see" servers, and "servers" only see clients.  This limits "client" to "client" infections.  

Chuck.

Pallanez, 2020-04-04:
I'd recommend to use the DNSServer library, it works flawlessly. MDNs seems to be cumbersome in some situations.

PC-maniak, 2020-04-05:
@Pallanez thanks I'll look at it

closed 2020-06-18 by stale[bot]