espressif/arduino-esp32 #3869: retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown

ElecDev, 2020-04-02:
Hi,

I'm facing a problem using BLE ESP32 whether it be Arduino or PlatformIo.
The problem is sometimes when trying to connect with my remote device (un EFR32BG), i have this type eff message and the program stuck, i had to reset my ESP32 to exit from this state.
The messages are as below: 
[E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_al
2020/04/02 19:31:57.196: l_descr: Unknown
[E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[E][BLERemoteService.cpp:185] retrieveCharacteristics(): esp_ble_gattc_get_all_char: Unknown

Do you have any idea what that is going on? sometimes i can make connection with no problem but sometimes not.
Best regards,

chegewara, 2020-04-02:
I think this is nothing unusual, some characteristics dont have descriptors:
```
[E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
```
Probably this should be WARN log, not ERR.

Why you cant connect is hard to say. Weak signal, or other. 
If you are trying to retrieve characteristics and descriptors you are definitely connected, unless in meantime device got disconnected.

ElecDev, 2020-04-03:
Thank you chegewara for your reply.
> If you are trying to retrieve characteristics and descriptors you are definitely connected, unless in meantime device got disconnected.
I can see that my remote BLE device still connected but ESP32 stuck to finish processing to get all BLE informations. The signal is very strong. i Get this when i doesn't work : 
```
 - Found our service
[E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[E][BLERemoteService.cpp:185] retrieveCharacteristics(): esp_ble_gattc_get_all_char: Unknown
Characteristics : 6
```
I have 14 Characteristics, the ESP prints only 6 in this case and stays in this state.
 
Best regards,

chegewara, 2020-04-03:
Still there is not enough info and/or logs.

1) If you can provide more logs after that it would be great. 
2) Can you confirm with nRF connect on android smartphone that device has 14 characteristics?
3) EFR32BG is only BLE chip, can you confirm with nRF connect there is 14 characteristics?
4) you can try to add custom GATTC and GAP event handlers
https://github.com/nkolban/esp32-snippets/issues/757#issuecomment-447953252

ElecDev, 2020-04-06:
Hi chegewara, 
1) here more log as asked 
`[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0xff (), length: 29, data: 060001092002b17566d909f72a9dae7df42ac8592720f8e5d2c016acc3
[D][BLEAdvertisedDevice.cpp:395] setManufacturerData(): - manufacturer data: 060001092002b17566d909f72a9dae7df42ac8592720f8e5d2c016acc3
Try to connect.
Create of timer
Start of timer
[D][BLEClient.cpp:458] handleGAPEvent(): BLEClient ... handling GAP event!
2020/04/06 17:58:34.444: Received Address00:0b:57:40:a4:ef
Device address :00:0b:57:40:a4:ef
Forming a connection to :00:0b:57:40:a4:ef
- client Created
[I][BLEDevice.cpp:593] addPeerDevice(): add conn_id: 1, GATT role: client
[D][FreeRTOS.cpp:189] take(): Semaphore taking: name: RegEvt (0x3fffbe1c), owner: <N/A> for connect
[D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: RegEvt (0x3fffbe1c), owner: connect
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClie
2020/04/06 17:58:34.458: nt.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][FreeRTOS.cpp:189] take(): Semaphore taking: name: OpenEvt (0x3fffc6d4), owner: <N/A> for connect
[D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: OpenEvt (0x3fffc6d4), owner: connect
2020/04/06 17:58:35.866: [D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:577] updatePeerDevice(): update conn_id: 4, GATT role: client
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
 - Connected to server
[D][FreeRTOS.cpp:189] take(): Semaphore taking: name: SearchCmplEvt (0x3fffc734), owner: <N/A> for getServices
[D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: SearchCmplEvt (0x3fffc734), owner: getServices
2020/04/06 17:58:36.738: [D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][
2020/04/06 17:58:36.757: BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
 - Fou
2020/04/06 17:58:36.852: nd our service
[D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLERemoteService.cpp:193] retrieveCharacteristics(): Found a characteristic: Handle: 26, UUID: 04741c6d-db74-4ac6-bfc0-d5754a681fce
[D][BLERemoteCharacteristic.cpp:280] retrieveDescriptors(): Found a descriptor: Handle: 27, UUID: 00002902-0000-1000-8000-00805f9b34fb
[D][BLERemoteService.cpp:193] retrieveCharacteristics(): Found a characteristic: Handle: 29, UUID: c66162f8-e074-4a99-813a-86e93c0ef287
[E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[D][BLERemoteService.cpp:193] retrieveCharacteristics(): Found a characteristic: Handle: 31, UUID: fe1809ad-f0ba-43a5-825d-89896ffa8f62
[E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[D][BLERemoteService.cpp:193] retrieveCharacteristics(): Found a characteristic: Handle: 33, UUID: e2398041-d20c-4c2a-bf62-65a80a020d09
[E][BLERemoteCharacteristi
2020/04/06 17:58:36.868: c.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[D][BLERemoteService.cpp:193] retrieveCharacteristics(): Found a characteristic: Handle: 35, UUID: 5ee5982c-27bf-4b8d-ad60-b470ad41c6d4
[E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[D][BLERemoteService.cpp:193] retrieveCharacteristics(): Found a characteristic: Handle: 37, UUID: 1ed4d43e-45b8-4a5a-81fa-ff101206a0e1
[E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_
2020/04/06 17:58:36.960: gattc_get_all_descr: Unknown
[D][BLERemoteService.cpp:193] retrieveCharacteristics(): Found a characteristic: Handle: 39, UUID: de471356-988d-4e43-93e5-b4082e836ca4
[E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[... cut ...] e_gattc_get_all_char: Unknown
Characteristics : 7
[D][FreeRTOS.cpp:189] take(): Semaphore taking: name: ReadCharEvt (0x3fffecf0), owner: <N/A> for readValue
[D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: ReadCharEvt (0x3fffecf0), owner: readValue
`
2) Yes i see 14 characteristics with  nRF connect on android smartphone of the service i'm handling, the device has more than  14 characteristics but i'm only manipulating the desired service who contains theses characteristics .

Best regards,

chegewara, 2020-04-06:
Ok, i will try to find some time to write test code with 14 characteristics, then test with client code.

ElecDev, 2020-04-07:
Ok thank you chegewara,
Just a question, suppose we have a service with x characteristics, if we don't handle them all, will that be a problem?
Best regards,

chegewara, 2020-04-07:
No, you dont need to handle all characteristics. Peripheral can have some functions that you are not interested, so you just ignore it.

ElecDev, 2020-04-07:
Ok, thanks,

I noticed that the problem appears when i have  this **retrieveCharacteristics(): esp_ble_gattc_get_all_char: Unknown**
Best regards,

closed 2020-06-20 by stale[bot]