espressif/arduino-esp32 #3870: Compiling Arduino-esp32 as library for PlatformIO

dougcfl, 2020-04-02:
Hi, I'm trying to rebuild the arduino-esp32 library to use in a PlatformIO project.  I need to change the values in menuconfig for my project and run on an esp32-solo chip.  I've tried the instructions in esp32-arduino-lib-builder on both windows and macOS.  If I use current versions of Arduino-esp32 and the ESP-IDF I get lots of errors with MOCKABLE_FUNCTION.
/Users/dougc/esp/esp32-arduino-lib-builder/components/arduino/libraries/AzureIoT/src/az_iot/c-utility/inc/azure_c_shared_utility/strings.h:36:1: error: expected primary-expression before ',' token
 **MOCKABLE_FUNCTION**(, int, STRING_replace, STRING_HANDLE, handle, char, target, char, replace);

I read about this and other build errors.  I tried to roll back the esp-idf to v3.3 and v4.0beta1.  I still get compile errors with each version trying to build the library (the errors are different with each version of the esp-idf).  Posts relating to this are a bit old and I'm looking for advice on how to rebuild the ardunio-esp32 library using the most current versions.

Right now, I'm on MacOS with esp-idf (master) and ardunio-esp (master).

I found binaries of updated library files here,
https://github.com/lbernstone/arduino-esp32-solo
but I'd like to be able to rebuild those library files.  Will someone describe the [latest] compatible versions of Arduino-esp32 and esp-idf, and provide a brief explanation on how to rebuild the Arduino-esp32 library for use in PlatformIO?  Thank you!

chegewara, 2020-04-02:
Try this in platformio.ini:
`platform=espressif32@1.11.2`

Thanks to @atanisoft

dougcfl, 2020-04-03:
Actually 
`platform=espressif32@1.11.0`
works in the platformio.ini file if you also use the library files from libernstone.  See here for the platformio discussion,
[https://community.platformio.org/t/esp32-solo-on-arduino-framework-setup/12805/2](url)

The help I'm requesting here is how to rebuild the arduino-esp32 libraries.  There seems to be tight coupling between the espidf and arduino-esp32 releases/commits.  There is also a dependency in the platformio package version.  There might be even more that I have yet to discover.  I'm asking if anyone knows the latest versions that all work together.  Thank you.

obertr0n, 2020-04-04:
I second this. I would also like to be able to compile idf and cherry pick some FreeRTOS libraries.
My goal is to have the tracing functionality enabled (https://www.freertos.org/uxTaskGetSystemState.html) and use this libraries with arduino.

closed 2020-06-18 by stale[bot]