espressif/arduino-esp32 #3871: Task watchdog got triggered. The following tasks did not reset the watchdog in time

ElecDev, 2020-04-03:
Hi,
i have an issue with my ESP32 concerning the watchdog. Sometimes i get this error and the ESP32 restarts. The message i get is this one:

```
[D][BLEAdvertisedDevice.cpp:247] pE (17446) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (17446) task_wdt:  - IDLE0 (CPU 0)
E (17446) task_wdt: Tasks currently running:
E (17446) task_wdt: CPU 0: Btc_task
E (17446) task_wdt: CPU 1: IDLE1
E (17446) task_wdt: Aborting.
abort() was called at PC 0x400e37d3 on core 0
```
How can i resolve this problem ?
I'm using the PlatformIo and arduino Framework.

Best regards,

lbernstone, 2020-04-03:
Please follow the [issue template](https://github.com/espressif/arduino-esp32/blob/master/docs/ISSUE_TEMPLATE.md).  It allows you to help us fix YOUR problem.  The task watchdog gets triggered when your code blocks the OS from switching tasks.

ElecDev, 2020-04-17:
Hi Ibernstone,

Sorry for the late response. Here the informations asked :
**Hardware :**
Board: TTGO T-Call, other ESP32 Dev Modules 
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash, PSRAM Disabled, Upload Speed=921600, Flash Frequency=80Mhz
Environment: Arduino, PlatformIO (framework-arduinoespressif32 3.10004.200129 (1.0.4))
Computer OS: Windows 

**Description:**

I’m implementing a simple ESP32 BLE device that do a BLE scan. In the advertised callback, I test if I find the UUID I’m looking for, when it’s found, I print it on the Serial port. The scan is done in continuous mode.
The problem I have that almost every hour my ESP32 restarts.

below a snippet of the code used:
`void initClient()
{

  //Starting Arduino BLE Client application
  BLEDevice::init("");
  // have detected a new device.  Specify that we want active scanning and start the
  pBLEScan = BLEDevice::getScan();
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks(), true);
  pBLEScan->setActiveScan(true);
 pBLEScan->start(50, scanCompleteCB);
}
`

**Debug Messages:**

> [D][BLEAdvertisedDevice.cpp:247] pE (17446) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (17446) task_wdt:  - IDLE0 (CPU 0)
E (17446) task_wdt: Tasks currently running:
E (17446) task_wdt: CPU 0: Btc_task
E (17446) task_wdt: CPU 1: IDLE1
E (17446) task_wdt: Aborting.
abort() was called at PC 0x400e37d3 on core 0

Best Regards,

lbernstone, 2020-04-17:
You need to add a delay() wherever your code is blocking for too long.  Since you did not post any code, I can only guess it is in your callback.

ElecDev, 2020-04-17:
Thanks, 

My code is very huge, i only give the part the causes the problem. I comment all part an let the BLE part. This problem produced in the callback indeed, because when it restarts, i have a serial print coming from this callback.
You suggest to a delay inside the callback ? it will penalize the BLE scanning, i will have less advertisement informations.
Best regards,

lbernstone, 2020-04-17:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/wdts.html#task-watchdog-timer

ElecDev, 2020-04-17:
Ok thank you, i'll try this and let you know.

Best regards,

closed 2020-07-01 by stale[bot]

moefear85, 2021-09-05:
From what I've come to understand... ESP-IDF creats a watchdog timer for idle tasks in addition to any tasks doing your work. IE, it creates two extra tasks IDLE0 & IDLE 1 (one for each core), whose sole purpose is to do nothing, ie idle. These idle tasks simply feed the watchdog whenever its respective core is idling. This means by design, esp-idf requires and forces that each core never be fully busy for too long. Whenever this happens (such as an infinite loop that doesn't call vTaskDelay or its equivalent anywhere), then it triggers the watchdog. IE you are defacto not allowed to fully consume the CPU and MUST have every task pause somewhere, to give the IDLE tasks some time to "run". The framework ensures this by giving them the lowest priority. IE, the reason is that the IDLE tasks are starving for "execution time", is because they have by default the lowest possible priority that the framework allows, so that any other task created will always have a higher priority, where if any task having a higher priority never takes a break (such as by calling vTaskDelay), then no task of lower priority (including the IDLE tasks), can run.

In other words there is no nice solution. Either disable the watchdog fully, or modify the framework to remove the IDLE tasks or somehow give them higher priorities (equal or greater than your task) while being careful however that they then don't block your work tasks, or simply don't allow any task to fully consume a core.

moefear85, 2021-09-05:
From what I've come to understand... ESP-IDF creats a watchdog timer for idle tasks in addition to any tasks doing your work. IE, it creates two extra tasks IDLE0 & IDLE 1 (one for each core), whose sole purpose is to do nothing, ie idle. These idle tasks simply feed the watchdog whenever its respective core is idling. This means by design, esp-idf requires and forces that each core never be fully busy for too long. Whenever this happens (such as an infinite loop that doesn't call vTaskDelay or its equivalent anywhere), then it triggers the watchdog, because that core is now fully busy, meaning it is no longer idling often enough, meaning the idle "tasks" can't run ie reset the watchdog ie effectively verify that the core has infact idled recently enough.. IE you are defacto not allowed to fully consume the CPU and MUST have every task pause somewhere, to give the IDLE tasks some time to "run". The framework ensures this by giving them the lowest priority. IE, the reason is that the IDLE tasks are starving for "execution time", is because they have by default the lowest possible priority that the framework allows, so that any other task created will always have a higher priority, where if any task having a higher priority never takes a break (such as by calling vTaskDelay), then no task of lower priority (including the IDLE tasks), can "run".

In other words there is no nice solution. Either disable the watchdog fully, or modify the framework to remove the IDLE tasks or somehow give them higher priorities (equal or greater than your task) while being careful however that they then don't block your work tasks (if they can do that), or simply don't allow any task to fully consume a core, so that it idles frequently enough for the watchdog to stay happy. Another alternative would be increasing the time to trigger the watchdog. I don't know how to do this, and I don't know what the max limit is.

Also: I'm not sure if the task, when it performs a blocking call (such as a blocking IO read that takes a long time) results in idling thus gives the IDLE tasks time to run, or whether it counts as "execution" thus further starving the IDLE tasks and triggering the watchdog. In the former case, you wouldn't need to add any delays, unless the calls end up not blocking long enough (then you'll have to artificially idle by calling vTaskDelay). In the latter it means adding a delay wouldn't help, you'll have to set time limits on the blocking call to force it to return before the watchdog triggers, and even then, unless you do something else that effectively idles, you'll still have to call vTaskDelay or equivalent.

Hantuch, 2021-12-17:
@moefear85 thanks it works!, but the funny thing is, according to the freeRTOS documentation https://www.freertos.org/RTOS-task-priority.html (and crosschecking of other's code), tskIDLE_PRIORITY is basically priority zero. I did check-run my code using 0 as priority and it works, unlike when I'm using 1