espressif/arduino-esp32 #3873: setTimeZone should not be static

Nagymadar, 2020-04-04:
setTimeZone is a pretty easy way to change time zone and summer time. This function should be available to be called from outside esp32-hal-time.c. Is there a particular reason to be static?

hitecSmartHome, 2021-09-24:
Why is it static?

me-no-dev, 2021-09-24:
It can not be called on it's own. NTP has to be started an that requires other things to be started as well.

closed 2021-09-24 by me-no-dev

jprouty, 2024-03-20:
For my particular use case, I want to use NTP to sync the time (roughly) once, after the first boot. Once the RTC time has been established, I will use a series of deep sleeps to conserve energy (on battery). On each subsequent boot (coming out of deep sleep) I only require setting the timezone (via setTimeZone) to then reinterpret the time in local time once again.

Electively, I could every so often sync with NTP to prevent skew.