espressif/arduino-esp32 #3874: How to increase UDP Packet Size?

diegovazquez, 2020-04-04:
### Hardware:
Board:			                        ESP32-CAM
Core Installation version:		1.11.2
IDE name:					Platform.io
Flash Frequency:				40Mhz
PSRAM enabled:                            no
Upload Speed:				115200
Computer OS:                                 Ubuntu

### Description:

Hello, english is not my mother tongue; please excuse any errors on my part. I whant to send a video using 1 UDP packet per frame, when i send the frame, gets fragmented in multiple UPD packets. 

I thing UDP MTU is set to 1500 (like TCP). How i can fix that? 

### Test:

tcpdump -i wlp62s0 udp port 4444

```
12:45:37.526372 IP _gateway.4444 > 255.255.255.255.8888: UDP, length 1460
12:45:37.539008 IP _gateway.4444 > 255.255.255.255.8888: UDP, length 1460
12:45:37.551666 IP _gateway.4444 > 255.255.255.255.8888: UDP, length 1460
12:45:37.564449 IP _gateway.4444 > 255.255.255.255.8888: UDP, length 1460
12:45:37.577128 IP _gateway.4444 > 255.255.255.255.8888: UDP, length 1460
12:45:37.583981 IP _gateway.4444 > 255.255.255.255.8888: UDP, length 732
```

### Sketch: 
```
#include "esp_camera.h"
#include <Arduino.h>
#include <WiFi.h>          
#include "WiFiUdp.h"

// -- WIFI --
#define WIFI_NAME           "FPV"                           // AP Name For mode JOIN or AP
#define WIFI_PASS           "12345678"                      // AP Password For mode JOIN or AP

# define UDP_TX_PORT        8888
# define UDP_RX_PORT        4444

// -- CAMERA --
# define CAMERA_FRAMESIZE               FRAMESIZE_CIF
# define CAMERA_VIDEO_COMPRESION        12     //0-63 lower number means higher quality

// AI Thinker Conf
// https://github.com/SeeedDocument/forum_doc/raw/master/reg/ESP32_CAM_V1.6.pdf
#define PWDN_GPIO_NUM     32
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27
#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
#define Y5_GPIO_NUM       21
#define Y4_GPIO_NUM       19
#define Y3_GPIO_NUM       18
#define Y2_GPIO_NUM        5
#define VSYNC_GPIO_NUM    25
#define HREF_GPIO_NUM     23
#define PCLK_GPIO_NUM     22

// ----------------------------------------------------------------------------------------------
WiFiUDP udp;
IPAddress broadcastIp;
camera_fb_t * fb = NULL;
char IP[] = "xxx.xxx.xxx.xxx";

void cameraSetup() {
  camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sscb_sda = SIOD_GPIO_NUM;
  config.pin_sscb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;

  // https://github.com/espressif/esp32-camera/issues/15
  // XCLK 20MHz or 10MHz for OV2640 double FPS (Experimental)
  // config.xclk_freq_hz = 20000000;
  config.xclk_freq_hz = 10000000;
    
  config.pixel_format = PIXFORMAT_JPEG;   //YUV422,GRAYSCALE,RGB565,JPEG
  //init with high specs to pre-allocate larger buffers
  if (psramFound()) {
    config.frame_size = FRAMESIZE_UXGA;   //QQVGA-QXGA Do not use sizes above QVGA when not JPEG
    config.jpeg_quality = 10;             //0-63 lower number means higher quality
    config.fb_count = 2;                  //if more than one, i2s runs in continuous mode. Use only with JPEG
  } else {
    config.frame_size = FRAMESIZE_SVGA;   //QQVGA-QXGA Do not use sizes above QVGA when not JPEG
    config.jpeg_quality = 12;             //0-63 lower number means higher quality
    config.fb_count = 1;                  //if more than one, i2s runs in continuous mode. Use only with JPEG
  }
   
  // camera init
  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    Serial.printf("Camera init failed with error 0x%x", err);
    return;
  }

  sensor_t * s = esp_camera_sensor_get();
  s->set_framesize(s, CAMERA_FRAMESIZE);
}

void setup() {
    // Serial port
    Serial.begin(115200);
    Serial.println("[INFO] Starting");
    Serial.println("[INFO] Serial set to 115200");

    // For OTA
    WiFi.mode(WIFI_STA);

    // Wifi
    Serial.println("[INFO] Setting AP (Access Point)");
    WiFi.softAP(WIFI_NAME, WIFI_PASS);
    IPAddress IP = WiFi.softAPIP();
    Serial.print("\tAP IP address: ");
    Serial.println(IP);

    // Camera 
    cameraSetup();
   
    broadcastIp = ~WiFi.subnetMask() | WiFi.gatewayIP();

    udp.begin(WiFi.localIP(), UDP_RX_PORT);

    Serial.print("[INFO] Broadcast IP ");
    Serial.println(broadcastIp);

    Serial.print("\tUDP TX PORT ");
    Serial.println(UDP_TX_PORT);
    
    Serial.print("\tUDP RX PORT ");
    Serial.println(UDP_RX_PORT);
}

void loop()
{   
    fb = esp_camera_fb_get();
    udp.beginPacket(broadcastIp, UDP_TX_PORT);
    udp.write(fb->buf, fb->len);
    udp.endPacket();
    Serial.println(fb->len);
    esp_camera_fb_return(fb);
    fb = NULL;

    sleep(5);
}
```

lbernstone, 2020-04-04:
UDP has no built in ordering, and UDP packets cannot be guaranteed to be properly reconstructed if fragmented.  There is a protocol called TCP that provides exactly what you want!

diegovazquez, 2020-04-04:
@lbernstone thank you for your answer. I will expand about what i want to do.

I want broadcast all video frames (esp32-cam) and make a receptor without join the ESP32 AP (similar to https://github.com/rodizio1/EZ-WifiBroadcast), for that, i need send the information without all TCP checks (loss packets is not a problem). This is very usefull for FPV (First-person view) for drones or planes. When your signal is lost, you dont need rejoin the AP, and you dont need get data in the esp32-cam.

For that is ideal 1 packet, 1 frame (small resolution (CIF)). 

Thank you

lbernstone, 2020-04-04:
You would need to [rebuild](https://github.com/espressif/arduino-esp32/blob/master/docs/lib_builder.md) the compiled libraries with changes to the lwip options.

diegovazquez, 2020-04-05:
thank you, i will try that and inform

closed 2020-06-18 by stale[bot]