espressif/arduino-esp32 #3875: ESP32 Watchdog getting Triggered when trying to run code on both cores

paulgiordano123, 2020-04-04:
**Hardware:**
Board: ESP32-WROVER-KIT
IDE name: Arduino IDE
Upload Speed: 115200

**Description:**
I am trying to have my chip run two processes simultaneously by running each one on a separate core. One process is an if statement that continually checks the value of a GPIO while the other is code to send picture data to a webserver I have deployed. I want each process to run forever in the forever loops of each core because I am trying to have a live stream video as well as a notification when motion has been detected. 
```
void Task1code( void * pvParameters ){
  Serial.print("Task1 running on core ");
  Serial.println(xPortGetCoreID());

  for(;;){
    if (digitalRead(4) == 1){
          //Establish Firebase Connection
          Firebase.begin("https://", "Ua");
          FirebaseJson json1;
          timeClient.update();
          formattedDate = timeClient.getFormattedDate();
          Serial.println(formattedDate);
          json1.set("timeDetected",formattedDate);
          Firebase.pushJSON(firebaseData, "00001", json1);
          Serial.println("Motion Detected");
          delay(25000);                      // Wait for two seconds (to demonstrate the active low LED)
      }
  }
}
```
```
void Task2code( void * pvParameters ){
  Serial.print("Task2 running on core ");
  Serial.println(xPortGetCoreID());

  for(;;){
    camera_fb_t *fb = esp_camera_fb_get();
    if(!fb){
      esp_camera_fb_return(fb);
      return;
    }

    if(fb->format != PIXFORMAT_JPEG){
      return;
    }

    client.sendBinary((const char*) fb->buf, fb->len);
    esp_camera_fb_return(fb);
    delay(1000);
  }
}
```
The problem is that the task watchdog gets triggered and makes the chip restart everytime motion has been detected. I either get the error message below or one that specifically says that the task watchdog has been triggered.
```
abort() was called at PC 0x4016f60b on core 0

Backtrace: 0x4008ef2c:0x3ffe7770 0x4008f15d:0x3ffe7790 0x4016f60b:0x3ffe77b0 0x4016f652:0x3ffe77d0 0x4016f6ff:0x3ffe77f0 0x4016f782:0x3ffe7810 0x4016fbe1:0x3ffe7830 0x4016fd1a:0x3ffe7850 0x40170059:0x3ffe7890 0x40170075:0x3ffe78b0 0x40170085:0x3ffe78d0 0x400d6539:0x3ffe78f0 0x400d65da:0x3ffe7980 0x400d58c6:0x3ffe79b0 0x400d2096:0x3ffe79d0 0x4008b675:0x3ffe79f0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
```

 

**Sketch:**
```
#include "esp_camera.h"
#include <WiFi.h>
#include <ArduinoWebsockets.h>
#include "FirebaseESP32.h"
#include "time.h"
#include <NTPClient.h>
#include "soc/rtc_wdt.h"
#include "soc/timer_group_struct.h"
#include "soc/timer_group_reg.h"

// Select camera model
//#define CAMERA_MODEL_WROVER_KIT
//#define CAMERA_MODEL_ESP_EYE
//#define CAMERA_MODEL_M5STACK_PSRAM
//#define CAMERA_MODEL_M5STACK_WIDE
#define CAMERA_MODEL_AI_THINKER

#include "camera_pins.h"

const char* ssid = "";
const char* password = "";
const char* websocket_server_host = "";
const uint16_t websocket_server_port =;
const char* ntpServer = "pool.ntp.org";
const long utcOffsetInSeconds = -18000;
const int   daylightOffset_sec = 3600;

FirebaseData firebaseData;

String formattedDate;

WiFiUDP ntpUDP;
NTPClient timeClient(ntpUDP, "pool.ntp.org", utcOffsetInSeconds);

using namespace websockets;
WebsocketsClient client;

TaskHandle_t Task1;
TaskHandle_t Task2;

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.println();

  camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sscb_sda = SIOD_GPIO_NUM;
  config.pin_sscb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 10000000;
  config.pixel_format = PIXFORMAT_JPEG;
  //init with high specs to pre-allocate larger buffers
  if(psramFound()){
    config.frame_size = FRAMESIZE_VGA;
    config.jpeg_quality = 40;
    config.fb_count = 2;
  } else {
    config.frame_size = FRAMESIZE_SVGA;
    config.jpeg_quality = 20;
    config.fb_count = 1;
  }

  // camera init
  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    Serial.printf("Camera init failed with error 0x%x", err);
    return;
  }

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print("waiting.....");
  }
  Serial.println("");
  Serial.println("WiFi connected");

  Serial.print("Camera Ready! Use 'http://");
  Serial.print(WiFi.localIP());
  Serial.println("' to connect");

  while(!client.connect(websocket_server_host, websocket_server_port, "/")){
    delay(500);
    Serial.print(".");
  }
  Serial.println("Websocket Connected!");

   //create a task that will be executed in the Task2code() function, with priority 1 and executed on core 1
  xTaskCreatePinnedToCore(
                    Task2code,   /* Task function. */
                    "Task2",     /* name of task. */
                    10000,       /* Stack size of task */
                    NULL,        /* parameter of the task */
                    6,           /* priority of the task */
                    &Task2,      /* Task handle to keep track of created task */
                    0);          /* pin task to core 1 */
    delay(500);
    //create a task that will be executed in the Task2code() function, with priority 1 and executed on core 1
  xTaskCreatePinnedToCore(
                    Task1code,   /* Task function. */
                    "Task1",     /* name of task. */
                    10000,       /* Stack size of task */
                    NULL,        /* parameter of the task */
                    6,           /* priority of the task */
                    &Task1,      /* Task handle to keep track of created task */
                    1);          /* pin task to core 1 */
    delay(500);

}
void Task1code( void * pvParameters ){
  Serial.print("Task1 running on core ");
  Serial.println(xPortGetCoreID());

  for(;;){
    if (digitalRead(4) == 1){
          //Establish Firebase Connection
          Firebase.begin("", "");
          FirebaseJson json1;
          timeClient.update();
          formattedDate = timeClient.getFormattedDate();
          Serial.println(formattedDate);
          json1.set("timeDetected",formattedDate);
          Firebase.pushJSON(firebaseData, "00001", json1);
          Serial.println("Motion Detected");
          delay(25000);                      // Wait for two seconds (to demonstrate the active low LED)
      }
  }
}

//Task2code: blinks an LED every 700 ms
void Task2code( void * pvParameters ){
  Serial.print("Task2 running on core ");
  Serial.println(xPortGetCoreID());

  for(;;){
    camera_fb_t *fb = esp_camera_fb_get();
    if(!fb){
      esp_camera_fb_return(fb);
      return;
    }

    if(fb->format != PIXFORMAT_JPEG){
      return;
    }

    client.sendBinary((const char*) fb->buf, fb->len);
    esp_camera_fb_return(fb);
    delay(1000);
  }
}

void loop() {

}
```

Pallanez, 2020-04-04:
Try putting a vTaskDelay() or yield() there: 
```
void Task1code( void * pvParameters ){
  Serial.print("Task1 running on core ");
  Serial.println(xPortGetCoreID());

  for(;;){
    if (digitalRead(4) == 1){
          //Establish Firebase Connection
          Firebase.begin("", "");
          FirebaseJson json1;
          timeClient.update();
          formattedDate = timeClient.getFormattedDate();
          Serial.println(formattedDate);
          json1.set("timeDetected",formattedDate);
          Firebase.pushJSON(firebaseData, "00001", json1);
          Serial.println("Motion Detected");
          delay(25000);                      // Wait for two seconds (to demonstrate the active low LED)
      }
     vTaskDelay(1);
  }
}
```

paulgiordano123, 2020-04-04:
Just tried it vTaskDelay and yield() and both still resulted in the same error and restart as before. Could it be something from task2 as the error message says abort caused on core 0 and that's where task2 is operating?

Pallanez, 2020-04-04:
Ok I think the problem may be this part: 

```
void Task2code( void * pvParameters ){
  Serial.print("Task2 running on core ");
  Serial.println(xPortGetCoreID());

  for(;;){
    camera_fb_t *fb = esp_camera_fb_get();
    if(!fb){
      esp_camera_fb_return(fb);
      return; //YOU CAN'T RETURN FROM A TASK!
    }

    if(fb->format != PIXFORMAT_JPEG){
      return; //YOU CAN'T RETURN FROM A TASK! 
    }

    client.sendBinary((const char*) fb->buf, fb->len);
    esp_camera_fb_return(fb);
    delay(1000);
  }
}
```

Try ditching the ```return``` statements. freeRTOS documentation is very clear about the fact that you can't return from a task, instead, you should call vTaskDelete(NULL) if you want the task to be completely terminated.

Pallanez, 2020-04-04:
Let me know if that fixed your issue!

paulgiordano123, 2020-04-04:
Just tried it and still got the same abort called on core 0 error and system restart. The weird part is that the code from core 0 in task2 works properly until the code in task1 happens. As soon as the task1 if statement gets triggered, the chip crashes and restarts and says core 0 aborted.

Pallanez, 2020-04-04:
Can you try decoding the backtrace using the TraceDecoder tool?

closed 2020-06-17 by stale[bot]