espressif/arduino-esp32 #3877: To use arduino as a component of ESP-IDF do not work!

Nexus1212, 2020-04-06:
Hello,
I am following this manual:

https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md

My goal is to flash a arduino sketch with esp-idf.

My system is a win10.

The following command in the commandshell does not work: "`make menuconfig`"
I get the error: "Der Befehl "make" ist entweder falsch geschrieben oder
konnte nicht gefunden werden." (In english: The command can not be found)

I tried another command: "`idf.py menuconfig`"
I get the error: 

```

Checking Python dependencies...
Python requirements from C:\Users\Chris\Desktop\esp-idf-4.0\requirements.txt are satisfied.
Executing action: menuconfig
Running cmake in directory c:\users\chris\desktop\esp-idf-4.0\hello_world\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 c:\users\chris\desktop\esp-idf-4.0\hello_world"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.21.0.windows.1")
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/Chris/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/Chris/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/Chris/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/Chris/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Users/Chris/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project version: v4.0-dirty
-- Building ESP-IDF components for target esp32
CMake Error at C:/Users/Chris/Desktop/esp-idf-4.0/tools/cmake/build.cmake:185 (message):
  Failed to resolve component 'ethernet'.
Call Stack (most recent call first):
  C:/Users/Chris/Desktop/esp-idf-4.0/tools/cmake/build.cmake:211 (__build_resolve_and_add_req)
  C:/Users/Chris/Desktop/esp-idf-4.0/tools/cmake/build.cmake:425 (__build_expand_requirements)
  C:/Users/Chris/Desktop/esp-idf-4.0/tools/cmake/project.cmake:337 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/Chris/Desktop/esp-idf-4.0/hello_world/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

```

lbernstone, 2020-04-06:
The recommended method is to use [lib-builder](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md).  You will need to checkout the release/v4.0 version of lib-builder to compile with IDF 4.0, and checkout the matching rev of IDF (0a2bec85f3ab4fc3ba0fa47f03f27fa7876c84d9).

Nexus1212, 2020-04-06:
I am not sure how that works...
Could you send me a link to a manual please or explain me it?

lbernstone, 2020-04-06:
If you don't know how to use git, this is not going to work out well for you.

Nexus1212, 2020-04-07:
Git is not the problem ^^ 

In the [manual ](https://github.com/espressif/arduino-esp32/blob/master/docs/lib_builder.md)of the lib builder is written at step 6:
`make menuconfig`
I think it will also not work, because the "make" command will not work on my windows machine.

Could you explain me please, how I can use this command?

lbernstone, 2020-04-07:
I don't do Windows.
There are directions for setting up esp-idf on windows at https://github.com/espressif/esp-idf.  It is complicated, and then slow once you get it working.

jlozanob2, 2020-04-12:
Hi guys, 
I'm stuck in a similar issue, I'm trying to compile Arduino as a component to be able to use some Arduino libraries into ESP-IDF development. To be honest, I'm not a guru on C or C++ compiling, so I invested several days on the configuration of several development environments on Windows 10 following the Espressif docs. Currently I have one development environment based on the ESP-IDF V4.0 and Eclipse in one computer and other environment based on ESP-IDF 3.2
 and Eclipse in other Windows 10 computer. I can compile and flash ESP32 with booth environments, but it was impossible to me to get in working the procedure of "To use Arduino as a component of ESP-IDF". For the ESP-IDF V4 environment I tried to compile following the procedure of components and I got the compiling error "... Ethernet module not found..." and nothing to do :(. For the v3.2 environment I had several errors trying to compile as components and stucked again. 
Reading in posts, I figure the Arduino SDK was developed from ESP-IDF release v3.3, so I think we have to have a development environment for the ESP-IDF v3.3 and then follow the process "To use Arduino as a component of ESP-IDF", I will try to configure it and compile, but any help/clarify about all this topic will be very appreciated :)

Best regards,
Jorge

lbernstone, 2020-04-12:
Your time will be better spent learning platform.IO, which has recently been upgraded to support ESP-IDF 4.0 with arduino as a component.  Compiling your own libraries is not intended for a novice.  In fact, the entire purpose of Arduino and other IDEs is to hide the complexities of compiling firmware so that it can be used with little knowledge of programming or systems development.

jlozanob2, 2020-04-12:
Hi lbernstone

Thanks for your reply. 
I'm moving from Arduino IDE to ESP-IDF due to the fact that Arduino has only one user program thread and one blocking operation of a library (for ex. Bluetooth SPP connect) blocks the entire program loop. Could I create applications for ESP32 with Plaptorm.IO capable of working with existing Arduino libraries (WIFI, LCD Screen, Bluetooth SPP) and several threads running at same time? 

BR

lbernstone, 2020-04-12:
Yes, but you can do that in Arduino IDE as well.  All the IDF functionality is available (well, most of it), so you can call xTaskCreate to make another thread.  After all, arduino-esp32 is simply an extension component of ESP-IDF.

jlozanob2, 2020-04-12:
In Arduino IDE I tried some examples of pseudo multitasking like millis() based and others, but not worked for Bluetooth blocking tasks. After your comment about xTaskCreate I found a [post ](https://www.savjee.be/2020/01/multitasking-esp32-arduino-freertos/) about it very interesting and probably matching my needs, I will try to code it into Arduino IDE and I will come back with the results :)

Thanks for your help. 

BR

jlozanob2, 2020-04-12:
Hi lbernstone,

It works on Arduino with xTaskCreate! 
I have 3 threads working at same time, one for Bluetooth SPP connection/reconnection managing, the second one for Bluetooth data reading and Serial data retransmision and third one is the Loop() with a Serial.println("something") each 2 sec. The 3 threads are working in parallel and doing its works without interferring the others.
I will try to implement the rest of the functionalities, but I think I'm in the correct way :)

Thanks again for your help.

Best regards.
Jorge.

closed 2020-06-25 by stale[bot]

rpolitex, 2020-09-26:
@Nexus1212 @lbernstone 
It's old conversation, but in case this helps for somebody.
You need to switch to the correct branch of https://github.com/espressif/arduino-esp32
The master branch currently does not work with ESP-IDF 4.0.

But *switching to the* `idf-release/v.4.0` branch helps me.