espressif/arduino-esp32 #3878: serial.end() doesn't work

aster94, 2020-04-06:
### Description:
`Serial.end()` should free the serial pins and leave the user the ability to do `digitalWrite()` on them, but it doesn't work. Instead to drive the TX pin to a LOW state i need to reset the board.

I tested the code below with the following cores:
- last stable release from platformio
- master branch
- master + #3713

I also made three logic analyzer captures.
An example from one of these, as you can see it shows that the pin is driven low only once (after upload of the code or reset)
[la.zip](https://github.com/espressif/arduino-esp32/files/4441212/la.zip)

[image]

### Sketch: 
```cpp

#include <Arduino.h>

HardwareSerial ser(2);
#define TX_PIN 17

void setup(){}

void loop()
{
    pinMode(TX_PIN, OUTPUT);
    delay(10);

    digitalWrite(TX_PIN, HIGH);
    delay(50);
    digitalWrite(TX_PIN, LOW);
    delay(300);
    digitalWrite(TX_PIN, HIGH);
    delay(50);

    ser.begin(9600);
    delay(10);

    for (uint8_t n = 1; n < 5; n++)
    {
        ser.write(n);
        delay(10);
    }
    ser.end();
    delay(300);
}

```

aster94, 2020-04-07:
@lbernstone do you have any idea about what can cause this behaviour?

lbernstone, 2020-04-07:
Turn on verbose debugging so you can see if any errors are shown.
Serial2 is instantiated by default.  Why are you not using it?
Make your delays longer- a full second whenever you attach or detach pins.
Try manually detaching after the ser.end:
`gpio_matrix_out(TX_PIN, 0x100, false, false);`

lyusupov, 2020-04-09:
One more ticket related to this issue:
https://github.com/espressif/arduino-esp32/issues/3563#issuecomment-567199429

aster94, 2020-04-11:
@lbernstone looks like you solved it 😁

[image]

At first, I tried as you suggested to put big delays, then slowly I removed all of them and no problem showed up in about 5 minutes running this sketch

I think it is an easy fix if the maintainers need I can make a PR @me-no-dev 

```cpp
#include <Arduino.h>

#define TX_PIN 17

void setup(){}

void loop()
{
    pinMode(TX_PIN, OUTPUT);

    digitalWrite(TX_PIN, HIGH);
    delay(50);
    digitalWrite(TX_PIN, LOW);
    delay(1000);
    digitalWrite(TX_PIN, HIGH);

    Serial2.begin(9600);

    for (uint8_t n = 1; n < 5; n++)
    {
        Serial2.write(n);
        delay(50);
    }
    Serial2.end();
    gpio_matrix_out(TX_PIN, 0x100, false, false);
}
```

Also, note that after the `begin` no bytes are lost
[image]

Here the LA log: 
[log.zip](https://github.com/espressif/arduino-esp32/files/4464774/log.zip)

lbernstone, 2020-04-11:
The gpio_matrix disconnect is already buried in the end method.  It just isn't working for some reason.  I'll see if I can track down where it is dropping the ball.

marcmathys, 2020-05-03:
Saw that you are interested in a flutter app to standardize access to BT wearables - did you get anywhere.  I also find this interesting and would contribute.

closed 2020-07-16 by stale[bot]

SuGlider, 2021-07-19:
Issue solved with PR #5385

commit 80418fad (2020-09-30): Fixes UART detach.  Fixes #3878 (#3894)

* Fixes UART detach.  Fixes #3878

* 0 is not a good holder value for pins!

* 0 is not a good holder value for pins!