espressif/arduino-esp32 #3885: Problem with connect to AWS mqtt usind Ethernet board enc28j60

marcoavaccaro, 2020-04-09:
Hello guys,

I'm using an esp32 dev board and an enc28j60 module to send messages to aws iot plataform using mqtt. I already done that using wifi but now i'm trying use ethernet.

I also already could post a tweet using this ethernet module with the lib Ethercard.h. So i can connect to internet via ethernet and can connect to aws mqtt using wifi. Now the problem is to do them both.

I could upload the new code to the esp but it crashes entering the function hornbill.connect() just after printing "test"

The log shows until the reboot is:

```Setting up DHCP
My IP: 192:168:1:10
Netmask: 255:255:255:0
GW IP: 192:168:1:1
DNS IP: 192:168:1:1
test
assertion "Invalid mbox" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c", line 267, function: tcpip_callback_with_block
abort() was called at PC 0x400fc527 on core 1

Backtrace: 0x4008dbe0:0x3ffb1bc0 0x4008de11:0x3ffb1be0 0x400fc527:0x3ffb1c00 0x4010a7aa:0x3ffb1c30 0x4010a775:0x3ffb1c50 0x4010a361:0x3ffb1c90 0x400ead65:0x3ffb1ce0 0x400d3300:0x3ffb1d30 0x400d246f:0x3ffb1e80 0x400d3922:0x3ffb1ed0 0x400d1aa2:0x3ffb1f70 0x400d5907:0x3ffb1fb0 0x4008a821:0x3ffb1fd0

Rebooting...
```
So i guess it connect to the internet but some error is happening and i don't know why.
The backtrace translaction is:
[https://i.imgur.com/5BOvjsr.png](url)

### Hardware:
Board:							ESP32 Dev Module 
IDE name:					        Arduino IDE 1.8.12
Flash Frequency:					80Mhz
Computer OS:                                         Windows 10

### Sketch: 
```cpp

#include <AWS_IOT.h>
//#include <WiFi.h>
#include <EtherCard.h>
#include <SPI.h>

// include your personal WiFi and AWS configuration. 
#include "config.h"

byte mymac[] = {  0x74,0x69,0x69,0x2D,0x30,0x31  };
byte Ethernet::buffer[700];

char TOPIC_NAME[50];
AWS_IOT hornbill;

//int status = WL_IDLE_STATUS;
int tick=0,msgCount=0,msgReceived = 0;
char payload[512];
char rcvdPayload[512];
const char* CLIENT_ID = "001";
void mySubCallBackHandler (char *topicName, int payloadLen, char *payLoad)
{
    strncpy(rcvdPayload,payLoad,payloadLen);
    rcvdPayload[payloadLen] = 0;
    msgReceived = 1;
}

void setup() {
  Serial.begin(115200);

  String auxTopic_Name = "teste/marco";
  auxTopic_Name.toCharArray(TOPIC_NAME, 50);

    if (ether.begin(sizeof Ethernet::buffer, mymac) == 0) 
    Serial.println(F("Failed to access Ethernet controller"));

    Serial.println(F("Setting up DHCP"));
    if (!ether.dhcpSetup())
    Serial.println(F("DHCP failed"));

  ether.printIp("My IP: ", ether.myip);
  ether.printIp("Netmask: ", ether.netmask);
  ether.printIp("GW IP: ", ether.gwip);
  ether.printIp("DNS IP: ", ether.dnsip);  
  Serial.println("test");
   /* while (status != WL_CONNECTED)
    {
        Serial.print("Attempting to connect to SSID: ");
        Serial.println(WIFI_SSID);
        // Connect to WPA/WPA2 network. Change this line if using open or WEP network:
        status = WiFi.begin(WIFI_SSID, WIFI_PASSWORD);

        // wait 5 seconds for connection:
        delay(5000);
    }

    Serial.println("Connected to wifi");*/

  if (!ether.dnsLookup(HOST_ADDRESS))
    Serial.println(F("DNS failed"));

  ether.printIp("SRV: ", ether.hisip);

  Serial.println("test");

    if(hornbill.connect(HOST_ADDRESS, CLIENT_ID,
                aws_root_ca_pem, certificate_pem_crt, private_pem_key)== 0)
    {
        Serial.println("Connected to AWS");
        delay(1000);

        if(0 == hornbill.subscribe(TOPIC_NAME,mySubCallBackHandler))
        {
            Serial.println("Subscribe Successfull");
        }
        else
        {
            Serial.println("Subscribe Failed, Check the Thing Name and Certificates");
            while(1);
        }
    }
    else
    {
        Serial.println("AWS connection failed, Check the HOST Address");
        while(1);
    }

    delay(2000);

}

void loop() {
    Serial.println("loop");
    if(msgReceived == 1)
    {
        msgReceived = 0;
        Serial.print("Received Message:");
        Serial.println(rcvdPayload);
    }
    if(tick >= 5)   // publish to topic every 5seconds
    {
        tick=0;
        sprintf(payload,"Hello from hornbill ESP32 : %d",msgCount++);
        if(hornbill.publish(TOPIC_NAME,payload) == 0)
        {        
            Serial.print("Publish Message:");
            Serial.println(payload);
        }
        else
        {
            Serial.println("Publish failed");
        }
    }  
    vTaskDelay(1000 / portTICK_RATE_MS); 
    tick++;
}
```

closed 2020-06-22 by stale[bot]

zigsolutions, 2020-07-14:
I am seeing the exact same thing with my setup when connecting to AWS using modem (in my case) and not an Ethernet board.

I can connect to AWS just fine over WiFi but as soon as I try with a modem I get the same error you're seeing @marcoavaccaro. 

Has anyone else come across this issue when trying to connect to AWS on another connection other than WiFi? Thank you so much in advance!

```
assertion "Invalid mbox" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c", line 267, function: tcpip_callback_with_block
abort() was called at PC 0x4012c027 on core 1

Backtrace: 0x4008cca8:0x3ffb1bb0 0x4008ced9:0x3ffb1bd0 0x4012c027:0x3ffb1bf0 0x40140042:0x3ffb1c20 0x4013fed9:0x3ffb1c40 0x4013f6b9:0x3ffb1c80 0x400fc399:0x3ffb1cd0 0x400d9f39:0x3ffb1d20 0x400d9254:0x3ffb1e70 0x400d8fdc:0x3ffb1eb0 0x400d2e24:0x3ffb1f50 0x400d7bbf:0x3ffb1f80 0x400e11dd:0x3ffb1fb0 0x400893bd:0x3ffb1fd0
```

1simone0, 2021-01-30:
Hello @marcoavaccaro , did you find a solution on how to connect ESP32 to AWS using Ethernet?

Thanks,
Simone

PranavD10, 2021-02-28:
Also facing the same problem with DOIT DEV KIT V1

woodlist, 2021-03-01:
I got the same problem on TTGO T1, upon successful establishing GPRS connection and mapping on DDNS, unable to run AsyncWebServer on 80 port. The WiFi is not initialized and do not required in given project

madaerodog, 2022-07-13:
Hello guys, I had the same problem with ESP32 and a  GPRS connection over Sim868.
The way I made it work and my understanding of the situation is as follows:

Although you might not want to use Wifi in your project for the time being you might want to initiate it with WiFi.begin();
because Esp32's internal clock, TCP, other components, and features are only initiated when wifi is initiated.

Even more so, because you are not connecting to a Wifi network that would have given you the current time you have to manually set this internal time, for this, I have used https://github.com/fbiego/ESP32Time and I have used the time provided by the GPRS module upon boot.

My use case involved accessing an HTTPS link with a firmware file to download and install remotely via GPRS with a provided certificate. So without the internal components started and the current time correct the request will surely fail.

So as a quick summary of the fix:

#include <WiFi.h>
#include <ESP32Time.h>
ESP32Time rtc(3600);  // offset in seconds GMT+1

and in setup before doing anything else:

WiFi.begin();
rtc.setTime(30, 58, 15, 13, 7, 2022); // you either set time manually or take it from your Ethernet / GPRS module

And this was the only way to make it work and not get the error.
Hope it helps!

mirkogeest, 2023-10-17:
This saved my day, thank you @madaerodog!

I only have added the WiFi.begin() without the setTime and now the aws iot sdk works like a charm.

My hardware is a LilyGo T-SIM7080G-S3 ESP32-S3 (TINY_GSM_MODEM_SIM7080), and the exact error I got when calling hornbill.connect was:
`assert failed: tcpip_send_msg_wait_sem IDF/components/lwip/lwip/src/api/tcpip.c:455 (Invalid mbox)`