espressif/arduino-esp32 #3886: mDNS stops working after two minutes

ademuri, 2020-04-09:
### Hardware:
Board:							DOIT devkit v1, TTGO oled + battery ESP32
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    not sure
Upload Speed:					115200
Computer OS:                                        Ubuntu

### Description:
I'm trying to use MDNS. Using the mDNS_Web_Server example, when the device starts, I am able to access the page at 'esp32.local/'. However, after two minutes, this stops working.

Manually resolving the mDNS address shows the same behavior - it works for the first two minutes after the device is started, then stops working:

Success:
```
$ avahi-resolve-host-name esp32.local
esp32.local     192.168.88.94
```

Failure:
```
$ avahi-resolve-host-name esp32.local                                        │+ wlp2s0 IPv4 Google-Cast-Group-32c30c3353a34b4ca0b6ae11e49bd15c-1 _googlecast._tcp     local
Failed to resolve host name 'esp32.local': Timeout reached
```

I did a Wireshark capture on the IP address of the ESP32. It broadcasts the mDNS messages, with a TTL of 2 minutes. It only ever broadcasts those messages at startup (I left the device for ~20m).

If I manually call `MDNS.begin` periodically, it fixes the issue.

I was able to reproduce this on two dev boards.

### Sketch:
mDNS_Web_Server example (only modified to set the ssid/password)

### Debug Messages:
Nothing

meditant, 2020-05-10:
I have problems too with mDNS !
It stop to work after some time....

everthu, 2020-06-02:
Same error here with mDNS, it seems to be more worse when the module has more distance to the router. However, in all cases it is still possible to directly connect via the IP adress.
Is there a way to gather more logging? The verbose logging itself doesn't show anything about mDNS.

thomastech, 2020-06-05:
+1 
I have this problem too: mDNS access is lost shortly after a ESP32 reboot. IP address access never fails.

> Also my solution is to recall MDNS.begin each several minutes
I tried that with various recall periods of 30 secs to several minutes. Does not help me.

Question for those of you that are experiencing the mDNS problem: Does your code also use the OTA functions?

Hopefully someone knows about a workaround for this issue.

- Thomas

R030t1, 2021-09-11:
I am encountering this issue. Reopen?

Anumann, 2023-09-14:
I also have this probelm