espressif/arduino-esp32 #3887: SSD1306 and AsyncWebServer causes weird artifacts on i2c LCD

gbaranski, 2020-04-09:
### Hardware:
Board:							WEMOS Lolin32
Core Installation version:			Sorry but no idea, couldn't find it, but the newest for this board from platformio

IDE name:						Platform.io
Flash Frequency:					?40Mhz?
PSRAM enabled:                                     No idea
Upload Speed:					115200?
Computer OS:                                         OS X

### Description:
I use AsyncWebServer, code for setting up server looks something like this 
```cpp
    server.on("/setAlarm", HTTP_GET, [](AsyncWebServerRequest * request) {
        if(request->hasParam("time")) {
            AsyncWebParameter* p = request->getParam("time");
            Serial.printf("Received %s with value %s from IP: \n", p->name().c_str(), p->value().c_str());
            Serial.println(request->client()->remoteIP());
            request->send(200, "text/plain", "OK");
            wifiTimeManager.saveAlarmTime(String(p->value()));
            String outputMsg = "New request!\n Time: " + String(p->value()) + "\n IP: " + request->client()->remoteIP().toString();
            lcdManager.printTextLcd("testing", 1);
        }
    });
```
As you can see on the bottom i have lcdManager.printTextLcd function called, first argument is String to print and second one is font size, function looks like this
```cpp
void ManageLcd::printTextLcd(String lcdText, int fontSize){
    clearLcd();
    display.setTextSize(fontSize);
    display.print(lcdText);
    display.display();
}
```
Here we got clearLcd(), it just clears the display
```cpp
void ManageLcd::clearLcd() {
    display.setTextColor(WHITE);
    display.setCursor(10, 0);
    display.clearDisplay();

    display.display();
}
```
Problem is when i want to print any text when request have been received, on ESP8266 i used schedule_function on async code, but i heard that on ESP32 i dont need that, and anyways schedule_function doesn't exist in ESP32 core, when i receive request i2c LCD connected to ESP prints me something like this https://imgur.com/a/hKBHXiY I also tried xTaskCreate but that gave me the same error

GPIO 4 and GPIO 5, setup for lcd looks like this 

```
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, -1);

bool ManageLcd::setupLcd() {
    Wire.begin(5,4);
    if(!display.begin(SSD1306_SWITCHCAPVCC, 0x3C)) {
        Serial.println("SSD1306 allocation failed");
        return false;
    }
    delay(1000);
    display.clearDisplay();
    display.setTextColor(WHITE);
    display.setTextSize(1);
    display.display();
    delay(1000);
    return true;
}
```

blotfi, 2020-04-10:
I use u8g2 lib for my OLED I2C display
and noticied that if I use HW I2C it gives sporadic resets with the AsyncWebServer but if I use SW emulated I2C then it works well:
https://github.com/olikraus/u8g2/issues/1150

stickbreaker, 2020-04-10:
part of the problem you are seeing is the SSD1306 display is a graphic display, to update the screen you have to send all the pixels, so a refresh cycle can take a while to send the 1024 bytes. If your updates come in too fast and concurrent calls are made to `Wire()` bad stuff will happen(glitches).

The standard AdaFruit driver sends the screen update in less than 32byte blocks to be compatible with the older Arduino hardware.  So to update the 128x64 screen it takes more than 32 I2C transactions.  I optimized Adafruits SSD1306 driver for the ESP32 here:[ssd1306](https://github.com/stickbreaker/SSD1306).  It is a drop in replacement. it uses a clipping window for updating and sends bigger blocks, it also has some error detection to handle screen glitches.  With it I have been able to get up to 70 screen updates per second.

Chuck.

closed 2020-06-23 by stale[bot]