espressif/arduino-esp32 #3899: ESP32 boards reset on 'preferences.begin( "a_namespace", false )'

CelliesProjects, 2020-04-12:
### Hardware:
Board:							ESP32 Wemos look-a-like with a ESP32S
Core Installation version:			1.0.4 and master
IDE name:						Arduino IDE 1.8.11 - 1.8.12
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					921600 - 460800
Computer OS:                                        Ubuntu

### Description:
The most recent batch of boards I bought have a strange problem:
The boards reset as soon as I use `preferences.begin( "a_namespace", false ); `
It resets a couple of times and then starts running normally.
My other boards that I tested don't have this problem.
The board below is maybe not the same as mine, but it looks exactly the same.
Below the debug output I put the decoded exception which tells me nothing.

Anybody familiar with this or better yet anybody knows how to solve it? 

[image]

### Sketch: 
```cpp
#include <WiFi.h>
#include <Preferences.h>

const char* ssid = "ssid";
const char* password = "psk";

Preferences preferences;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("Stability test ESP32S module");
  Serial.printf( "Connecting to %s", ssid );
  WiFi.begin( ssid, password );
  while ( WiFi.status() != WL_CONNECTED )
  {
    delay(10);
    Serial.print(".");
  }
  Serial.println();
  Serial.println( "connected" );
  Serial.println( WiFi.localIP().toString() );

  preferences.begin( "a_namespace", false );  
  Serial.println("Setup done.");
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Messages:
```
st:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10864
load:0x40080400,len:6432
entry 0x400806b8
Stability test ESP32S module
Connecting to huiskamerGuru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
[... 204 lines cut ...]
entry 0x400806b8
Stability test ESP32S module
Connecting to huiskamer..................
connected
192.168.0.105
Setup done.
```
### Decoded exception:
```bash
PC: 0x400d865e: rtc_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/rtc_module.c line 1965
EXCVADDR: 0x00000000

Decoding stack results
0x400d865e: rtc_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/rtc_module.c line 1965
0x401212c6: ram_get_fm_sar_dout at phy_chip_v7_cal.c line 279
0x401225e9: ram_rfcal_pwrctrl at phy_chip_v7_cal.c line 1639
0x40122b6a: tx_pwctrl_init_cal at phy_chip_v7_cal.c line 2143
0x40122bc9: tx_pwctrl_init at phy_chip_v7_cal.c line 2182
0x40120536: bb_init at phy_chip_v7.c line 2623
0x40120b0a: register_chipv7_phy at phy_chip_v7.c line 3371
0x400d64bf: esp_phy_rf_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/phy_init.c line 169
0x400d6742: esp_phy_load_cal_and_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/phy_init.c line 659
0x4008df39: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

```

MarkyAD, 2020-04-12:
Preferences need a begin() and an end() while declaring your preference variables within them.
```c
preferences.begin("myproject", false);
preferences.putUChar("myvar", var);
preferences.end();
```
Same for reading them back...
```c
preferences.begin("myproject", true);
var = preferences.getUChar("myvar", var);
preferences.end();
```

lbernstone, 2020-04-12:
This is not correct, @MarkyAD.  He has defined preferences globally, so it does not need to be ended.  This is likely a power issue.  OP will update when he gets a different power supply.

MarkyAD, 2020-04-12:
Sure he's defined them globally, but the MCVE sketch he posted shows him doing nothing else with them.

MarkyAD, 2020-04-12:
Yeah youre right, the sketch works here it just doesn't do anything, but it doesn't crash. I have several of those 32S mini and they dont seem to draw a whole lot of power... less in fact than the wroom and wrover boards.

CelliesProjects, 2020-04-13:
@lbernstone It is really a power problem. Just setup a fresh board and fed it 5V on the VCC pin and still no luck. Also I removed `preferences.begin( "a_namespace", false ); ` which is not the cause after all.

CelliesProjects, 2020-04-13:
Just checked the LDO on the board with the marking `4B2X` which seems to be a [LP2983](https://www.ti.com/lit/ds/symlink/lp2983.pdf) with a rated output of 150mA.
There is the culprit. 

Only option for these boards is to feed them 3.3V directly.
@MarkyAD Could you check the LDO on one of your boards?

closed 2020-04-13 by CelliesProjects

MarkyAD, 2020-04-13:
I opened up a new one and it has the 4B2X regulator, so I assume that all of the others here have the same. Good info to know about the weak reg, thanks.

lbernstone, 2020-04-13:
Well, the device needs 100-150mA at init.  It may be possible to get it to an operational state if you start the system at 80MHz.  Delay for a second while the main system inits. Fire up the WiFi and waitForConnectResult.  If you can get that far, you should be able to switch over to a higher MHz and operate normally.

CelliesProjects, 2020-04-14:
Thank you both for the effort.

@lbernstone For my current project that is not an option. Maybe for a future project. Thanks for the tip.