espressif/arduino-esp32 #3901: Eclipse compile the arduino example: /arduino - esp32 / libraries/WebServer/examples/HttpBasicAuth /

LusterWong, 2020-04-12:
WIN7 SP1 system.
The first error is the famous "#include MBEDTLS CONFIG FILE", which has been resolved
But there are other errors.
My code has not been modified, it is directly copied and compiled, and the eclipse compilation environment should be configured correctly, because other examples have been compiled successfully.

The error message is as follows:

C:/Users/xxxxxx/.arduinocdt/packages/esp32/hardware/esp32/1.0.4/libraries/ESPmDNS/src/ESPmDNS.cpp: In member function 'void MDNSResponder::enableArduino(uint16_t, bool)':
<command-line>:0:17: error: 'esp32' was not declared in this scope
C:/Users/xxxxxx/.arduinocdt/packages/esp32/hardware/esp32/1.0.4/libraries/ESPmDNS/src/ESPmDNS.cpp:51:18: note: in definition of macro 'STR'
 #define STR(tok) tok
                  ^
C:/Users/xxxxxx/.arduinocdt/packages/esp32/hardware/esp32/1.0.4/libraries/ESPmDNS/src/ESPmDNS.cpp:93:45: note: in expansion of macro 'ARDUINO_VARIANT'
         {(char*)"board"         ,(char*)STR(ARDUINO_VARIANT)},
                                             ^
makefile:459: recipe for target 'libraries/ESPmDNS/src/ESPmDNS.cpp.o' failed
make: *** [libraries/ESPmDNS/src/ESPmDNS.cpp.o] Error 1

The code was copy from /arduino - esp32 / libraries/WebServer/examples/HttpBasicAuth /

LusterWong, 2020-04-12:
[image]
This is about the setting of ESP32's arduino library

lbernstone, 2020-04-12:
platform.txt provides a define for ARDUINO_VARIANT in Arduino IDE.  You will need to provide that through eclipse.

LusterWong, 2020-04-13:
> platform.txt provides a define for ARDUINO_VARIANT in Arduino IDE. You will need to provide that through eclipse.
Thanks.

I don't quite understand the role of platform.txt. I thought it would work as long as ESP's arduino plugin was installed.

The platform.txt automatically downloaded by the plugin is different from the file on github, which is very weird.Do I need to overwrite with github files?

Is there any way to directly set ARDUINO_VARIANT with eclipse IDE?

LusterWong, 2020-04-13:
`
name=ESP32 Arduino
version=

tools.esptool_py.path={runtime.tools.esptool_py.path}
tools.esptool_py.cmd=esptool
tools.esptool_py.cmd.linux=esptool.py
tools.esptool_py.cmd.windows=esptool.exe

tools.esptool_py.network_cmd=python "{runtime.platform.path}/tools/espota.py"
tools.esptool_py.network_cmd.windows="{runtime.platform.path}/tools/espota.exe"

tools.gen_esp32part.cmd=python "{runtime.platform.path}/tools/gen_esp32part.py"
tools.gen_esp32part.cmd.windows="{runtime.platform.path}/tools/gen_esp32part.exe"

compiler.warning_flags=-w
compiler.warning_flags.none=-w
compiler.warning_flags.default=
compiler.warning_flags.more=-Wall -Werror=all
compiler.warning_flags.all=-Wall -Werror=all -Wextra

compiler.path={runtime.tools.xtensa-esp32-elf-gcc.path}/bin/
compiler.sdk.path={runtime.platform.path}/tools/sdk
compiler.cpreprocessor.flags=-DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-I{compiler.sdk.path}/include/config" "-I{compiler.sdk.path}/include/app_trace" "-I{compiler.sdk.path}/include/app_update" "-I{compiler.sdk.path}/include/asio" "-I{compiler.sdk.path}/include/bootloader_support" "-I{compiler.sdk.path}/include/bt" "-I{compiler.sdk.path}/include/coap" "-I{compiler.sdk.path}/include/console" "-I{compiler.sdk.path}/include/driver" "-I{compiler.sdk.path}/include/esp-tls" "-I{compiler.sdk.path}/include/esp32" "-I{compiler.sdk.path}/include/esp_adc_cal" "-I{compiler.sdk.path}/include/esp_event" "-I{compiler.sdk.path}/include/esp_http_client" "-I{compiler.sdk.path}/include/esp_http_server" "-I{compiler.sdk.path}/include/esp_https_ota" "-I{compiler.sdk.path}/include/esp_ringbuf" "-I{compiler.sdk.path}/include/ethernet" "-I{compiler.sdk.path}/include/expat" "-I{compiler.sdk.path}/include/fatfs" "-I{compiler.sdk.path}/include/freemodbus" "-I{compiler.sdk.path}/include/freertos" "-I{compiler.sdk.path}/include/heap" "-I{compiler.sdk.path}/include/idf_test" "-I{compiler.sdk.path}/include/jsmn" "-I{compiler.sdk.path}/include/json" "-I{compiler.sdk.path}/include/libsodium" "-I{compiler.sdk.path}/include/log" "-I{compiler.sdk.path}/include/lwip" "-I{compiler.sdk.path}/include/mbedtls" "-I{compiler.sdk.path}/include/mdns" "-I{compiler.sdk.path}/include/micro-ecc" "-I{compiler.sdk.path}/include/mqtt" "-I{compiler.sdk.path}/include/newlib" "-I{compiler.sdk.path}/include/nghttp" "-I{compiler.sdk.path}/include/nvs_flash" "-I{compiler.sdk.path}/include/openssl" "-I{compiler.sdk.path}/include/protobuf-c" "-I{compiler.sdk.path}/include/protocomm" "-I{compiler.sdk.path}/include/pthread" "-I{compiler.sdk.path}/include/sdmmc" "-I{compiler.sdk.path}/include/smartconfig_ack" "-I{compiler.sdk.path}/include/soc" "-I{compiler.sdk.path}/include/spi_flash" "-I{compiler.sdk.path}/include/spiffs" "-I{compiler.sdk.path}/include/tcp_transport" "-I{compiler.sdk.path}/include/tcpip_adapter" "-I{compiler.sdk.path}/include/ulp" "-I{compiler.sdk.path}/include/vfs" "-I{compiler.sdk.path}/include/wear_levelling" "-I{compiler.sdk.path}/include/wifi_provisioning" "-I{compiler.sdk.path}/include/wpa_supplicant" "-I{compiler.sdk.path}/include/xtensa-debug-module" "-I{compiler.sdk.path}/include/esp-face" "-I{compiler.sdk.path}/include/esp32-camera" "-I{compiler.sdk.path}/include/esp-face" "-I{compiler.sdk.path}/include/fb_gfx"

compiler.c.cmd=xtensa-esp32-elf-gcc
compiler.c.flags=-std=gnu99 -Os -g3 -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wpointer-arith {compiler.warning_flags} -Wno-maybe-uninitialized -Wno-unused-function -Wno-unused-but-set-variable -Wno-unused-variable -Wno-deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -Wno-old-style-declaration -MMD -c

compiler.cpp.cmd=xtensa-esp32-elf-g++
compiler.cpp.flags=-std=gnu++11 -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib {compiler.warning_flags} -Wno-error=maybe-uninitialized -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-unused-but-set-parameter -Wno-missing-field-initializers -Wno-sign-compare -fno-rtti -MMD -c

compiler.S.cmd=xtensa-esp32-elf-gcc
compiler.S.flags=-c -g3 -x assembler-with-cpp -MMD -mlongcalls

compiler.c.elf.cmd=xtensa-esp32-elf-gcc
compiler.c.elf.flags=-nostdlib "-L{compiler.sdk.path}/lib" "-L{compiler.sdk.path}/ld" -T esp32_out.ld -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.libgcc.ld -T esp32.rom.spiram_incompatible_fns.ld -u ld_include_panic_highint_hdl -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,--undefined=uxTopUsedPriority  -u __cxa_guard_dummy -u __cxx_fatal_exception
compiler.c.elf.libs=-lgcc -lesp32 -lphy -lesp_http_client -lmbedtls -lrtc -lesp_http_server -lbtdm_app -lspiffs -lbootloader_support -lmdns -lnvs_flash -lfatfs -lpp -lnet80211 -ljsmn -lface_detection -llibsodium -lvfs -ldl_lib -llog -lfreertos -lcxx -lsmartconfig_ack -lxtensa-debug-module -lheap -ltcpip_adapter -lmqtt -lulp -lfd -lfb_gfx -lnghttp -lprotocomm -lsmartconfig -lm -lethernet -limage_util -lc_nano -lsoc -ltcp_transport -lc -lmicro-ecc -lface_recognition -ljson -lwpa_supplicant -lmesh -lesp_https_ota -lwpa2 -lexpat -llwip -lwear_levelling -lapp_update -ldriver -lbt -lespnow -lcoap -lasio -lnewlib -lconsole -lapp_trace -lesp32-camera -lhal -lprotobuf-c -lsdmmc -lcore -lpthread -lcoexist -lfreemodbus -lspi_flash -le [... cut ...] --flash_size detect 0xe000 "{runtime.platform.path}/tools/partitions/boot_app0.bin" 0x1000 "{runtime.platform.path}/tools/sdk/bin/bootloader_{build.boot}_{build.flash_freq}.bin" 0x10000 "{build.path}/{build.project_name}.bin" 0x8000 "{build.path}/{build.project_name}.partitions.bin"
tools.esptool_py.upload.network_pattern={network_cmd} -i "{serial.port}" -p "{network.port}" "--auth={network.password}" -f "{build.path}/{build.project_name}.bin"
`

The above code is the platform.txt of my computer

lbernstone, 2020-04-13:
You need to ask on whatever site you got your instructions for installing eclipse.  This is a problem with your code, not with the code base.  Read the error carefully and figure out where you are missing quotes.

closed 2020-04-14 by LusterWong

hreintke, 2020-04-14:
@LusterWong 
Which arduino/eclipse environment are you using ?

LusterWong, 2020-05-23:
> @LusterWong
> Which arduino/eclipse environment are you using ?
Eclipse IDE for C/C++ Developers
Version: 2020-03 (4.15.0)
Build id: 20200313-1211
OS: Windows 7, v.6.1, x86_64 / win32
Java version: 1.8.0_171