espressif/arduino-esp32 #3902: ESP32 library missing WebServerSecure ?

Twilight-Logic, 2020-04-12:
### Hardware:
Board:							ESP32 Devkit V1, NodeMCU-32s compatible
								ESP32 WROOM-32 WiFi module
Core Installation version:			1.0.4
IDE name:						Arduino IDE 1.8.10, 1.8.12
Flash Frequency:					not sure
PSRAM enabled:                                    not sure
Upload Speed:					115200
Computer OS:                                        Windows 7, Linux Mint 19.2

### Description:

On the ESP32, WebServer corresponds to the ESP8266WebServer module. However, I am puzzled by the fact that when I type in #include <WebServerSecure.h> it turns bold red as though it was a recognized module, but I then get a compiler error telling me:

`WebServerSecure.h: No such file or directory
`
It seemed a good idea to compared the corresponding paths in each library:

`~/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/ESP8266WebServer`

and

`~/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WebServer`

Among other things, this contains a file called keywords.txt. The ESP32 version contains as follows:

Code: [Select]

#######################################
# Datatypes (KEYWORD1)
#######################################

WebServer KEYWORD1
**WebServerSecure KEYWORD1**
HTTPMethod KEYWORD1

So clearly 'WebServerSecure' is present as a defined keyword, however, although the ESP8266 path contains ESP8266WebServerSecure.h and ESPWebServerSecure.cpp files, the ESP32 does NOT contain either a corresponding WebServerSecure.h or WebServerSecure.cpp files as perhaps might be expected. Having had a look at the ESP32 Git repository I discover the same.

Can anyone shed any light on this? Is this a file omission from the library, or was the secure version  never implemented?

### Sketch: 

n/a
```cpp

//Change the code below by your sketch
#include <Arduino.h>
#include <WebServer.h>
#include <WebServerSecure.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Arduino: 1.8.12 (Linux), Board: "NodeMCU-32S, 80MHz, 921600"

Multiple libraries were found for "WiFi.h"
AR488-ESP32-test2:73:33: fatal error: WebServerSecure.h: No such file or directory
 Used: /home/johnc/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi
compilation terminated.
 Not used: /opt/arduino/libraries/WiFi
exit status 1
WebServerSecure.h: No such file or directory

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

Twilight-Logic, 2020-06-14:
Could we have a reply to this please?

OBorce, 2020-08-01:
+1 for WebServerSecure

Twilight-Logic, 2020-08-03:
Still waiting for a reply from the devs?

luc-github, 2020-08-03:
why not use  https://github.com/fhessel/esp32_https_server ? 
there is even a compat layer : https://github.com/fhessel/esp32_https_server_compat

Twilight-Logic, 2020-08-03:
Tried that. I did spend a few days re-developing my code for this library and even contacted the author because I had some issues with it which he was able to help me resolve. However, this library is completely different, MUCH bigger in terms of code size and memory usage, much more complex, and seems VERY slow. Furthermore this library is not available for the ESP8266 so this means I would have to support two separate code bases for the ESP8266 and ESP32 when surely a compatible and consistent approach could have been implemented?

luc-github, 2020-08-03:
hmm https://github.com/fhessel/esp32_https_server_compat do not use same syntax for API ?

Twilight-Logic, 2020-08-03:
I had not noticed the esp32_https_server_compat library before. Thanks for pointing it out. This will add another layer and hence further overhead, but might be worth a try. The quoted inefficiency of POST handling and Basic authentication only might be issues but still worth a test perhaps.

luc-github, 2020-08-03:
Looking at https://github.com/fhessel/esp32_https_server_compat#state-of-development still some work need to be done so not sure it will meet your expectation anyway

For what it worth, on my side, I have noticed arduino Esp32 core webserver is slower than esp8266 so I do not think https version will be faster - I did not tested esp32_https_server yet but got good feedback and it is actively maintained 

Another solution would be to use the native http / https from IDF - it is async and looks faster as it is used for camera but API is limited and need to be ported :

https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/esp_http_server/esp_http_server.h / https://github.com/espressif/arduino-esp32/tree/master/tools/sdk/include/esp_https_server

ZakCodes, 2020-10-03:
I think this should stay opened

Twilight-Logic, 2020-10-05:
I think they are probably expecting us to use the incompatible fhessel library but would be helpful to have this confirmed by the devs please.

Twilight-Logic, 2020-12-05:
Ok, looks like we are not going to get any reply to this then.

Rp777, 2020-12-15:
+1 for WebServerSecure

ZakCodes, 2021-02-13:
let's keep this open

FotoFieber, 2021-02-18:
+1 for WebServerSecure

Twilight-Logic, 2021-02-18:
Any update from developers?

me-no-dev, 2021-02-18:
https server will be added in the next major version of Arduino

ZakCodes, 2021-06-20:
Let's keep this open

VojtechBartoska, 2022-03-30:
Hello @Twilight-Logic , sorry for late reply. Are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? You can take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

Twilight-Logic, 2022-03-30:
Thank you for your response. I have tried both versions 1.8.19 and 2.0.0 rc5 of the IDE:

```
Arduino: 1.8.19 (Linux), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

AR488-ESP32-test2:80:33: fatal error: WebServerSecure.h: No such file or directory
 Used: /home/johnc/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/WiFi
compilation terminated.
 Not used: /opt/arduino/libraries/WiFi
exit status 1
WebServerSecure.h: No such file or directory

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

```
IDE2.0 rc5
/mnt/ntfs/Data/Data/Arduino/AR488/AR488-ESP32-test2/AR488-ESP32-test2.ino:80:33: fatal error: WebServerSecure.h: No such file or directory
compilation terminated.
Multiple libraries were found for "WiFi.h"
  Used: /home/johnc/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/WiFi
  Not used: /opt/arduino/libraries/WiFi
Compilation error: exit status 1
```

igrr, 2022-04-05:
Let's keep this feature request open.

Twilight-Logic, 2022-04-06:
Than you. Really could do with a lightweight secure webserver on the ESP32 just like on the ESP8266. The alternative unfortunately is rather unwieldly and memory hungry.

(If anyone has manually added the original ESP8266 WebseverSecure library - and I have heard that is has been done - I would be interested to know how it was done? The ESP32 library seems to have a somewhat different folder structure.)

hemangjoshi37a, 2024-06-12:
has this been solved yet ?

closed 2025-01-16 by Parsaabasi